coolsnowwolf/lede #3702: 3月10日git pull的adbyby没有域名白名单了。

norikohxb, 2020-03-10:
[image]
请lean大雕拨冗修复一下。不胜感激。

norikohxb, 2020-03-10:
[image]
[image]

adbyby的hostslist无法更新。
然后新的ssrplus无法访问www.wujieliulan.com

norikohxb, 2020-03-10:
加入了强制走代理的域名也无法访问
已经吧www.wujieliulan.com加入到强制走代理的域名。

closed 2020-03-10 by coolsnowwolf

hplee0120, 2020-03-10:
这个广告屏蔽大师有域名黑名单 白名单 ip黑名单 为什么单独少了一个ip白名单

norikohxb, 2020-03-10:
lean大反馈一下，wujieliulan不可以访问是因为开启了ssr的去广告，wujieliulan的网站在屏蔽列表里面
https://gitee.com/privacy-protection-tools/anti-ad/raw/master/anti-ad-for-dnsmasq.conf
里面有www.wujieliulan.com

norikohxb, 2020-03-10:
lean大再反馈一下，我之前反映的adbyby不可以升级blacklist是因为我编译的时候去掉了
urandom-seed. /etc/urandom.seed handling for OpenWrt   
为何去掉这个会造成adbyby今天更新的版本无法升级blacklist，之前都可以，之前我都是这么干的。

norikohxb, 2020-03-10:
这个ssr的屏蔽规则能否添加白名单，我查看了引用的list把无界浏览给屏蔽了。
今天我朋友让我帮他下载无界浏览的exe，然后突然发现网站打不开。还有能否解答一下，为什么昨天以前的一个月内，我没有编译urandom-seed 我也可以更新adbyby，今天就不行了？

hplee0120, 2020-03-10:
你说的这个urandom-seed可能跟规则更新没有关系。。我猜测的。因为看你提的规则无法更新的时候大雕后面又推了一次更新，然后关闭了你这个问题。就是针对这个规则更新的。我猜是你提了这个问题之后是不是有重新git pull过？

norikohxb, 2020-03-10:
不选择urandom编译确实无法更新blacklist

norikohxb, 2020-03-11:
我今天又更新了，不编译urandom是可以更新blocklist了，估计我太着急了，昨天lean大雕还没有完全更新好代码，我就去gitpull了。导致了。