coolsnowwolf/lede #3705: ssr“不走代理的域名”不生效

kl9800, 2020-03-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题

A：
1.我开着ssr+（绕过模式），访问www.toutiao.com，网页无新闻也没有图片，关闭ssr+以后，正常，怀疑是把头条的IP当成了外国IP？
2.然后我把toutiao.com加入了“不走代理的域名”里面，也无效,域名这个功能没生效吗？

### (2) 路由器型号和固件版本

路由器型号：x86 j1900
固件版本：OpenWrt R20.3.8 / LuCI Master (git-20.068.29105-b2cf61d)

### (3) 详细日志

A：

R3pl4c3r, 2020-03-10:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [X ] 没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：
> 1.我开着ssr+（绕过模式），访问www.toutiao.com，网页无新闻也没有图片，关闭ssr+以后，正常，怀疑是把头条的IP当成了外国IP？ [...]

因为你开了广告屏蔽被误杀了

kl9800, 2020-03-10:
我去查查，先谢谢

kl9800, 2020-03-10:
> 因为你开了广告屏蔽被误杀了

果然是！谢谢！！！

closed 2020-03-10 by kl9800