coolsnowwolf/lede #3707: 固件编译问题

TomNotchYmk, 2020-03-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：非固件运行问题，编译问题。

[image]

[image]

### (2) 编译机器与配置

A：vps: 谷歌云香港asia-east2-b区 n1-standard （1核）
      系统：谷歌云提供的ubuntu 18.04 LTS
      menuconfig: luci全部application + 全部theme, 其余default

### (3) 详细日志

A：

make[4]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/arprefresh'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/arprefresh'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/bmf'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/bmf'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/dot_draw'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/dot_draw'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/dyn_gw'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/dyn_gw'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/dyn_gw_plain'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/dyn_gw_plain'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/httpinfo'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/httpinfo'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/jsoninfo'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/jsoninfo'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/netjson'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/netjson'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/mdns'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/mdns'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/nameservice'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/nameservice'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/p2pd'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/p2pd'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/pgraph'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/pgraph'
make[5]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/pud'
make[6]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/pud/nmealib'
make[6]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/pud/nmealib'
make[6]: Entering directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/pud/wireformat'
make[6]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/pud/wireformat'
[CC] src/gpsdclient.c
src/gpsdclient.c: In function 'nmeaInfoFromGpsd':
src/gpsdclient.c:291:101: warning: format '%f' expects argument of type 'double', but argument 12 has type 'timespec_t {aka struct timespec}' [-Wformat=]
         syslog(LOG_INFO, "Using %s device%s%s on %s in %s mode at %u baud (%u%c%u), refresh time %.3f (min. %.3f)", //
                                                                                                  ~~~^
src/gpsdclient.c:301:13:
             dev->cycle, //
             ~~~~~~~~~~
src/gpsdclient.c:291:112: warning: format '%f' expects argument of type 'double', but argument 13 has type 'timespec_t {aka struct timespec}' [-Wformat=]
         syslog(LOG_INFO, "Using %s device%s%s on %s in %s mode at %u baud (%u%c%u), refresh time %.3f (min. %.3f)", //
                                                                                                             ~~~^
src/gpsdclient.c:302:13:
             dev->mincycle);
             ~~~~~~~~~~~~~
In file included from ../../lib/pud/nmealib/include/nmealib/info.h:21:0,
                 from src/gpsdclient.h:51,
                 from src/gpsdclient.c:46:
../../lib/pud/nmealib/include/nmealib/util.h:48:21: error: invalid operands to binary != (have 'timespec_t {aka struct timespec}' and 'timespec_t {aka struct timespec}')
 #define isNaN(x) (x != x)
                     ^
src/gpsdclient.c:370:8: note: in expansion of macro 'isNaN'
   if (!isNaN(gpsdata->fix.time)) {
        ^~~~~
src/gpsdclient.c:372:33: error: incompatible type for argument 1 of 'fabs'
     double fraction = modf(fabs(gpsdata->fix.time), &seconds);
                                 ^~~~~~~
In file included from ../../lib/pud/wireformat/include/OlsrdPudWireFormat/wireFormat.h:57:0,
                 from src/configuration.h:59,
                 from src/gpsdclient.c:48:
/home/2215776228/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/math.h:203:13: note: expected 'double' but argument is of type 'timespec_t {aka struct timespec}'
 double      fabs(double);
             ^~~~
src/gpsdclient.c:390:7: error: wrong type argument to unary exclamation mark
   if (!gpsdata->online) {
       ^
src/gpsdclient.c:457:29: error: 'struct gps_data_t' has no member named 'separation'; did you mean 'version'?
     info->height = gpsdata->separation;
                             ^~~~~~~~~~
                             version
src/gpsdclient.c: In function 'readFromGpsd':
src/gpsdclient.c:581:19: error: too few arguments to function 'gps_read'
     gpsReadCode = gps_read(gpsdata);
                   ^~~~~~~~
In file included from src/gpsdclient.h:49:0,
                 from src/gpsdclient.c:46:
/home/2215776228/lede/staging_dir/target-x86_64_musl/usr/include/gps.h:2307:12: note: declared here
 extern int gps_read(struct gps_data_t *, char *message, int message_len);
            ^~~~~~~~
../../Makefile.inc:351: recipe for target 'src/gpsdclient.o' failed
make[5]: *** [src/gpsdclient.o] Error 1
make[5]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/lib/pud'
Makefile:241: recipe for target 'libs' failed
make[4]: *** [libs] Error 2
make[4]: Leaving directory '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7'
Makefile:293: recipe for target '/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/.built' failed
make[3]: *** [/home/2215776228/lede/build_dir/target-x86_64_musl/olsrd-0.9.7/.built] Error 2
make[3]: Leaving directory '/home/2215776228/lede/feeds/routing/olsrd'
time: package/feeds/routing/olsrd/compile#0.32#0.12#0.46
package/Makefile:111: recipe for target 'package/feeds/routing/olsrd/compile' failed
make[2]: *** [package/feeds/routing/olsrd/compile] Error 2
make[2]: Leaving directory '/home/2215776228/lede'
package/Makefile:107: recipe for target '/home/2215776228/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/2215776228/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/2215776228/lede'
/home/2215776228/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2020-03-10:
目前代码不支持olsrd，等上游修复

TomNotchYmk, 2020-03-10:
应该去掉哪个luci-app ?

closed 2020-03-11 by coolsnowwolf

AmadeusGhost, 2020-03-11:
> 应该去掉哪个luci-app ?

你找跟olsrd相关的包啊