coolsnowwolf/lede #3708: 咱们编译的固件是不支持HDMI输出么？

liyuhaolol, 2020-03-10:
我下载了您恩山论坛3月5日发布的X86编译固件遇到一个问题
链接是这个https://www.right.com.cn/forum/thread-2798572-1-1.html
我的软路由只有一个HDMI输出口，之前一直在用koolshare的固件，今日用了大神的固件发现写入磁盘后，重启电脑会一直黑屏什么都不显示，我还以为是我弄坏了，后来偶然发现，192.168.1.1可以进行访问。
我想问这是大神没有加入hdmi输出支持的问题么？可以给稍微解答一下么？

EnnawYang, 2020-03-10:
你该更新了

liyuhaolol, 2020-03-10:
> 你该更新了

更新是指什么意思？我并不会自己编译固件，3月5日的版本是作者目前在恩山发布的最新版啊？

R3pl4c3r, 2020-03-10:
> 我下载了您恩山论坛3月5日发布的X86编译固件遇到一个问题
> 链接是这个https://www.right.com.cn/forum/thread-2798572-1-1.html
> 我的软路由只有一个HDMI输出口，之前一直在用koolshare的固件，今日用了大神的固件发现写入磁盘后，重启电脑会一直黑屏什么都不显示，我还以为是我弄坏了，后来偶然发现，192.168.1.1可以进行访问。
> 我想问这是大神没有加入hdmi输出支持的问题么？可以给稍微解答一下么？

你看一下这个 #3575 把核显支持加上

hxlls, 2020-03-10:
首先大雕很久没专门发布什么最新版，默认编译的版本这个github 就可以下载，其次，你下载是3方编译的版本，HDMI 无法显示的原因可能是因为加入了GPU的驱动，openwrt 能调用GPU ，但是HDMI 显示会出现问题

closed 2020-03-11 by coolsnowwolf