coolsnowwolf/lede #3713: UnblockNeteaseMusicGO是否可以增加在线更新功能？

yinzhefuyun, 2020-03-11:
UnblockNeteaseMusicGO是否可以像nodejs一样增加在线更新功能？
最新的go版本0.1.9 https://github.com/cnsilvan/UnblockNeteaseMusic/commit/8dd3f9779fcdaa271414a85b80f43f11300e08d3 修复了 https://github.com/coolsnowwolf/lede/issues/3613 的bug
而源代码还是0.1.6

AmadeusGhost, 2020-03-11:
https://github.com/coolsnowwolf/lede/commit/830096f96025de7c5da980cc6644584902518da9 不是已经1.0.9了吗

yinzhefuyun, 2020-03-11:
> [830096f](https://github.com/coolsnowwolf/lede/commit/830096f96025de7c5da980cc6644584902518da9) 不是已经1.0.9了吗

我是建议加一个在线更新的功能和nodejs一样

Kisoul, 2020-03-11:
nodejs,我的无法启动，不知道为什么？编译时间是3月10日

coolsnowwolf, 2020-03-11:
go 程序太大，不提供在线升级功能

mjyhj, 2020-03-12:
> > [830096f](https://github.com/coolsnowwolf/lede/commit/830096f96025de7c5da980cc6644584902518da9) 不是已经1.0.9了吗
> 
> 我是建议加一个在线更新的功能和nodejs一样

不太可能，go版要编译，node版是直接拉取项目文件运行

DongfeiSay, 2020-03-13:
我倒是希望能增加一个是否启用FLAC解锁的选择

closed 2020-03-28 by coolsnowwolf