coolsnowwolf/lede #3717: xxrp更新v2订阅后，danmasq无法启动

Aspiriniii, 2020-03-11:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：xxrp更新v2订阅后，无论更新是否成功，danmasq都会挂掉

### (2) 路由器型号和固件版本

A：x86_64     commit f9cb999

### (3) 详细日志

A：
**ssrp log**
2020-03-11 11:20:35 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
2020-03-11 11:20:33 订阅更新成功
2020-03-11 11:20:33 新增节点数量: 0 删除节点数量: 0
2020-03-11 11:20:33 当前主服务器节点已被删除，正在自动更换为第一个节点。
2020-03-11 11:20:33 成功解析节点数量: 13
2020-03-11 11:20:33 成功解析: v2ray 节点, xxx
2020-03-11 11:20:33 (51.0 MB/s) - written to stdout [4546/4546]

     0K ....                                                  100% 51.0M=0s

Saving to: 'STDOUT'
Length: 4546 (4.4K) [application/octet-stream]
HTTP request sent, awaiting response... 200 OK
Connecting to xxx|xxx|:443... connected.
Resolving xxx... xxx
--2020-03-11 11:20:32--  https://xxxx
2020-03-11 11:20:31 服务正在暂停

**system log**
Wed Mar 11 11:16:12 2020 cron.info crond[12497]: crond (busybox 1.30.0) started, log level 5
Wed Mar 11 11:16:13 2020 cron.info crond[12644]: crond (busybox 1.30.0) started, log level 5
Wed Mar 11 11:20:05 2020 daemon.err uhttpd[2987]: cat: can't open '/tmp/ssrplus.log': No such file or directory
Wed Mar 11 11:20:32 2020 cron.info crond[11354]: crond (busybox 1.30.0) started, log level 5
Wed Mar 11 11:20:34 2020 cron.info crond[11909]: crond (busybox 1.30.0) started, log level 5
Wed Mar 11 11:20:35 2020 daemon.info pdnsd[12219]: pdnsd-1.2.9b-par starting.
Wed Mar 11 11:20:35 2020 daemon.warn pdnsd[12219]: EOF encountered while reading entry count from disk cache file.
Wed Mar 11 11:20:41 2020 daemon.crit dnsmasq[13091]: cannot access directory /usr/share/adbyby/rules/data: No such file or directory
Wed Mar 11 11:20:41 2020 daemon.crit dnsmasq[13091]: FAILED to start up
Wed Mar 11 11:20:46 2020 daemon.crit dnsmasq[13215]: cannot access directory /usr/share/adbyby/rules/data: No such file or directory
Wed Mar 11 11:20:46 2020 daemon.crit dnsmasq[13215]: FAILED to start up
Wed Mar 11 11:20:51 2020 daemon.crit dnsmasq[13355]: cannot access directory /usr/share/adbyby/rules/data: No such file or directory
Wed Mar 11 11:20:51 2020 daemon.crit dnsmasq[13355]: FAILED to start up
Wed Mar 11 11:20:56 2020 daemon.crit dnsmasq[13496]: cannot access directory /usr/share/adbyby/rules/data: No such file or directory
Wed Mar 11 11:20:56 2020 daemon.crit dnsmasq[13496]: FAILED to start up
Wed Mar 11 11:21:01 2020 daemon.crit dnsmasq[13658]: cannot access directory /usr/share/adbyby/rules/data: No such file or directory
Wed Mar 11 11:21:01 2020 daemon.crit dnsmasq[13658]: FAILED to start up
Wed Mar 11 11:21:06 2020 daemon.crit dnsmasq[13802]: cannot access directory /usr/share/adbyby/rules/data: No such file or directory
Wed Mar 11 11:21:06 2020 daemon.crit dnsmasq[13802]: FAILED to start up
Wed Mar 11 11:21:06 2020 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash

Aspiriniii, 2020-03-11:
试了下跟更新Adblock Plus规则没什么关系，只有更新xxrp订阅才会dnsmasq FAILED to start up

Aspiriniii, 2020-03-11:
xxrp开的gfwlist模式

coolsnowwolf, 2020-03-11:
/usr/share/adbyby/rules/data 都写着没有这个目录了

closed 2020-03-11 by coolsnowwolf