coolsnowwolf/lede #3721: 挂载硬盘失败

coolice231, 2020-03-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
编译的3.8固件挂载硬盘失败，编译已加入了automount，3.5及以前版本没有问题
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：主机型号	Newifi-D2
架构	MediaTek MT7621 ver:1 eco:3 x 4
固件版本	OpenWrt R20.3.8 / LuCI Master (git-20.068.29105-b2cf61d)
内核版本	4.14.171

### (3) 详细日志

A：Wed Mar 11 13:07:18 2020 kern.info kernel: [  225.880840] usb 2-1: USB disconnect, device number 2
Wed Mar 11 13:07:18 2020 kern.warn kernel: [  225.886172] xhci-mtk 1e1c0000.xhci: Mismatch between completed Set TR Deq Ptr command & xHCI internal state.
Wed Mar 11 13:07:18 2020 kern.warn kernel: [  225.895958] xhci-mtk 1e1c0000.xhci: ep deq seg = 8f5e8480, deq ptr = af62b010
Wed Mar 11 13:07:26 2020 kern.info kernel: [  234.240942] usb 2-1: new SuperSpeed USB device number 3 using xhci-mtk
Wed Mar 11 13:07:26 2020 kern.info kernel: [  234.275249] scsi host0: uas
Wed Mar 11 13:07:26 2020 kern.err kernel: [  234.279222] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 4
Wed Mar 11 13:07:26 2020 kern.err kernel: [  234.287645] xhci-mtk 1e1c0000.xhci: @000000000f6a53c0 0f5e9200 00000000 05000000 01058001
Wed Mar 11 13:07:26 2020 kern.err kernel: [  234.295785] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 6
Wed Mar 11 13:07:26 2020 kern.err kernel: [  234.304182] xhci-mtk 1e1c0000.xhci: @000000000f6a53d0 0f5e9100 00000000 05000000 01078001
Wed Mar 11 13:07:47 2020 kern.info kernel: [  255.300586] scsi 0:0:0:0: tag#0 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD IN
Wed Mar 11 13:07:47 2020 kern.info kernel: [  255.308148] scsi 0:0:0:0: tag#0 CDB: opcode=0x12 12 00 00 00 24 00
Wed Mar 11 13:07:47 2020 kern.warn kernel: [  255.314544] xhci-mtk 1e1c0000.xhci: Mismatch between completed Set TR Deq Ptr command & xHCI internal state.
Wed Mar 11 13:07:47 2020 kern.warn kernel: [  255.324338] xhci-mtk 1e1c0000.xhci: ep deq seg = 8f5f2b80, deq ptr = af64c010
Wed Mar 11 13:07:49 2020 kern.err kernel: [  256.410565] scsi host0: uas_eh_device_reset_handler FAILED to get lock err -16
Wed Mar 11 13:07:49 2020 kern.info kernel: [  256.417820] scsi 0:0:0:0: Device offlined - not ready after error recovery

R3pl4c3r, 2020-03-11:
把 5dc5b107a219179d96bd6b3217a0d780d70fca09 这个commit里面的depends后面的东西都勾上

liyiranlab, 2020-03-12:
我的7621 r3g，OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7)
Extra packages  --->   <*> automount
kmod-block2mtd +kmod-ata-ahci +kmod-ata-core +kmod-scsi-generic +kmod-usb-ohci +kmod-usb-uhci +kmod-usb2 +kmod-usb3
硬盘也挂不上，usb3.0的u可以挂上
Thu Mar 12 13:00:44 2020 kern.info kernel: [  477.207210] usb 2-1: new SuperSpeed USB device number 3 using xhci-mtk
Thu Mar 12 13:00:44 2020 kern.info kernel: [  477.241569] scsi host0: uas
Thu Mar 12 13:00:44 2020 kern.err kernel: [  477.245608] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 4
Thu Mar 12 13:00:44 2020 kern.err kernel: [  477.254082] xhci-mtk 1e1c0000.xhci: @000000000ed653c0 0ecfa200 00000000 05000000 01058001
Thu Mar 12 13:00:44 2020 kern.err kernel: [  477.262267] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 6
Thu Mar 12 13:00:44 2020 kern.err kernel: [  477.270697] xhci-mtk 1e1c0000.xhci: @000000000ed653d0 0ecfa100 00000000 05000000 01078001

liyiranlab, 2020-03-12:
u盘挂载信息
Thu Mar 12 13:03:36 2020 kern.info kernel: [  649.386870] usb 2-1: new SuperSpeed USB device number 4 using xhci-mtk
Thu Mar 12 13:03:36 2020 kern.info kernel: [  649.419198] usb-storage 2-1:1.0: USB Mass Storage device detected
Thu Mar 12 13:03:36 2020 kern.info kernel: [  649.426442] scsi host0: usb-storage 2-1:1.0
Thu Mar 12 13:03:37 2020 kern.notice kernel: [  650.477825] scsi 0:0:0:0: Direct-Access     Kingston DataTraveler 3.0 PMAP PQ: 0 ANSI: 6
Thu Mar 12 13:03:37 2020 kern.notice kernel: [  650.487389] sd 0:0:0:0: Attached scsi generic sg0 type 0
Thu Mar 12 13:03:37 2020 kern.notice kernel: [  650.493359] sd 0:0:0:0: [sda] 30277632 512-byte logical blocks: (15.5 GB/14.4 GiB)
Thu Mar 12 13:03:37 2020 kern.notice kernel: [  650.501474] sd 0:0:0:0: [sda] Write Protect is off
Thu Mar 12 13:03:37 2020 kern.debug kernel: [  650.506254] sd 0:0:0:0: [sda] Mode Sense: 45 00 00 00
Thu Mar 12 13:03:37 2020 kern.notice kernel: [  650.506562] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
Thu Mar 12 13:03:39 2020 kern.info kernel: [  651.953126]  sda: sda1
Thu Mar 12 13:03:39 2020 kern.notice kernel: [  651.958301] sd 0:0:0:0: [sda] Attached SCSI removable disk

coolice231, 2020-03-12:
好像是大佬更新了automount后就挂不上了

R3pl4c3r, 2020-03-12:
执行一下这个
```bash
git pull
make target/kernel{clean,compile} -j4
rm -rf tmp/ .config* 
```
之后再`make menuconfig`配置并编译

liyiranlab, 2020-03-12:
machine:~/lede$ make target/kernel{clean,compile} -j4
 make[1] target/kernelclean
make -r target/kernelclean: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/ran/lede/include/toplevel.mk:225: recipe for target 'target/kernelclean' failed
make: *** [target/kernelclean] Error 1
出错了，我只能全部清理了

liyiranlab, 2020-03-12:
r3g usb3.0还是不行，另外只<*> automount的小米 mini usb2.0 可以正确挂载硬盘
git pull
make clean
rm -rf tmp/ .config* 
make menuconfig
编译时候添加
Extra packages ---> <*> automount
kmod-block2mtd +kmod-ata-ahci +kmod-ata-core +kmod-scsi-generic +kmod-usb-ohci +kmod-usb-uhci +kmod-usb2 +kmod-usb3 +kmod-nls-cp936 

Wed Mar 11 23:04:49 2020 kern.info kernel: [  120.597472] usb 2-1: new SuperSpeed USB device number 2 using xhci-mtk
Wed Mar 11 23:04:49 2020 kern.info kernel: [  120.631859] scsi host0: uas
Wed Mar 11 23:04:49 2020 kern.err kernel: [  120.635932] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 4
Wed Mar 11 23:04:49 2020 kern.err kernel: [  120.644380] xhci-mtk 1e1c0000.xhci: @000000000ed36170 0d0d9000 00000000 05000000 01058001
Wed Mar 11 23:04:49 2020 kern.err kernel: [  120.652532] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 6
Wed Mar 11 23:04:49 2020 kern.err kernel: [  120.660933] xhci-mtk 1e1c0000.xhci: @000000000ed36180 0d0d9100 00000000 05000000 01078001

R3pl4c3r, 2020-03-12:
Base System里面的blockd勾上再试试

R3pl4c3r, 2020-03-12:
> Base System里面的blockd勾上再试试

这个包的Description是：Block Device Automount.
倘若还不行就只能自己手动mount硬盘了，使用方法：
```bash
mkdir /mnt/sdxN    #创建分区文件夹
mount /dev/sdxN /mnt/sdxN    #挂载磁盘
```
这里的sdxN的x是你的磁盘代表的字母，比如我的主磁盘是sda，U盘是sdb. 而N是分区(partition) ，从1开始计数，我的主磁盘的第一个分区就是sda1，第二个分区是sda2，U盘的第一个分区是sdb1，第二个分区是sdb2.
假如我要挂载我的主磁盘上第5个分区，那么我就要输入：
```bash
mkdir /mnt/sda5
mount /dev/sda5 /mnt/sda5
```

liyiranlab, 2020-03-12:
在前面的基础上添加了Base System里面的blockd还是连不上移动硬盘，
mkdir /mnt/sda1
root@OpenWrt:~# mount -t ntfs-3g -o /dev/sda1 /mnt/sda1
mount: can't find /mnt/sda1 in /etc/fstab
好像是不能识别硬盘吧，不是什么挂载不上！

R3pl4c3r, 2020-03-12:
移动硬盘？你kernel modules里面USB support里面的kmod-usb-extra-uas是否勾选？

liyiranlab, 2020-03-12:
```
-*- kmod-usb-storage..................................... USB Storage support
-*- kmod-usb-storage-extras.................... Extra drivers for usb-storage
-*- kmod-usb-storage-uas.................... USB Attached SCSI (UASP) support 
```
自动勾勾选的吧。

liyiranlab, 2020-03-12:
换了个大的电源，可以排除电源带不动的情况，3.0的u盘却可以正常使用，奇怪了

coolice231, 2020-03-24:
已修复，谢谢大佬

closed 2020-03-24 by coolice231

ccy361, 2020-12-09:
> 换了个大的电源，可以排除电源带不动的情况，3.0的u盘却可以正常使用，奇怪了

你确定驱动是uas,而不是usb-storage?
我这里u盘使用usb3.0时，可以挂载，读写，但是使用lsusb -t 查看发现使用的驱动不是uas,而是usb-storage。
如下：
root@OpenWrt:~# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
    |__ Port 1: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M
根据资料usb-storage是usb2.0的驱动。uas才是