coolsnowwolf/lede #3722: 遇到奇怪的端口转发问题，share

demonlj, 2020-03-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：打开flow offloading，会出现端口转发故障。具体表现为：
1、openwrt的监听端口，没问题
2、openwrt转发到局域网的端口，非选择性故障。 比如8688 -> 192.168.9.6:80正常。而2622 -> 192.168.9.6:22则无法完成ssh握手

### (2) 路由器型号和固件版本

A：gen8 + proxmox + OpenWrt R20.3.5 / LuCI Master (git-20.068.29105-b2cf61d)

### (3) 详细日志

A：关闭flow offloading一切正常

goukey, 2020-03-11:
确实有此问题，我不得已已经关闭flow offloading加速，只保留了bbr

Aspiriniii, 2020-03-12:
确实有问题，开启flow offloading的时候，配置转发wan:8088 -> 192.168.1.252:3389，通过外网ip连接8088端口无法完成握手，关闭flow offloading则没有问题。
但是开flow offloading的情况下，配置转发wan:3389 -> 192.168.1.252:3389，通过外网ip连接3389端口则可以正常连接。
怀疑flow offloading对dnat产生了莫名其妙的影响。

yihongyuelan, 2020-03-15:
> 确实有此问题，我不得已已经关闭flow offloading加速，只保留了bbr

我这是只需要关闭Flow Offloading ，BBR和DNS加速都保留了的，只是每次开关Flow Offloading 都需要重启路由

tty228, 2020-03-28:
> > 确实有此问题，我不得已已经关闭flow offloading加速，只保留了bbr
> 
> 我这是只需要关闭Flow Offloading ，BBR和DNS加速都保留了的，只是每次开关Flow Offloading 都需要重启路由

试试关闭网易云跟adbyby

closed 2020-07-04 by coolsnowwolf