coolsnowwolf/lede #3726: 新版的去广告大师好像有BUG

hxlls, 2020-03-11:
全新编译，并且已经全新安装，不保存之前的设置，在更新adblock 的规则时候按键一直在更新中。。
查看tmp/adnew.conf 文件已经生成，大概容量在
[image]

还有对应的adbyby.update文件，生成了一个20B左右的文件，里面只有日期
[image]
在luci 界面，还是一直显示更新中
看系统日志没有对应的日志，也无法判断是否已经运行

hxlls, 2020-03-11:
我看了下生成的adnew.conf 文件，生成大概是8.5W条数据，
[image]
不懂是否是已经超出了65535的原因？

shenyz, 2020-03-11:
easylist-downloads.adblockplus.org  域名走代理 再去更新

hxlls, 2020-03-11:
> easylist-downloads.adblockplus.org 域名走代理 再去更新

已经更新了啊，你看规则，都生成了8万多条，和域名没关系吧

coolsnowwolf, 2020-03-11:
rom 空间可能满了

hxlls, 2020-03-11:
> rom 空间可能满了

这个可能性不太大吧，因为这个是X86的，并且内存是4G

int2e, 2020-03-11:
> > rom 空间可能满了
> 
> 这个可能性不太大吧，因为这个是X86的，并且内存是4G

他说你磁盘空间可能满了

panther706, 2020-03-11:
/dev/root | /rom | 0.00 B / 17.00 MB | 100% (17.00 MB)
-- | -- | -- | --
tmpfs | /tmp | 56.01 MB / 61.12 MB | 8% (5.11 MB)
/dev/ubi0_1 | /overlay | 81.64 MB / 86.91 MB | 1% (816.00 KB)
overlayfs:/overlay | / | 81.64 MB / 86.91 MB | 1% (816.00 KB)
tmpfs | /dev | 512.00 KB / 512.00 KB | 0% (0.00 B)

确实ROM没有空间，昨天那源码可以更新的

jasonliu747, 2020-03-11:
刚编译最新代码，广告大师一直显示 Adbyby Plus+ 未运行，好像不止一个人遇到类似问题，请问该怎么解决？
[image]

coolsnowwolf, 2020-03-11:
不同问题请分开，这个关闭了

closed 2020-03-11 by coolsnowwolf