coolsnowwolf/lede #3733: Adbyby 无法正常运行

jasonliu747, 2020-03-11:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：刚编译最新代码，广告大师一直显示 Adbyby Plus+ 未运行，好像不止一个人遇到类似问题，请问该怎么解决？

### (2) 路由器型号和固件版本

A：FriendlyARM NanoPi NEO 2
OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7)

### (3) 详细日志

A：[image]

[image]

tcsr200722, 2020-03-12:
x86可以正常运行，就是adblock Plus 规则数据库更新不了

coolsnowwolf, 2020-03-12:
只支持 32bit

coolsnowwolf, 2020-03-12:
arm 64bit 不能运行

jasonliu747, 2020-03-12:
原来如此，那请问 @coolsnowwolf 要如何在 Neo2 上面正确运行广告大师呢？

coolsnowwolf, 2020-03-12:
编译 32bit 的 ARM 版本

jasonliu747, 2020-03-13:
不好意思再问个问题，我没看到对应 Neo2 的 32bit 版本，请问能说一下编译时该怎么选择吗？谢谢！ @coolsnowwolf

lesliedh, 2020-03-15:
编译friendlywrt的固件加入adbyby插件可以运行

jasonliu747, 2020-03-15:
@lesliedh 您也是Neo2吗？

lesliedh, 2020-03-15:
r1s-h5

lesliedh, 2020-03-15:
r1s-h5，应该和你的平台差不多

closed 2020-03-19 by jasonliu747