coolsnowwolf/lede #3739: 反馈：R20.3.11 SSR PLUS最新编译依旧存在凌晨1点当掉（未运行）的情况。

lovelyOK, 2020-03-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：SSR PLUS最新编译依旧存在凌晨1点当掉（未运行）的情况。自动更新、自动切换等设置未关闭

### (2) 路由器型号和固件版本

A：R20.3.11         X86路由器

### (3) 详细日志

A：

lyin888, 2020-03-12:
遇到同样问题提

swxk521, 2020-03-12:
遇到同样问题提

coolsnowwolf, 2020-03-12:
把订阅的连接私信我，否则无法解决，这个问题和机场有关

lesliedh, 2020-03-12:
似乎是电信公司的问题，公司固定公网ip没问题，个人pppoe拨号会有问题

coolsnowwolf, 2020-03-12:
把 ssr plus 的日志和系统日志，还有订阅地址私信下即可

lesliedh, 2020-03-12:
怎么私信，github不熟悉（40多岁跟不上节奏了），直接发email可以吗？

flin-X, 2020-03-13:
前几天同样的情况，试了下把自动更新订阅的时间换成了白天，这两天没出现过了。

closed 2020-03-13 by lovelyOK

Jefffish09, 2020-05-13:
最新版编译OpenWrt R20.5.9，ssr plus也是最新的。

默认使用凌晨2点自动更新订阅，也是订阅失败，导致ssr plus自己关掉，第二天看到显示“ssr plus未运行”。

但其他时间，我随机时间手动点“更新订阅”的时候，都没有失败过。

不知道是不是机场问题？不过如果本身有订阅服务器了，哪怕即使订阅失败，顶多就拿不到新的订阅信息，也可以继续使用旧的订阅信息继续运行，而不应该停掉吧。

我已经试着把自动订阅时间改成白天且通过代理进行订阅，看看还会不会出现同样问题。