coolsnowwolf/lede #3741: adbyby规则更新卡死导致路由器断网

swxk521, 2020-03-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
adbyby Adblock Plus 规则数据更新是如果科学上网没开，就会卡死路由器
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
adbyby Adblock Plus 规则数据更新是如果科学上网没开，就会卡死路由器
### (1) 具体问题

A：adbyby Adblock Plus 规则数据更新是如果科学上网没开，就会卡死路由器

### (2) 路由器型号和固件版本

A：X86-64   R20.3.11

### (3) 详细日志

A：

norikohxb, 2020-03-12:
我也遇到了。感觉最近添加的功能不是很合拍。，今天还发生了很多国内视频音乐无法打开。好像关闭adbyby再开启又好了，但是很不流畅。国内视频，小度在家有些视频都打开不了（开着adbyby）mibox访问youtube迟钝（很奇怪电话倒是不会）然后各种用户喊卡，我切换回3月9日的了

norikohxb, 2020-03-12:
之前还发生过，开启那个ssr的广告或者关闭的时候造成ssr功能失效，需要重启路由器。

norikohxb, 2020-03-12:
好像程序在启动了ssr的广告过滤和关闭以后并不是一个可逆的，就是说当你重新刷完rom这个时候是没有开启ssr的过滤，如果你启用了过滤，然后再关闭，你的整个系统其实和刚刷完rom没有开启过滤规则的时候一样。 adbyby也是。感觉有点串门的样子。所以造成国内的小度在家都不能播放视频。

norikohxb, 2020-03-12:
更新adbyby的blocklist的时候科学上网是必须开的，因为他是依赖翻墙的环境。不过我觉得这个最新版本很怪。好像一些按钮你不该去点击更新，点击更新以后，就发生了国内视频看不了，翻墙也不怎么好用。

gpeditmsc1986, 2020-03-12:
我使用正常，规则更新3网都正常，只是电信与联通更新都很快，移动比较慢。现在的版本不需要设置全局电视盒子也可以正常了

norikohxb, 2020-03-12:
我估计你正常是你没有手贱去点击，左边那个按钮，我是每个更新的按钮都点，点击完就不正常了。今天撤回了3月7日的版本，因为我是git pull最新版本，我现在在找朋友编译之前的。
我adybyb是必须要的，非常喜欢他，都是开全局模式。然后有按钮进入后台没事干就点击一次。这个规则更新本来就需要翻墙。只不过好像这个也开ssrplus的广告过滤也开，就好像不正常了，我目前也说不上为什么。

norikohxb, 2020-03-12:
还有你的adbyby规则是full全局模式吗？ 我这半年都是full 模式在跑的，没有开过plus。会不会这个也有区别？

Wilson-coder, 2020-03-12:
> 我估计你正常是你没有像我手贱去点击，左边那个按钮，我是每个更新的按钮都点，点击完就不正常了。今天撤回了3月7日的版本，因为我是git pull最新版本，我现在在找朋友编译之前的。
> 我adybyb是必须要的，非常喜欢他，都是开全局模式。然后有按钮进入后台没事干就点击一次。这个规则更新本来就需要翻墙。只不过好像这个也开ssrplus的广告过滤也开，就好像不正常了，我目前也说不上为什么。

我确认也有同样的情况。现在暂时是把ssr+那里的去广告取消打钩，然后把默认的那个.conf地址复制到Adbyby那个规则订阅的位置，然后在Adbyby那里选择更新。重新启动了路由器之后卡顿的现象没再发生。不过广告过滤效果没有看出来，电脑的IP地址已经添加到全局过滤列表了，然后用浏览器开优酷，看视频仍然需要等广告。也不知道是不是自己设置不对。

norikohxb, 2020-03-12:
我的adybby半年内都是开全局模式，因为我特别讨厌爱奇艺app的广告。也是运行很好。就是到了最后这个版本，感觉adbyby的广告屏蔽影响，和ssrplus的广告屏蔽影响都会造成要么爱奇艺这种视频看不了。youtube看不了，反正轮流的，我现在在重新编译，用默认方式不用精简方式，lean大说了，小白莫瞎搞，我是精简得很彻底的就4个插件。，我现在打算用直接勾选luci以后的试试看，不精简了，看看有没有问题

norikohxb, 2020-03-12:
好像不开全局模式的adbyby的人就没有遇到我的问题。

norikohxb, 2020-03-12:
我重新编译了一下，不用精简模式，好像没有问题了。之前我精简掉太多的东西了，估计lean大的编译越来越high调用的东西比较多。我现在运行了几个小时正常。但是ssr的广告屏蔽还没有开，晚一点开启看看

gpeditmsc1986, 2020-03-12:
我现在用的是Plus+模式，其他的现在都是默认设置，之前要用全局模式才有效，效果在盒子上面很好，ssr的广告屏蔽我从来不用，这次更新去广告的版本真的不错。

gpeditmsc1986, 2020-03-12:
[image]
[image]对了，默认的主题兼容性最好，贼快贼快

norikohxb, 2020-03-12:
我现在不精简的编译了一下，运行正常。但是不敢去点击那个“手动更新规则”因为那个点击了也没有任何内容出现。现在先用等lean大更新了更新的再来测试是不是点击这个“手动更新规则”导致出错的。

mjyhj, 2020-03-12:
我刚刚编译完更新固件，也卡死了。LUCI十分卡顿，DHCP没有主动分配IP。
关闭ADBYBY和SSR一样卡，然后去防火墙LUCI那重启了一下就OK了

norikohxb, 2020-03-13:
不死心，刚才又用了一下今天的版本，受不了各种无法访问，又换回3月7日的版本了。看来还是不要尝新

coolsnowwolf, 2020-03-13:
看看日志，如果 dnsmasq 起不来就会十分卡，因为系统会不断自动拉起

有两个原因：
1 规则有无效的语句
2 目录不存在

hcym, 2020-03-13:
--test试试看吧，我是很久不用了，就是hosts，简单粗暴执法

swxk521, 2020-03-13:
> 看看日志，如果 dnsmasq 起不来就会十分卡，因为系统会不断自动拉起
> 
> 有两个原因：
> 1 规则有无效的语句
> 2 目录不存在

试过把SSR科学上网关了就正常，点ADBYBY手动更新就不会卡死。

- [ ]

jackberry, 2020-03-13:
借楼问下腾讯视频的广告需要更新什么规则或者怎么设置才能去掉？

closed 2020-03-13 by swxk521