coolsnowwolf/lede #3742: Net无法回流

sunnyandsummy, 2020-03-12:
在外网用域名加端口可以正常访问家里的nas，但是在家里用域名加端口就访问不了 必须要用内网ip地址才能访问 ...

固件版本 | OpenWrt R20.3.8 (Compiled by Lusty on Mar.10, 2020) / LuCI Master (git-20.062.50551-a6189c5)
-- | --
内核版本 | 4.19.106

wspzf, 2020-03-12:
改路由器的host可以解决

sunnyandsummy, 2020-03-12:
是防火墙 - 自定义规则吗？  那我改如何修改呢？

coolsnowwolf, 2020-03-12:
编译时不加 docker 即可

sunnyandsummy, 2020-03-12:
我是用别人编译的固件，我也尝试过在现有的固件中移除docker但还没没用！

sunnyandsummy, 2020-03-12:
> 编译时不加docker即可
我是用别人编译的固件，我也尝试过在现有的固件中可移除docker但还没没用！

sunnyandsummy, 2020-03-12:
> 改路由器的host可以解决

是防火墙-自定义规则吗？那我改如何修改呢？

minfengyk, 2020-03-12:
我是3月8号编译的版本，没加docker，也有同样的问题，最近更新解决了吗？

wspzf, 2020-03-12:
在 /etc/hosts里加一行
nas的ip 你的域名，例如：
192.168.1.111 baidu.com

这是一种解决办法，把你的的域名劫持到nas的ip上

wspzf, 2020-03-12:
或者是在网络-DHCP/DNS里的自定义劫持域名填上也应该是可以的

sunnyandsummy, 2020-03-12:
> 在 /etc/hosts里加一行
> nas的ip 你的域名，例如：
> 192.168.1.111 baidu.com
> 
> 这是一种解决办法，把你的的域名劫持到nas的ip上

那如果我不止转发一个的话这种方法应该是不行吧  例如5000 5001 3389 这些我都要转发到内网来！

wspzf, 2020-03-12:
> > 在 /etc/hosts里加一行
> > nas的ip 你的域名，例如：
> > 192.168.1.111 baidu.com
> > 这是一种解决办法，把你的的域名劫持到nas的ip上
> 
> 那如果我不止转发一个的话这种方法应该是不行吧 例如5000 5001 3389 这些我都要转发到内网来！

这个是解决你家里不能用域名访问NAS的问题，因为之前你在家里用域名访问解析的是公网IP，加host后解析的是内网IP。和端口没有关系，保留你以前的配置

sunnyandsummy, 2020-03-12:
> > > 在 /etc/hosts里加一行
> > > nas的ip 你的域名，例如：
> > > 192.168.1.111 baidu.com
> > > 这是一种解决办法，把你的的域名劫持到nas的ip上
> > 
> > 
> > 那如果我不止转发一个的话这种方法应该是不行吧 例如5000 5001 3389 这些我都要转发到内网来！
> 
> 这个是解决你家里不能用域名访问NAS的问题，因为之前你在家里用域名访问解析的是公网IP，加host后解析的是内网IP。和端口没有关系，保留你以前的配置

但我在家里可以用我的域名来解析到我正确的公网ip，我在家里甚至可以直接用我的公网IP来登录我的Openwrt后台就是除了这个后台其他的都不能在家里用公网ip来上去

wspzf, 2020-03-12:
> > > > 在 /etc/hosts里加一行
> > > > nas的ip 你的域名，例如：
> > > > 192.168.1.111 baidu.com
> > > > 这是一种解决办法，把你的的域名劫持到nas的ip上
> > > 
> > > 
> > > 那如果我不止转发一个的话这种方法应该是不行吧 例如5000 5001 3389 这些我都要转发到内网来！
> > 
> > 
> > 这个是解决你家里不能用域名访问NAS的问题，因为之前你在家里用域名访问解析的是公网IP，加host后解析的是内网IP。和端口没有关系，保留你以前的配置
> 
> 但我在家里可以用我的域名来解析到我正确的公网ip，我在家里甚至可以直接用我的公网IP来登录我的Openwrt后台就是除了这个后台其他的都不能在家里用公网ip来上去

这种改host回导致你在家里不能用域名访问到OpenWrt后台，在外面没事😂

sunnyandsummy, 2020-03-12:
> 这种改host回导致你在家里不能用域名访问到OpenWrt后台，在外面没事😂

那这样还是个BUG啊   我以前用OpenWrt R9.10.24的版本用好久都完全没这样的问题

Wilson-coder, 2020-03-12:
是不是默认的端口转发页面底部那个“NAT回环”选项没有选中？我理解这个就是解决局域网内解析外网地址的。

sunnyandsummy, 2020-03-12:
> 是不是默认的端口转发页面底部那个“NAT回环”选项没有选中？我理解这个就是解决局域网内解析外网地址的。

这个已经是勾上了

ankygit, 2020-03-12:
同样问题，我还以为我的配置出问题了，但是之前是没问题的。

sunnyandsummy, 2020-03-12:
> 同样问题，我还以为我的配置出问题了，但是之前是没问题的。
我以前一直在用OpenWrt R9.10.24版本一直都没问题的，只是最近都在家里就打算重刷一下升级到最新版本就出现这样的问题了！

zesming, 2020-03-12:
turbo acc 中 flow offloading 关了就好了

Zack0617, 2020-03-12:
试试在ssr plus的访问控制中将nas ip设置为不代理

sunnyandsummy, 2020-03-12:
> turbo acc 中 flow offloading 关了就好了

尝试过但没有任何效果

sunnyandsummy, 2020-03-12:
> 试试在ssr plus的访问控制中将nas ip设置为不代理

我都将SSR Plus关闭都没用

Choinyu, 2020-03-12:
同样的问题，端口转发中的“NAT回环”完全不起作用，尝试将DDNS域名放到SSR+的白名单也没用。

coolsnowwolf, 2020-03-13:
你们用我 action 默认生成的固件就正常了

minfengyk, 2020-03-13:
今天编译新固件，这个版本已经解决了NAT环回。编完测试还是不行，无意中发现手机可以环回，电脑浏览器还是不行，最后清Cookie，就好了。有类似问题的朋友试试看。

trackycn, 2020-03-15:
> 今天编译新固件，这个版本已经解决了NAT环回。编完测试还是不行，无意中发现手机可以环回，电脑浏览器还是不行，最后清Cookie，就好了。有类似问题的朋友试试看。

昨天看到你说的, 测试了一下还是不行。

如果是4.19内核, 编译时不勾选docker, 默认是可以支持:
1. 外网到路由器自身的端口映射, 例如映射外网 8443端口到路由器443端口;
2. 外网到路由器局域网中其他机器同端口映射, 例如映射外网8888端口到局域网中 NAS 的 8888 端口

默认不支持:
1. 外网到路由器局域网中其他机器非相同端口映射, 例如映射外网8888端口到局域网中 NAS 的 80 端口

针对不支持的情况, 关闭 turbo acc 中的 flow offloading 就可以了。

另外, 按照大雕在 #3777 中的回复, 用 4.9 的内核应该也是可以的, 这个我没测试过。

minfengyk, 2020-03-16:
> 昨天看到你说的, 测试了一下还是不行。
>
> 如果是4.19内核, 编译时不勾选docker, 默认是可以支持:
>
> 1. 外网到路由器自身的端口映射, 例如映射外网 8443端口到路由器443端口;
> 2. 外网到路由器局域网中其他机器同端口映射, 例如映射外网8888端口到局域网中 NAS 的 8888 端口
>
> 默认不支持:
>
> 1. 外网到路由器局域网中其他机器非相同端口映射, [...]

我测试外网到路由器局域网中NAS非相同端口映射是可以的，并且这时flow offloading是打开的。就是UPnP内网中NAS非相同端口映射不行，只好手动设置端口映射。

ElectronicElephant, 2020-03-22:
类似的功能在梅林里叫 NAT Loopback

我比较喜欢直接设置 DHCP Host，并且内外网同一端口

ivybridgei, 2020-03-27:
N1的OpenWrt R20.3.19 / LuCI Master (git-20.062.50551-a6189c5)版本当主路由使用仍然有这个毛病

SupernovaCooper, 2020-04-08:
我也遇到了这个问题。OpenWRT路由器设置了端口转发，外网访问域名可以直接转发到内网那台电脑，但是内网通过域名访问就没法到那台电脑。尝试卸载docker了，依然无效。求大佬更新。

closed 2020-04-08 by coolsnowwolf

clock88731, 2022-02-08:
参见 https://github.com/coolsnowwolf/lede/issues/2103