coolsnowwolf/lede #3748: 星际宝盒	MobiPromo CM520-79F 编译失败，报错

ankygit, 2020-03-12:
选了机型之后其他全默认的情况下，编译到以下步骤就报错：
SHELL= flock /home/ubuntu/20200310/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ubuntu/20200310/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/strongswan-5.6.3/.pkgdir/strongswan-mod-xcbc/
. /home/ubuntu/20200310/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/ubuntu/20200310/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.strongswan-mod-xcbc_installed
echo "strongswan-mod-xcbc" >> /home/ubuntu/20200310/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/strongswan.default.install
make[3]: Leaving directory '/home/ubuntu/20200310/lede/feeds/packages/net/strongswan'
time: package/feeds/packages/strongswan/compile#298.59#35.45#282.88
make[2]: Leaving directory '/home/ubuntu/20200310/lede'
package/Makefile:107: recipe for target '/home/ubuntu/20200310/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/ubuntu/20200310/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/20200310/lede'
Build failed - please re-run with -j1 to see the real error message
/home/ubuntu/20200310/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1

请问该怎么修改config文件才可以编译出来呢？ 感激！

467815891a, 2020-03-15:
我今天才编译的星际宝盒，没问题。你这个问题我看不出来，或许需要`make clean && make V=99 -j1`重新编译一下才能看出来

ankygit, 2020-03-15:
> 我今天才编译的星际宝盒，没问题。你这个问题我看不出来，或许需要`make clean && make V=99 -j1`重新编译一下才能看出来

按你说的，重新git clone了一边15号的源码，再跑一次，还是一样的错误卡住了。
编译其他路由器或者x86的固件都没有问题 。

ankygit, 2020-03-15:
15号的源码，编译卡住的错误还是这里，一模一样。。。

package/Makefile:107: recipe for target '/home/ubuntu/20200310/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed

467815891a, 2020-03-18:
`rm .config && make clean && make V=99 -j1`重新配置吧，肯定是你的问题，但是凭这一两行看不出问题。

closed 2020-04-08 by coolsnowwolf