coolsnowwolf/lede #3752: 有谁知道系统日志里面这个错是哪里出问题了？（我自己的问题）

norikohxb, 2020-03-12:
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186275] pci 0000:00:15.3: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186337] pci 0000:00:15.3: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186399] pci 0000:00:15.4: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186457] pci 0000:00:15.4: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186519] pci 0000:00:15.5: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186577] pci 0000:00:15.5: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186638] pci 0000:00:15.6: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186696] pci 0000:00:15.6: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186758] pci 0000:00:15.7: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186816] pci 0000:00:15.7: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186878] pci 0000:00:16.3: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186936] pci 0000:00:16.3: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.186998] pci 0000:00:16.4: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187056] pci 0000:00:16.4: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187148] pci 0000:00:16.5: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187206] pci 0000:00:16.5: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187269] pci 0000:00:16.6: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187327] pci 0000:00:16.6: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187389] pci 0000:00:16.7: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187447] pci 0000:00:16.7: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187515] pci 0000:00:17.3: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187574] pci 0000:00:17.3: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187636] pci 0000:00:17.4: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187694] pci 0000:00:17.4: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187756] pci 0000:00:17.5: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187814] pci 0000:00:17.5: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187876] pci 0000:00:17.6: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187934] pci 0000:00:17.6: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.187996] pci 0000:00:17.7: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188054] pci 0000:00:17.7: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188116] pci 0000:00:18.2: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188175] pci 0000:00:18.2: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188236] pci 0000:00:18.3: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188295] pci 0000:00:18.3: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188357] pci 0000:00:18.4: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188415] pci 0000:00:18.4: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188477] pci 0000:00:18.5: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188535] pci 0000:00:18.5: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188597] pci 0000:00:18.6: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188655] pci 0000:00:18.6: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188717] pci 0000:00:18.7: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188775] pci 0000:00:18.7: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188840] pci 0000:00:18.7: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188898] pci 0000:00:18.7: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.188960] pci 0000:00:18.6: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189018] pci 0000:00:18.6: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189080] pci 0000:00:18.5: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189138] pci 0000:00:18.5: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189200] pci 0000:00:18.4: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189259] pci 0000:00:18.4: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189322] pci 0000:00:18.3: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189381] pci 0000:00:18.3: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189443] pci 0000:00:18.2: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189501] pci 0000:00:18.2: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189563] pci 0000:00:17.7: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189621] pci 0000:00:17.7: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189683] pci 0000:00:17.6: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189741] pci 0000:00:17.6: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189803] pci 0000:00:17.5: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189861] pci 0000:00:17.5: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189923] pci 0000:00:17.4: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.189982] pci 0000:00:17.4: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:17.3: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:17.3: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:16.7: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:16.7: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:16.6: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:16.6: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:16.5: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:16.5: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:16.4: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:16.4: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:16.3: BAR 7: no space for [io  size 0x1000]
[... cut ...]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:15.5: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:15.4: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:15.4: BAR 7: failed to assign [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:15.3: BAR 7: no space for [io  size 0x1000]
Thu Mar 12 20:49:18 2020 kern.info kernel: [    0.190030] pci 0000:00:15.3: BAR 7: failed to assign [io  size 0x1000]

norikohxb, 2020-03-12:
有没有路过的大神可以帮我看看，不胜感激。

norikohxb, 2020-03-13:
自己顶一下

norikohxb, 2020-03-13:
有没有大神可以帮我看看

norikohxb, 2020-03-14:
请路过的大神赐教

norikohxb, 2020-03-15:
请路过的大神帮我看看啊。

closed 2020-03-24 by norikohxb

klmyhd, 2020-10-22:
你好楼主，遇到相同问题，请问有什么解决办法吗？

lvusyy, 2021-03-03:
参考下这里
https://discuss.96boards.org/t/no-enough-memory-address-space-for-pcie-f4000000/9592

jackiezhang29, 2021-05-20:
> Thu Mar 12 20:49:18 2020 kern.info kernel: [ 0.186275] pci 0000:00:15.3: BAR 7: no space for [io size 0x1000]
> Thu Mar 12 20:49:18 2020 kern.info kernel: [ 0.186337] pci 0000:00:15.3: BAR 7: failed [...]

我现在也出现了这个问题，esxi7，请问后来是怎么解决的