coolsnowwolf/lede #3753: IP/MAC绑定功能经常性失效

Fyred, 2020-03-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
openwrt自带的IP/MAC功能经常无效，ip neigh命令查看并未绑定

### (1) 具体问题

A：IP/MAC中已设置了绑定，但ip neigh命令查看arp表的时候，发现条目状态为REACHABLE而并非PERMANENT，该主机关闭后，本条arp信息即失效，未形成永久绑定关系。
这会导致通过端口映射发送来的远程唤醒包无法抵达IP/MAC绑定的主机，主机无法通过外网端口映射唤醒。
查看/etc/init.d/arpbind，发现原因可能是33行：
`ip neigh add $1 lladdr $2 nud permanent dev $3`
执行失败导致（ip neigh add仅在没有条目时才能执行成功）
尝试将其改为：
`ip neigh add $1 lladdr $2 nud permanent dev $3 || ip neigh change $1 lladdr $2 nud permanent dev $3`
症状明显缓解，但仍偶有绑定无效发生

### (2) 路由器型号和固件版本

A：x86，OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7)

### (3) 详细日志

A：无日志，有图有真相。
[image]

Wilson-coder, 2020-03-12:
是的，这个问题从R9.12.31版本之后好像不记得哪个版本开始就一直存在了，反正我自从升级到2020之后的版本后，WOL远程开机功能就彻底无效了。

Fyred, 2020-03-12:
> 是的，这个问题从R9.12.31版本之后好像不记得哪个版本开始就一直存在了，反正我自从升级到2020之后的版本后，WOL远程开机功能就彻底无效了。

1、修改/etc/init.d/arpbind中第33行add为replace
2、手动指定电脑ip
3、ip在dhcp地址池之外
4、不要设置该ip为dhcp静态分配

保证以上4条，基本绑定就不会失效了，不知道arpbind受什么影响，感觉dhcp总跑在它前边，优先级比他高似的。

Wilson-coder, 2020-03-13:
> > 是的，这个问题从R9.12.31版本之后好像不记得哪个版本开始就一直存在了，反正我自从升级到2020之后的版本后，WOL远程开机功能就彻底无效了。
> 
> 1、修改/etc/init.d/arpbind中第33行add为replace
> 2、手动指定电脑ip
> 3、ip在dhcp地址池之外
> 4、不要设置该ip为dhcp静态分配
> 
> 保证以上4条，基本绑定就不会失效了，不知道arpbind受什么影响，感觉dhcp总跑在它前边，优先级比他高似的。

谢谢回复，完全按照你这四条建议改过了，仍然无效。

Fyred, 2020-03-13:
> > > 是的，这个问题从R9.12.31版本之后好像不记得哪个版本开始就一直存在了，反正我自从升级到2020之后的版本后，WOL远程开机功能就彻底无效了。
> > 
> > 
> > 1、修改/etc/init.d/arpbind中第33行add为replace
> > 2、手动指定电脑ip
> > 3、ip在dhcp地址池之外
> > 4、不要设置该ip为dhcp静态分配
> > 保证以上4条，基本绑定就不会失效了，不知道arpbind受什么影响，感觉dhcp总跑在它前边，优先级比他高似的。
> 
> 谢谢回复，完全按照你这四条建议改过了，仍然无效。

重启路由器。。。ip neigh查看你ip条目后边有permanent就是绑定成功了，广域网唤醒不好用，就是别的问题了

coolsnowwolf, 2020-03-13:
广域网唤醒有两个工具，你可以分别都试试

Wilson-coder, 2020-03-13:
> 广域网唤醒有两个工具，你可以分别都试试

谢谢您关注这个话题。我手机上和平板上安装的是两个不同作者、但都可以正常使用的App，包括设置什么都没有改过。现在都不行了，所以暂时不怀疑是工具自身问题。

不过早上突然想起来最早的时候也是广域唤醒不正常，一直在自己电脑和网卡设置上找原因。刚好在那个时间段因为无线网卡驱动的兼容性问题导致编译出错，就把内核人为切换成了4.9，结果广域唤醒无意间就正常了。

我刚才又把内核切换成4.9版本，重新编译了一个新固件，等下刷新了再看看效果。

coolsnowwolf, 2020-03-13:
好像高版本内核是会这样

leopardciaw, 2020-03-13:
> > 广域网唤醒有两个工具，你可以分别都试试
> 
> 谢谢您关注这个话题。我手机上和平板上安装的是两个不同作者、但都可以正常使用的App，包括设置什么都没有改过。现在都不行了，所以暂时不怀疑是工具自身问题。
> 
> 不过早上突然想起来最早的时候也是广域唤醒不正常，一直在自己电脑和网卡设置上找原因。刚好在那个时间段因为无线网卡驱动的兼容性问题导致编译出错，就把内核人为切换成了4.9，结果广域唤醒无意间就正常了。
> 
> 我刚才又把内核切换成4.9版本，重新编译了一个新固件，等下刷新了再看看效果。

结果怎么样哈。我也准备明天刷入路由中，远程唤醒，看到这个issue，担心4.14的内核是否无法唤醒。
另外请问下，如何更换内核，查了一些资料，看着要更改很多个地方。看到资料说4.9的内核，已不再支持一些CPU或机型。我的是acrh17。多谢！

Wilson-coder, 2020-03-14:
> 好像高版本内核是会这样

发现4.9内核一个新问题，就是好像不能准确识别我的赛扬3205u的CPU，系统频率几乎锁死在1.5GHz。后来搜索OpenWrt官方，好像适配的内核是4.14，昨晚又重新用4.14编译了一遍，早上更新，保留原来设置不动的前提下，已经可以实现广域唤醒了。

原来还遇到过刚关机可以唤醒，长时间关机后就失灵的情况，网上搜索也说是IPMAC绑定的问题，现在需要再观察一下这一点。

初步用下来感觉4.14内核下，各个功能呈现出来的情况和预期最接近。包括4.19下系统日志里面看到3322.org DDNS服务经常要重试很多次才能成功，改成4.9或者4.14之后，都是一次成功。当然这个因为次数有限，还不确认是不是必然结果。

总之非常感谢您的辛苦付出，才让我们这种编程小白有机会定制自己理想的路由器功能。

Wilson-coder, 2020-03-14:
> 结果怎么样哈。我也准备明天刷入路由中，远程唤醒，看到这个issue，担心4.14的内核是否无法唤醒。
> 另外请问下，如何更换内核，查了一些资料，看着要更改很多个地方。看到资料说4.9的内核，已不再支持一些CPU或机型。我的是acrh17。多谢！

前面解释了4.9遇到的问题，我现在已经改成4.14内核，暂时测试正常了，还需要做多一些观察。不过我的系统是X86软路由，发现的结论未必适合你的平台。

leopardciaw, 2020-03-14:
> 前面解释了4.9遇到的问题，我现在已经改成4.14内核，暂时测试正常了，还需要做多一些观察。不过我的系统是X86软路由，发现的结论未必适合你的平台。

谢谢回复。

tty228, 2020-03-14:
也有可能是 arp 命令的问题，我之前就发现 arp -d 是无效的，即使设备已经离线，根本清除不掉

Fyred, 2020-03-14:
> 发现4.9内核一个新问题，就是好像不能准确识别我的赛扬3205u的CPU，系统频率几乎锁死在1.5GHz。后来搜索OpenWrt官方，好像适配的内核是4.14，昨晚又重新用4.14编译了一遍，早上更新，保留原来设置不动的前提下，已经可以实现广域唤醒了。
>
> 原来还遇到过刚关机可以唤醒，长时间关机后就失灵的情况，网上搜索也说是IPMAC绑定的问题，现在需要再观察一下这一点。 [...]

我觉得你这样换内核应该解决不了问题，我去翻了翻我之前改的arpbind文件的保存时间是2018年11月，说明arp无法绑定的问题至少从2018年4.9内核时就存在了，刚关机可唤醒、长时间就失灵也是由于REACHABLE类型的动态绑定有老化时间，老化成FAILED失效有几分钟时间所致。
请你在4.14内核下再测试一段时间，排除偶然因素，特别是用ip neigh查看arp表的情况。

Wilson-coder, 2020-03-15:
> 我觉得你这样换内核应该解决不了问题，我去翻了翻我之前改的arpbind文件的保存时间是2018年11月，说明arp无法绑定的问题至少从2018年4.9内核时就存在了，**刚关机可唤醒、长时间就失灵也是由于REACHABLE类型的动态绑定有老化时间，老化成FAILED失效有几分钟时间所致**。
> 请你在4.14内核下再测试一段时间，排除偶然因素，特别是用ip neigh查看arp表的情况。

你好，目前4.14内核看似已经解决这个问题了，昨天刻意很早关机，过了一夜，已经超过12小时了，仍旧可以通过广域网唤醒。ip neigh查看arp表显示仍是REACHABLE，并不是PERMANENT.

Fyred, 2020-03-15:
个人认为换内核、换广域网唤醒工具来解决这个问题都是跑偏了。通过我1年多来的观察，我认为还是arpbind这个2015年的启动脚本不适应现在所致。
广域网唤醒工作原理是通过端口映射将魔术唤醒包从外网发送到内网对应主机，而端口映射只能映射ip，无法映射mac地址，被唤醒主机又是关机没有ip的，所以就需要路由器里有一条永久的arp静态绑定存在，直接告诉路由器这个被唤醒ip的mac地址是什么，魔术唤醒包就被发送到这个mac地址了。
openwrt的IP/MAC绑定是通过这个2015年的arpbind启动脚本实现的，开机的时候通过ip neigh add命令往arp表里新增一条静态绑定。但目前的openwrt，不知什么原因，在arpbind启动执行的时候，arp表里已经有对应的ip和mac的动态绑定条目了，ip neigh add是肯定执行不成功的，所以我才用ip neigh replace代替add，这样即便有动态绑定条目（REACHABLE）也能改写成静态绑定（PERMANENT），这样能解决路由器开机绑定不上的问题（占所有情况的99%）。
如果改这个arpbind启动脚本能解决100%的问题，我也不会提这个issue了，后来我发现，在路由器启动后，还会有什么程序改动arp表，将开机ip neigh replace写好的静态绑定又改成动态绑定，永久绑定又失效了，这个我确实找不出来原因，怀疑是dhcp静态分配什么的，也都是猜测。
导致广域网唤醒失效的原因很多，别人什么原因失效，我不敢妄下结论，但我能肯定的是IP/MAC绑定这个功能是一定有问题的，这也是这个issue想让L大看到并解决的初衷。

Fyred, 2020-03-15:
> 你好，目前4.14内核看似已经解决这个问题了，昨天刻意很早关机，过了一夜，已经超过12小时了，仍旧可以通过广域网唤醒。ip neigh查看arp表显示仍是REACHABLE，并不是PERMANENT.

REACHABLE条目是动态绑定，迟早失效，要么你是开机后用这台被唤醒的电脑查看的，所以是REACHABLE，要么是你那出现了什么高科技，电脑关机还在不停发arp广播包，维持了REACHABLE。
正常情况下，你现在看完是REACHABLE，关机10分钟，广域网唤醒一定失效。

Wilson-coder, 2020-03-15:
> > > > > 好像高版本内核是会这样
> > > > 
> > > > 
> > > > 发现4.9内核一个新问题，就是好像不能准确识别我的赛扬3205u的CPU，系统频率几乎锁死在1.5GHz。后来搜索OpenWrt官方，好像适配的内核是4.14，昨晚又重新用4.14编译了一遍，早上更新，保留原来设置不动的前提下，已经可以实现广域唤醒了。
> > > > 原来还遇到过刚关机可以唤醒，长时间关机后就失灵的情况，网上搜索也说是IPMAC绑定的问题，现在需要再观察一下这一点。
> > > > 初步用下来感觉4.14内核下，各个功能呈现出来的情况和预期最接近。包括4.19下系统日志里面看到3322.org DDNS服务经常要重试很多次才能成功，改成4.9或者4.14之后，都是一次成功。当然这个因为次数有限，还不确认是不是必然结果。
> > > > 总之非常感谢您的辛苦付出，才让我们这种编程小白有机会定制自己理想的路由器功能。
> > > 
> > > 
> > > 我觉得你这样换内核应该解决不了问题，我去翻了翻我之前改的arpbind文件的保存时间是2018年11月，说明arp无法绑定的问题至少从2018年4.9内核时就存在了，**刚关机可唤醒、长时间就失灵也是由于REACHABLE类型的动态绑定有老化时间，老化成FAILED失效有几分钟时间所致**。
> > > 请你在4.14内核下再测试一段时间，排除偶然因素，特别是用ip neigh查看arp表的情况。
> > 
> > 
> > 你好，目前4.14内核看似已经解决这个问题了，昨天刻意很早关机，过了一夜，已经超过12小时了，仍旧可以通过广域网唤醒。ip neigh查看arp表显示仍是REACHABLE，并不是PERMANENT.
> 
> REACHABLE条目是动态绑定，迟早失效，要么你是开机后用这台被唤醒的电脑查看的，所以是REACHABLE，要么是你那出现了什么高科技，电脑关机还在不停发arp广播包，维持了REACHABLE。
> 正常情况下，你现在看完是REACHABLE，关机10分钟，REACHABLE会老化成FAILED，广域网唤醒一定失效。

我对网络技术基本是小白，只能看懂你说的意思，但是对背后的工作原理几乎一无所知。不过我前面确实都是通过需要远程唤醒那台电脑ssh登录路由器来检测arp表的状态，那我再关机10分钟之后用另外的电脑看看啥情况吧。

不过从一个用户角度来说，我观察到的情况前面已经回复，确实解决了问题。瞎猜测一下，会不会是不同的内核下，处理arp绑定的工作机制不一样？

Fyred, 2020-03-15:
> > > > > > 好像高版本内核是会这样
> > > > > 
> > > > > 
> > > > > 发现4.9内核一个新问题，就是好像不能准确识别我的赛扬3205u的CPU，系统频率几乎锁死在1.5GHz。后来搜索OpenWrt官方，好像适配的内核是4.14，昨晚又重新用4.14编译了一遍，早上更新，保留原来设置不动的前提下，已经可以实现广域唤醒了。
> > > > > 原来还遇到过刚关机可以唤醒，长时间关机后就失灵的情况，网上搜索也说是IPMAC绑定的问题，现在需要再观察一下这一点。
> > > > > 初步用下来感觉4.14内核下，各个功能呈现出来的情况和预期最接近。包括4.19下系统日志里面看到3322.org DDNS服务经常要重试很多次才能成功，改成4.9或者4.14之后，都是一次成功。当然这个因为次数有限，还不确认是不是必然结果。
> > > > > 总之非常感谢您的辛苦付出，才让我们这种编程小白有机会定制自己理想的路由器功能。
> > > > 
> > > > 
> > > > 我觉得你这样换内核应该解决不了问题，我去翻了翻我之前改的arpbind文件的保存时间是2018年11月，说明arp无法绑定的问题至少从2018年4.9内核时就存在了，**刚关机可唤醒、长时间就失灵也是由于REACHABLE类型的动态绑定有老化时间，老化成FAILED失效有几分钟时间所致**。
> > > > 请你在4.14内核下再测试一段时间，排除偶然因素，特别是用ip neigh查看arp表的情况。
> > > 
> > > 
> > > 你好，目前4.14内核看似已经解决这个问题了，昨天刻意很早关机，过了一夜，已经超过12小时了，仍旧可以通过广域网唤醒。ip neigh查看arp表显示仍是REACHABLE，并不是PERMANENT.
> > 
> > 
> > REACHABLE条目是动态绑定，迟早失效，要么你是开机后用这台被唤醒的电脑查看的，所以是REACHABLE，要么是你那出现了什么高科技，电脑关机还在不停发arp广播包，维持了REACHABLE。
> > 正常情况下，你现在看完是REACHABLE，关机10分钟，REACHABLE会老化成FAILED，广域网唤醒一定失效。
> 
> 我对网络技术基本是小白，只能看懂你说的意思，但是对背后的工作原理几乎一无所知。不过我前面确实都是通过需要远程唤醒那台电脑ssh登录路由器来检测arp表的状态，那我再关机10分钟之后用另外的电脑看看啥情况吧。
> 
> 不过从一个用户角度来说，我观察到的情况前面已经回复，确实解决了问题。瞎猜测一下，会不会是不同的内核下，处理arp绑定的工作机制不一样？
> 
> 补充：关机差不多二十分钟，从手机登录路由器查看arp表，对应IP显示是PERMANENT状态。
> [image]

如果一直是PERMANENT，那肯定没问题了，确实搞不懂为何换内核能解决问题，arpbind自身和内核有关的，也就是Hotplug事件的触发了吧，或许4.14内核下，对arpbind有干扰的东西被干掉了？对了，你的电脑是直连在openwrt路由器的网口上，还是通过交换机转接的呢？

Wilson-coder, 2020-03-15:
> 如果一直是PERMANENT，那肯定没问题了，确实搞不懂为何换内核能解决问题，arpbind自身和内核有关的，也就是Hotplug事件的触发了吧，或许4.14内核下，对arpbind有干扰的东西被干掉了？对了，你的电脑是直连在openwrt路由器的网口上，还是通过交换机转接的呢？

电脑是直连路由器网口的，我等下可以再测试一下通过交换机（无线AP多余LAN口）连接的情况。

Fyred, 2020-03-15:
> 电脑是直连路由器网口的，我等下可以再测试一下通过交换机（无线AP多余LAN口）连接的情况。

直连openwrt网口的话，电脑关机，网卡降速，应该会触发一下路由的Hotplug事件，会重新执行一次arpbind绑定脚本，应该比通过交换机连要可靠吧，内核日志里会有ifup的记录。

Wilson-coder, 2020-03-15:
> > > > > > > > > 好像高版本内核是会这样
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 发现4.9内核一个新问题，就是好像不能准确识别我的赛扬3205u的CPU，系统频率几乎锁死在1.5GHz。后来搜索OpenWrt官方，好像适配的内核是4.14，昨晚又重新用4.14编译了一遍，早上更新，保留原来设置不动的前提下，已经可以实现广域唤醒了。
> > > > > > > > 原来还遇到过刚关机可以唤醒，长时间关机后就失灵的情况，网上搜索也说是IPMAC绑定的问题，现在需要再观察一下这一点。
> > > > > > > > 初步用下来感觉4.14内核下，各个功能呈现出来的情况和预期最接近。包括4.19下系统日志里面看到3322.org DDNS服务经常要重试很多次才能成功，改成4.9或者4.14之后，都是一次成功。当然这个因为次数有限，还不确认是不是必然结果。
> > > > > > > > 总之非常感谢您的辛苦付出，才让我们这种编程小白有机会定制自己理想的路由器功能。
> > > > > > > 
> > > > > > > 
> > > > > > > 我觉得你这样换内核应该解决不了问题，我去翻了翻我之前改的arpbind文件的保存时间是2018年11月，说明arp无法绑定的问题至少从2018年4.9内核时就存在了，**刚关机可唤醒、长时间就失灵也是由于REACHABLE类型的动态绑定有老化时间，老化成FAILED失效有几分钟时间所致**。
> > > > > > > 请你在4.14内核下再测试一段时间，排除偶然因素，特别是用ip neigh查看arp表的情况。
> > > > > > 
> > > > > > 
> > > > > > 你好，目前4.14内核看似已经解决这个问题了，昨天刻意很早关机，过了一夜，已经超过12小时了，仍旧可以通过广域网唤醒。ip neigh查看arp表显示仍是REACHABLE，并不是PERMANENT.
> > > > > 
> > > > > 
> > > > > REACHABLE条目是动态绑定，迟早失效，要么你是开机后用这台被唤醒的电脑查看的，所以是REACHABLE，要么是你那出现了什么高科技，电脑关机还在不停发arp广播包，维持了REACHABLE。
> > > > > 正常情况下，你现在看完是REACHABLE，关机10分钟，REACHABLE会老化成FAILED，广域网唤醒一定失效。
> > > > 
> > > > 
> > > > 我对网络技术基本是小白，只能看懂你说的意思，但是对背后的工作原理几乎一无所知。不过我前面确实都是通过需要远程唤醒那台电脑ssh登录路由器来检测arp表的状态，那我再关机10分钟之后用另外的电脑看看啥情况吧。
> > > > 不过从一个用户角度来说，我观察到的情况前面已经回复，确实解决了问题。瞎猜测一下，会不会是不同的内核下，处理arp绑定的工作机制不一样？
> > > > 补充：关机差不多二十分钟，从手机登录路由器查看arp表，对应IP显示是PERMANENT状态。
> > > > [image]
> > > 
> > > 
> > > 如果一直是PERMANENT，那肯定没问题了，确实搞不懂为何换内核能解决问题，arpbind自身和内核有关的，也就是Hotplug事件的触发了吧，或许4.14内核下，对arpbind有干扰的东西被干掉了？对了，你的电脑是直连在openwrt路由器的网口上，还是通过交换机转接的呢？
> > 
> > 
> > 电脑是直连路由器网口的，我等下可以再测试一下通过交换机（无线AP多余LAN口）连接的情况。
> 
> 直连openwrt网口的话，电脑关机和开机，网卡变速，应该都会触发路由的Hotplug事件，会重新执行两次arpbind绑定脚本，应该比通过交换机连要可靠吧，内核日志里会有ifup的记录。

把电脑网线从路由器网口移到无线AP下的端口，测试了两次，第一次是电脑已经正常关机后直接拔掉网线，并接入AP自带LAN口，然后用WOL软件远程开机，结果正常。第二次是保持目前的连接状态，关机等待一个小时，再用WOL，也能正常开机。

closed 2020-03-25 by coolsnowwolf

windybird, 2022-09-14:
不要设置该ip为dhcp静态分配,经测试这样是可以的。。