coolsnowwolf/lede #3755: R619AC 编译出错

xxdd239, 2020-03-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
R619AC 编译出错
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
编译出错
### (1) 具体问题
9号还能编译成功，11号git pull后就不行了
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
tools/v8_gypfiles/v8_base_without_compiler.host.mk:26: warning: ignoring old recipe for target `/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Runtime.h'
tools/v8_gypfiles/v8_base_without_compiler.target.mk:26: warning: overriding recipe for target `/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Schema.h'
tools/v8_gypfiles/v8_base_without_compiler.host.mk:26: warning: ignoring old recipe for target `/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Schema.h'
  touch /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/tools/v8_gypfiles/v8_version.stamp
  touch /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/tools/v8_gypfiles/v8_headers.stamp
  g++ -m32 -o /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o ../deps/v8/src/base/bits.cc '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-DV8_TARGET_ARCH_ARM' '-DCAN_USE_ARMV7_INSTRUCTIONS' '-DCAN_USE_VFP3_INSTRUCTIONS' '-DCAN_USE_VFP32DREGS' '-DCAN_USE_NEON' '-DV8_EMBEDDER_STRING="-node.16"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DV8_USE_SNAPSHOT' '-DV8_CONCURRENT_MARKING' '-DV8_EMBEDDED_BUILTINS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DBUILDING_V8_BASE_SHARED' '-DUSE_EABI_HARDFLOAT=1' -I/home/dong/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I../deps/v8 -I../deps/v8/include  -pthread -Wno-unused-parameter -Wno-return-type -fno-strict-aliasing -m32 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/.deps//home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o.d.raw   -c
In file included from /usr/include/features.h:399:0,
                 from /usr/include/stdint.h:25,
                 from /usr/lib/gcc/x86_64-redhat-linux/4.8.5/include/stdint.h:9,
                 from ../deps/v8/src/base/bits.h:8,
                 from ../deps/v8/src/base/bits.cc:5:
/usr/include/gnu/stubs.h:7:27: fatal error: gnu/stubs-32.h: No such file or directory
 # include <gnu/stubs-32.h>
                           ^
compilation terminated.
make[5]: *** [/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o] Error 1
make[5]: Leaving directory `/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out'
make[4]: *** [node] Error 2
make[4]: Leaving directory `/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0'
make[3]: *** [/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/.built] Error 2
make[3]: Leaving directory `/home/dong/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#5.14#3.47#26.64
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory `/home/dong/lede'
make[1]: *** [/home/dong/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2

A：

ankygit, 2020-03-13:
我的星际宝盒，同cpu，同问题。

georip, 2020-03-13:
建议楼主搭上梯子，清空重来，肯定能过~

jdjingdian, 2020-03-13:
`
Applying /home/jarvis/rw619/lede/target/linux/generic/pending-4.19/411-mtd-partial_eraseblock_write.patch using plaintext: 
patching file drivers/mtd/mtdpart.c
Hunk #4 FAILED at 606.
1 out of 4 hunks FAILED -- saving rejects to file drivers/mtd/mtdpart.c.rej
Patch failed!  Please fix /home/jarvis/rw619/lede/target/linux/generic/pending-4.19/411-mtd-partial_eraseblock_write.patch!
Makefile:26: recipe for target '/home/jarvis/rw619/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.108/.prepared_af770c967ae9f9a5157ff88e42db0899' failed
make[4]: *** [/home/jarvis/rw619/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.108/.prepared_af770c967ae9f9a5157ff88e42db0899] Error 1
make[4]: Leaving directory '/home/jarvis/rw619/lede/target/linux/ipq40xx'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/jarvis/rw619/lede/target/linux'
time: target/linux/compile#7.93#3.82#9.11
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/home/jarvis/rw619/lede'
target/Makefile:18: recipe for target '/home/jarvis/rw619/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile' failed
make[1]: *** [/home/jarvis/rw619/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/jarvis/rw619/lede'
/home/jarvis/rw619/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2`

我也是，是不是因为更新了内核所以出来问题？

closed 2020-03-13 by coolsnowwolf