coolsnowwolf/lede #3756: ShadowSocksR Plus+添加服务器节点时不能通过链接导入配置信息

minfengyk, 2020-03-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
    ShadowSocksR Plus+人工添加Ttojan服务器节点时不能通过链接（不是通过订阅链接）导入配置信息
### (1) 具体问题
    添加如：trojan://Kgy7TeRaJpw59Kp5@au1-1.sstr-api.xyz:443这样链接，（密码我改动了一下，抱歉）。以前版本可以自动导入相关信息，测试passwall可以，说明不是节点问题。

### (2) 路由器型号和固件版本
型号：Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz : 1 Core 2 Thread (CpuMark : 37214.783573 Scores)
版本：OpenWrt R20.3.5 / LuCI Master (git-20.062.50551-a6189c5)

### (3) 详细日志
导入失败后系统日志上和ShadowSocksR Plus+客户端上没有记录到报错信息。
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain googlepages.com
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain googleusercontent.com
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain gstatic.com
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain google.com.hk
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain google.com
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain wsj.net
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain whatsapp.net
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain whatsapp.com
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain uploaded.net
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain t66y.com
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain trouter.io
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using nameserver 127.0.0.1#5335 for domain textnow.me
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: using 5234 more nameservers
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: read /etc/hosts - 4 addresses
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq[9727]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Mar 12 21:46:50 2020 daemon.info dnsmasq-dhcp[9727]: read /etc/ethers - 0 addresses
Thu Mar 12 21:48:02 2020 daemon.warn dnsmasq[9727]: possible DNS-rebind attack detected: internal-lb-all.bytedns.net

ShadowSocksR Plus+上的日志

2020-03-12 22:09:32 ShadowsocksR No Problem.
2020-03-12 22:09:29 Main server is NOT avilable.Continue using current server.
2020-03-12 22:09:17 Current server is not default Main server, try to switch back.
2020-03-12 21:58:10 ShadowsocksR No Problem.
2020-03-12 21:58:09 Main server is NOT avilable.Continue using current server.
2020-03-12 21:57:57 Current server is not default Main server, try to switch back.
2020-03-12 21:46:50 ShadowsocksR server switch OK
2020-03-12 21:46:47 Welcome to trojan 1.14.1 , 2 Threads Started!
2020-03-12 21:46:44 Another server is avilable, now switching server.
2020-03-12 21:46:23 Current server error, try to switch another server.
2020-03-12 21:35:01 ShadowsocksR No Problem.
2020-03-12 21:23:43 ShadowsocksR No Problem.
2020-03-12 21:12:36 ShadowsocksR No Problem.
2020-03-12 21:01:28 ShadowsocksR No Problem.
2020-03-12 20:50:18 ShadowsocksR No Problem.
2020-03-12 20:39:08 ShadowsocksR No Problem.
2020-03-12 20:28:00 ShadowsocksR No Problem.
2020-03-12 20:16:52 ShadowsocksR No Problem.
2020-03-12 20:05:43 ShadowsocksR No Problem.
2020-03-12 19:54:28 ShadowsocksR No Problem.
2020-03-12 19:43:20 ShadowsocksR No Problem.
2020-03-12 19:32:06 ShadowsocksR No Problem.
2020-03-12 19:20:55 ShadowsocksR No Problem.
2020-03-12 19:09:47 ShadowsocksR No Problem.
2020-03-12 18:58:37 ShadowsocksR No Problem.
2020-03-12 18:47:30 ShadowsocksR No Problem.
2020-03-12 18:36:16 ShadowsocksR No Problem.
2020-03-12 18:25:08 ShadowsocksR No Problem.
2020-03-12 18:13:56 ShadowsocksR No Problem.
2020-03-12 18:02:48 ShadowsocksR No Problem.
2020-03-12 17:51:34 ShadowsocksR No Problem.
2020-03-12 17:40:18 ShadowsocksR No Problem.
[... 102 lines cut ...]
2020-03-12 05:21:07 ShadowsocksR No Problem.
2020-03-12 05:09:59 Try restart current server.
2020-03-12 05:09:55 Welcome to trojan 1.14.1 , 2 Threads Started!
2020-03-12 05:09:52 Current server error, try to switch another server.
2020-03-12 04:58:42 ShadowsocksR No Problem.
2020-03-12 04:47:34 ShadowsocksR No Problem.
2020-03-12 04:36:27 ShadowsocksR No Problem.
2020-03-12 04:25:20 ShadowsocksR No Problem.
2020-03-12 04:14:13 ShadowsocksR No Problem.
2020-03-12 04:03:06 ShadowsocksR No Problem.
2020-03-12 03:51:58 ShadowsocksR No Problem.
2020-03-12 03:40:51 ShadowsocksR No Problem.
2020-03-12 03:29:44 ShadowsocksR No Problem.
2020-03-12 03:18:37 ShadowsocksR No Problem.
2020-03-12 03:07:30 ShadowsocksR No Problem.
2020-03-12 02:56:22 ShadowsocksR No Problem.
2020-03-12 02:45:15 ShadowsocksR No Problem.
2020-03-12 02:34:08 ShadowsocksR No Problem.
2020-03-12 02:23:01 ShadowsocksR No Problem.
2020-03-12 02:11:54 ShadowsocksR No Problem.
2020-03-12 02:00:46 ShadowsocksR No Problem.
2020-03-12 01:49:38 ShadowsocksR No Problem.
2020-03-12 01:38:31 ShadowsocksR No Problem.
2020-03-12 01:27:24 ShadowsocksR No Problem.
2020-03-12 01:16:16 Welcome to trojan 1.14.1 , 2 Threads Started!

coolsnowwolf, 2020-03-12:
@SharerMax

coolsnowwolf, 2020-03-12:
@jjm2473

SharerMax, 2020-03-12:
我明天看看

jjm2473, 2020-03-12:
> @jjm2473

我之前添加的是支持trojan的订阅，并没有更改手动添加链接那里的js代码。

出现bug的js代码是：
package/lean/luci-app-ssr-plus/luasrc/view/shadowsocksr/ssrurl.htm 224行

coolsnowwolf, 2020-03-12:
要不就这么简单粗暴的算了，官方的 URL 定义写的实在太简单：

https://github.com/trojan-gfw/trojan-url

[image]

[image]

SharerMax, 2020-03-12:
> 要不就这么简单粗暴的算了，官方的 URL 定义写的实在太简单：
> 
> https://github.com/trojan-gfw/trojan-url
> 
> [image]
> 
> [image]

恩，本来是简单的，但是后面加了订阅，我就按订阅的格式扩展了一下

可尝试将224行的大于0改为大于1，应该就修复了

之前测试的时候可能是在路由器上直接改的，忘记同步了，我的锅

SharerMax, 2020-03-12:
https://github.com/coolsnowwolf/lede/blob/3aceea25a524860e83b9fe5012c46395ea152453/package/lean/luci-app-ssr-plus/luasrc/view/shadowsocksr/ssrurl.htm#L224

就是这了问题了，改为 大于 1 即可

PR #3767 (2020-03-13): luci-app-ssr-plus: fix bug that user can't import trojan url
close #3756
closed 2020-03-13 by coolsnowwolf