coolsnowwolf/lede #3757: 新版 subscribe.lua 无法订阅节点，清空重试问题依旧

tty228, 2020-03-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：手动运行 `/usr/bin/lua /usr/share/shadowsocksr/subscribe.lua` 报错如下

> /usr/bin/lua: /usr/share/shadowsocksr/subscribe.lua:23: attempt to call method 'get_first' (a nil value)
> stack traceback:
>         /usr/share/shadowsocksr/subscribe.lua:23: in main chunk
>         [C]: ?

依照之前的脚本，将脚本第 22 行 `local ucic = uci.cursor()` 修改为 `local ucic = luci.model.uci.cursor()` 之后可以正常运行

改动编号为 https://github.com/coolsnowwolf/lede/pull/3665

https://github.com/coolsnowwolf/lede/commit/d5ddb7e8a52e46b2709916bb9c62cfb68497c2df#diff-8bc30179814d8a098b7bbd588ca7e359

不确定是不是个别问题，因为我使用了 19.07 的luci

### (2) 路由器型号和固件版本

A：斐讯K3
luci 19.07

closed 2020-03-12 by tty228

jw113, 2020-03-15:
19.07也是一样的情况，手动修改后也可以用了

garypang13, 2020-03-23:
遇到了相同的情况.

fanck0605, 2020-04-05:
```
	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:23: in main chunk
stack traceback:
/usr/bin/lua: /usr/share/shadowsocksr/subscribe.lua:23: attempt to call method 'get_first' (a nil value)
```
问题仍然存在，官方openwrt-19.07手动添加了lean的源