coolsnowwolf/lede #3759: 编译uefi启动固件，启动后提示不能在（sda3）Block上找到有效的f2fs filesystem

minfengyk, 2020-03-12:
L大：
    您好！
    这个问题困扰我很久，百度，Google，泡论坛都找不到答案，只好向您请教！请您百忙中指点一下。
    初学编译，成功编译出心仪的固件，编译出的combined固件完美运行，但是同时编译出的uefi固件启动后总是报错：F2FS-fs (sda3)：Can't find valid F2FS filesystem in superblock。确定编译时在Kernel modules的filesystem中选了f2fs相关选项。这样的固件也能启动运行，但是wan口设置后不能拨号，需要删除后重新设置，并且重启才能解决，再次重启有可能不能拨号。请问怎么解决，谢谢🙏
    下面是启动后日志，内核日志和这一样：
[image]

mjyhj, 2020-03-12:
这个分区有问题吧，应该是sda2 是/rom 160m  没有sda2 sda3
重新写盘试试吧

dolphinpaopao, 2020-03-13:
试试直接重写不用升级的方式，刷过不同版本的固件？

minfengyk, 2020-03-13:
> 这个分区有问题吧，应该是sda2 是/rom 160m 没有sda2 sda3
> 重新写盘试试吧

分区是在编译完自动生成的，没有问题的，combined固件和uefi固件文件系统不一样，所以combined固件是两个分区，uefi固件4个分区。我怀疑发生这种原因是编译时menuconfig某个选项没正确设置。

> 试试直接重写不用升级的方式，刷过不同版本的固件？

我是装在ESXI虚拟机上，每次编译完固件，都是全新安装；写盘安装，转盘安装都试过，甚至用Ubuntu14，Ubuntu18，反复编译过不同版本。一样的menuconfig设置，同时输出的combined固件和uefi固件，combined固件正常，uefi固件出现如前所述错误。所以应该和安装方式没关系。

linyuhc001, 2020-03-13:
automount不要选

minfengyk, 2020-03-13:
> automount不要选
> 另外这个问题应该不影响wan口拨号。。。

感谢指点，再编译一次固件试试。

minfengyk, 2020-03-13:
> automount不要选
> 另外这个问题应该不影响wan口拨号。。。

问题解决了！高手！！

minfengyk, 2020-03-16:
> automount不要选
> 另外这个问题应该不影响wan口拨号。。。

最新发现：这不算一个错误，闲来无事用balkid命令查了一下，sda2是efi system，sda3是BIOS启动区。就推测uefi固件是不是也支持BIOS启动，在ESXI上测试了下，果然这固件支持双启动。
[image]

closed 2020-04-08 by coolsnowwolf