coolsnowwolf/lede #3771: 411-mtd-partial_eraseblock_write.patch 更新，引起编译失败

keivenliao, 2020-03-13:
看了一下
lede/target/linux/generic/pending-4.19/411-mtd-partial_eraseblock_write.patch

改动了不少，此次改动之前编译都没有问题。

Applying /home/keiven/lede/target/linux/generic/pending-4.19/411-mtd-partial_eraseblock_write.patch using plaintext: 
patching file drivers/mtd/mtdpart.c
Hunk #4 FAILED at 606.
1 out of 4 hunks FAILED -- saving rejects to file drivers/mtd/mtdpart.c.rej
Patch failed!  Please fix /home/keiven/lede/target/linux/generic/pending-4.19/411-mtd-partial_eraseblock_write.patch!
make[4]: *** [Makefile:37: /home/keiven/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.108/.prepared_cbb27ad7eb5c346ea4e0e4995bb87cb6] Error 1
make[4]: Leaving directory '/home/keiven/lede/target/linux/x86'
make[3]: *** [Makefile:13: compile] Error 2
make[3]: Leaving directory '/home/keiven/lede/target/linux'
time: target/linux/compile#8.50#6.64#15.59
make[2]: *** [target/Makefile:25: target/linux/compile] Error 2
make[2]: Leaving directory '/home/keiven/lede'
make[1]: *** [target/Makefile:18: /home/keiven/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/keiven/lede'
make: *** [/home/keiven/lede/include/toplevel.mk:227：world] 错误 2

SharerMax, 2020-03-13:
#3772 同样的问题
@coolsnowwolf 可能还需要更新 patch 文件
https://github.com/openwrt/openwrt/blob/5cc0493c2bac3089de3bfb52256748c6a9249254/target/linux/generic/pending-4.19/411-mtd-partial_eraseblock_write.patch

LeeMenHin, 2020-03-13:
是的要更新patch文件

closed 2020-03-13 by coolsnowwolf