coolsnowwolf/lede #3779: 4.14.172内核，竞斗云报错

ksong008, 2020-03-13:
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

 2. 详细叙述
今天git pull后，竞斗云编译选择4.14内核，出错，提示缺少sha1-arm.ko文件

日志如下：

ERROR: module '/home/shaka/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.14.172/arch/arm/crypto/sha1-arm.ko' is missing.
modules/crypto.mk:734: recipe for target '/home/shaka/lede/bin/targets/ipq40xx/generic/packages/kmod-crypto-sha1_4.14.172-1_arm_cortex-a7_neon-vfpv4.ipk' failed
make[3]: *** [/home/shaka/lede/bin/targets/ipq40xx/generic/packages/kmod-crypto-sha1_4.14.172-1_arm_cortex-a7_neon-vfpv4.ipk] Error 1
make[3]: Leaving directory '/home/shaka/lede/package/kernel/linux'
time: package/kernel/linux/compile#5.41#1.09#5.88
package/Makefile:111: recipe for target 'package/kernel/linux/compile' failed
make[2]: *** [package/kernel/linux/compile] Error 2
make[2]: Leaving directory '/home/shaka/lede'
package/Makefile:107: recipe for target '/home/shaka/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/shaka/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2

coolsnowwolf, 2020-03-13:
官方今天已经中止 IPQ 4.14 内核的支持了，请跟进

ksong008, 2020-03-13:
> 官方今天已经中止 IPQ 4.14 内核的支持了，请跟进

明白，谢谢

closed 2020-03-13 by ksong008

lexcomeing, 2022-10-31:
> 官方今天已经中止 IPQ 4.14 内核的支持了，请跟进

怎么解决呢？

PR #3786 (2020-03-13): ipq40xx: fix build error on 4.14
fix #3779