coolsnowwolf/lede #3785: 更新后n270编译时报错

kenen, 2020-03-13:
### (1) 具体问题

A：编译最新的固件，提示报错。求知道如何解决的大佬解惑，小弟不胜感激。。。

### (2) 路由器型号和固件版本

A：n270的软路由 ，最新的固件

### (3) 详细日志

A：Generating files
configure: creating ./config.status
creating main/internal_functions.c
creating main/internal_functions_cli.c
+--------------------------------------------------------------------+
| License:                                                           |
| This software is subject to the PHP License, available in this     |
| distribution in the file LICENSE.  By continuing this installation |
| process, you are bound by the terms of this license agreement.     |
| If you do not agree with the terms of this license, you must abort |
| the installation process at this point.                            |
+--------------------------------------------------------------------+

Thank you for using PHP.

config.status: creating php7.spec
config.status: creating main/build-defs.h
config.status: creating scripts/phpize
config.status: creating scripts/man1/phpize.1
config.status: creating scripts/php-config
config.status: creating scripts/man1/php-config.1
config.status: creating sapi/cli/php.1
config.status: creating sapi/fpm/php-fpm.conf
config.status: creating sapi/fpm/www.conf
config.status: creating sapi/fpm/init.d.php-fpm
config.status: creating sapi/fpm/php-fpm.service
config.status: creating sapi/fpm/php-fpm.8
config.status: creating sapi/fpm/status.html
config.status: creating sapi/cgi/php-cgi.1
config.status: creating main/php_config.h
config.status: executing libtool commands
config.status: executing default commands
configure: WARNING: unrecognized options: --enable-gd-native-ttf
touch /home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/.configured_ff94e781a6dcfebaebad9220fe39fa2f
rm -f /home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/.built
touch /home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9:php-7.2.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libiconv-stub/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libintl-stub/include  -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/usr/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/include/fortify -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libiconv-stub/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libintl-stub/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9:php-7.2.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libiconv-stub/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libintl-stub/include  -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/usr/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/include/fortify -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libiconv-stub/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libintl-stub/include " LDFLAGS="-L/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib -L/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/lib -L/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/usr/lib -L/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/lib -znow -zrelro -L/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libiconv-stub/lib -L/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libintl-stub/lib " make  -C /home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/. AR="i486-openwrt-linux-musl-gcc-ar" AS="i486-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9:php-7.2.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libiconv-stub/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libintl-stub/include" LD=i486-openwrt-linux-musl-ld NM="i486-openwrt-linux-musl-gcc-nm" CC="i486-openwrt-linux-musl-gcc" GCC="i486-openwrt-linux-musl-gcc" CXX="i486-openwrt-linux-musl-g++" RANLIB="i486-openwrt-linux-musl-gcc-ranlib" STRIP=i486-openwrt-linux-musl-strip OBJCOPY=i486-openwrt-linux-musl-objcopy OBJDUMP=i486-openwrt-linux-musl-objdump SIZE=i486-openwrt-linux-musl-size CROSS="i486-openwrt-linux-musl-" ARCH="i386" ;
make[4]: Entering directory '/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9'
/bin/bash /home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/libtool --silent --preserve-dup-deps --mode=compile i486-openwrt-linux-musl-gcc  -Iext/curl/ -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/ext/curl/ -DPHP_ATOM_INC -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/include -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/main -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9 -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/ext/date/lib -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/include/freetype2 -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/include/libpng16 -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libiconv-stub/include -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/ext/mbstring/oniguruma -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/ext/mbstring/libmbfl -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/ext/mbstring/libmbfl/mbfl -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/ext/zip/lib -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/TSRM -I/home/kennen/Openwrt/Lean_R9/lede/build_dir/target-i386_core2_musl/php-7.2.9/Zend  -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/usr/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/include/fortify -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libiconv-stub/include -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/usr/lib/libintl-stub/include  -I/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_co [... cut ...]
make[2]: Leaving directory '/home/kennen/Openwrt/Lean_R9/lede'
package/Makefile:107: recipe for target '/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/stamp/.package_compile' failed
make[1]: *** [/home/kennen/Openwrt/Lean_R9/lede/staging_dir/target-i386_core2_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kennen/Openwrt/Lean_R9/lede'
/home/kennen/Openwrt/Lean_R9/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

haxc, 2020-03-13:
i486是什么鬼？编译ｘ８６_64的就可以啦

hcym, 2020-03-13:
刚刚x86的正常编译

kenen, 2020-03-13:
> 刚刚x86的正常编译
刚才编译了一下还是同样的报错

closed 2020-04-08 by coolsnowwolf

gitfxt, 2020-05-01:
Got the same Error, how to fix it?