coolsnowwolf/lede #3787: 编译工具链时报错

cjkzwe, 2020-03-13:
+馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
编译交叉编译工具时编译报错
### (1) 具体问题
在Centos 9.9 Linux izx112aj3x4ytpz 3.10.0-1062.12.1.el7.x86_64 x86_64 GNU/Linux 上编译 ,
使用make menuconfig只选择 [*] Package the OpenWrt-based Toolchain 
编译命令: make V=99
环境:/root/lede-master
user:root
注：已忽略root编译报警
报错如下：
......

Applying /root/lede-master/target/linux/generic/pending-4.9/680-NET-skip-GRO-for-foreign-MAC-addresses.patch using plaintext: 
patching file include/linux/netdevice.h
patching file include/linux/skbuff.h
Hunk #1 succeeded at 747 with fuzz 2 (offset -10 lines).
patching file net/core/dev.c
Hunk #1 succeeded at 4572 (offset -10 lines).
Hunk #2 succeeded at 5870 (offset -10 lines).
patch: **** malformed patch at line 104: @@ -6075,6 +6121,7 @@ void netdev_upper_dev_unlink(struct net_

Patch failed!  Please fix /root/lede-master/target/linux/generic/pending-4.9/680-NET-skip-GRO-for-foreign-MAC-addresses.patch!
make[3]: *** [/root/lede-master/build_dir/toolchain-mipsel_mips32_gcc-7.5.0_musl/linux-4.9.215/.prepared] Error 1
make[3]: Leaving directory `/root/lede-master/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#9.15#3.28#14.20
make[2]: *** [toolchain/kernel-headers/compile] Error 2
make[2]: Leaving directory `/root/lede-master'
make[1]: *** [/root/lede-master/staging_dir/toolchain-mipsel_mips32_gcc-7.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory `/root/lede-master'
make: *** [world] Error 2

### (2) 路由器型号和固件版本

### (3) 详细日志

A：WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-system', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-program_options', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-date_time', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusic/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusicGo/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/v2ray/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/v2ray-plugin/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist
make[1]: Entering directory `/root/lede-master'
make[2]: Entering directory `/root/lede-master'
+ mkdir -p /root/lede-master/staging_dir/target-mipsel_mips32_musl
+ cd /root/lede-master/staging_dir/target-mipsel_mips32_musl
+ mkdir -p bin lib include stamp
mkdir -p /root/lede-master/build_dir/target-mipsel_mips32_musl/stamp
touch /root/lede-master/staging_dir/target-mipsel_mips32_musl/.prepared
make[3]: Entering directory `/root/lede-master/toolchain/gdb'
make[3]: Nothing to be done for `compile'.
make[3]: Leaving directory `/root/lede-master/toolchain/gdb'
time: toolchain/gdb/compile#0.00#0.00#0.01
make[3]: Entering directory `/root/lede-master/toolchain/binutils'
make[3]: Nothing to be done for `compile'.
make[3]: Leaving directory `/root/lede-master/toolchain/binutils'
time: toolchain/binutils/compile#0.00#0.00#0.01
make[3]: Entering directory `/root/lede-master/toolchain/gcc/initial'
make[3]: Nothing to be done for `compile'.
make[3]: Leaving directory `/root/lede-master/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#0.00#0.00#0.01
make[3]: Entering directory `/root/lede-master/toolchain/kernel-headers'
[... 788 lines cut ...]
patching file net/ipv6/route.c

Applying /root/lede-master/target/linux/generic/pending-4.9/671-net-provide-defines-for-_POLICY_FAILED-until-all-cod.patch using plaintext: 
patching file include/uapi/linux/fib_rules.h
patching file include/uapi/linux/icmpv6.h
patching file include/uapi/linux/rtnetlink.h

Applying /root/lede-master/target/linux/generic/pending-4.9/680-NET-skip-GRO-for-foreign-MAC-addresses.patch using plaintext: 
patching file include/linux/netdevice.h
patching file include/linux/skbuff.h
Hunk #1 succeeded at 747 with fuzz 2 (offset -10 lines).
patching file net/core/dev.c
Hunk #1 succeeded at 4572 (offset -10 lines).
Hunk #2 succeeded at 5870 (offset -10 lines).
patch: **** malformed patch at line 104: @@ -6075,6 +6121,7 @@ void netdev_upper_dev_unlink(struct net_

Patch failed!  Please fix /root/lede-master/target/linux/generic/pending-4.9/680-NET-skip-GRO-for-foreign-MAC-addresses.patch!
make[3]: *** [/root/lede-master/build_dir/toolchain-mipsel_mips32_gcc-7.5.0_musl/linux-4.9.215/.prepared] Error 1
make[3]: Leaving directory `/root/lede-master/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#9.15#3.28#14.20
make[2]: *** [toolchain/kernel-headers/compile] Error 2
make[2]: Leaving directory `/root/lede-master'
make[1]: *** [/root/lede-master/staging_dir/toolchain-mipsel_mips32_gcc-7.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory `/root/lede-master'
make: *** [world] Error 2

cjkzwe, 2020-03-13:
补充：恩山论坛也有类似情况：https://www.right.com.cn/forum/thread-265852-2-1.html

SharerMax, 2020-03-14:
你可能需要更新一下代码 #3776 已修复

cjkzwe, 2020-03-17:
依然有patch错误：

Applying /home/cjk/lede-master/target/linux/generic/hack-4.9/930-crashlog.patch using plaintext: 
patching file include/linux/crashlog.h
patching file init/Kconfig
patching file kernel/Makefile
patching file kernel/crashlog.c
patching file kernel/module.c
patching file mm/bootmem.c
patching file mm/memblock.c

Applying /home/cjk/lede-master/target/linux/generic/hack-4.9/950-net-patch-linux-kernel-to-support-shortcut-fe.patch using plaintext: 
patching file include/linux/skbuff.h
Hunk #1 succeeded at 743 (offset -10 lines).
patching file net/Kconfig
patching file net/core/dev.c
Hunk #1 FAILED at 2972.
Hunk #2 succeeded at 2971 with fuzz 2 (offset -8 lines).
Hunk #3 succeeded at 4123 (offset -10 lines).
Hunk #4 succeeded at 4175 (offset -10 lines).
Hunk #5 succeeded at 4204 (offset -10 lines).
1 out of 5 hunks FAILED -- saving rejects to file net/core/dev.c.rej
patching file net/netfilter/nf_conntrack_proto_tcp.c
Patch failed!  Please fix /home/cjk/lede-master/target/linux/generic/hack-4.9/950-net-patch-linux-kernel-to-support-shortcut-fe.patch!
Makefile:104: recipe for target '/home/cjk/lede-master/build_dir/toolchain-mipsel_mips32_gcc-7.5.0_musl/linux-4.9.215/.prepared' failed
make[3]: *** [/home/cjk/lede-master/build_dir/toolchain-mipsel_mips32_gcc-7.5.0_musl/linux-4.9.215/.prepared] Error 1
make[3]: Leaving directory '/home/cjk/lede-master/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#7.12#2.61#39.52
toolchain/Makefile:98: recipe for target 'toolchain/kernel-headers/compile' failed
make[2]: *** [toolchain/kernel-headers/compile] Error 2
make[2]: Leaving directory '/home/cjk/lede-master'
toolchain/Makefile:96: recipe for target '/home/cjk/lede-master/staging_dir/toolchain-mipsel_mips32_gcc-7.5.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/cjk/lede-master/staging_dir/toolchain-mipsel_mips32_gcc-7.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/cjk/lede-master'
/home/cjk/lede-master/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

SharerMax, 2020-03-17:
代码是最新的么？能否对比一下你的文件和最新的文件

closed 2020-04-08 by coolsnowwolf