coolsnowwolf/lede #3792: 请教lean大大一个问题，希望百忙中能答复一下。

norikohxb, 2020-03-13:
发现lean大的更新现在都是通过git pull了
而
./scripts/feeds update -a && ./scripts/feeds install -a
好像都不更新文件了
那么请教一个问题。
什么时候需要重新编辑config文件？
什么时候需要运行
rm -rf ./tmp && rm -rf .config
我发现我这几次跟着报bug，大部分都是因为我沿用了之前的config，然后还是精简的config，
所以这次新更新发生了，一堆奇奇怪怪的问题。

所以我想要请教lean大大，什么时候才需要rm掉config，重新生成一个。

guochan2019, 2020-03-13:
之前的config可以用，但是有更新后至少要运行一次，因为某些依赖会发生变化，另外这种问题最好不要在这里问，技能上的问题去论坛或油管上自学，至少现在你说的问题都不存在。

norikohxb, 2020-03-14:
youtube很少编译lede的更深入的。因为都是大神级的人都是简单的一笔带过

fgprodigal, 2020-03-14:
git pull和feeds update之后make menuconfig会自动更新.config

norikohxb, 2020-03-14:
但是我确实有发现，即使在系统没有红色警告提示config文件很老旧需要更新（系统会有红字提示这个时候你肯定只能删除config），在系统没有红色警告提示的时候就算是你用之前旧的config也可能会编译出错。我是清楚知道在系统红字提示你config已经太旧的时候你肯定需要删除掉，但是就算没有红字也不是每次你用上一次编译成功的config，这次一定可以编译通过的，我指的是这个问题。难道决定要不要删除掉config，是以当前confgi能不能编译通过为准吗？

fgprodigal, 2020-03-14:
我咋没看懂你在说什么，我一直用的同一个config文件，没遇到过要删除的情况

norikohxb, 2020-03-14:
查了一下，redsock2应该是搭配shadowsock的，我一开始就用trojan了，所以没有必要保留redsocks，至于dns2sock应该也是一种利用dns 到 sock的翻墙方式，和trojan并不是依存关系，是另外一种翻墙方式，所以2个都可以干掉
[image]

norikohxb, 2020-03-14:
> 我咋没看懂你在说什么，我一直用的同一个config文件，没遇到过要删除的情况

你有使用make clean吗？

closed 2020-03-14 by norikohxb

fgprodigal, 2020-03-14:
> > 我咋没看懂你在说什么，我一直用的同一个config文件，没遇到过要删除的情况
> 
> 你有使用make clean吗？

不用

norikohxb, 2020-03-14:
> > > 我咋没看懂你在说什么，我一直用的同一个config文件，没遇到过要删除的情况
> > 
> > 
> > 你有使用make clean吗？
> 
> 不用

那就说明我之前走火入魔挖空心思精简lean大的lede有关系，因为不是自己学会了编译了，就一直想要让lede更轻装上阵，强迫症，可能还是和我config有关系，就是之前在用的config，文件已经是去掉了很多东西，我不单纯去掉luci-app，我去掉了一堆东西。
Target Images -》
勾选Build EFI
去掉Use Console；Build VMware；Pad images；Build GRUB

global bulid setting-》
去掉Compile with support ；Enable ipv6；

base system-》
去掉block-mount.；dropbear；mtd；opkg；urandom-seed；urngd；getrandom

Administration-》
去掉htop

Network-》去掉ppp；wpad

Kernel modules-》
-》filesystems-》全部去掉
-》usb support-》去掉 kmod-usb-net；kmod-usb-storage-extras；kmod-usb-storage；kmod-usb-hid 可以全部去掉
-》network devices -》全去掉仅保留kmod-e1000e 
-》network support-》kmod-bonding                                             
-》other modules-》去掉kmod-button-hotplug
-》wireless drivers-》全去掉
-》Cryptographic API modules**所有可以去掉的都可以去掉
-》I2C support-》全去掉
-》Netfilter Extensions-》可以去掉的通通去掉
-》Libraries-》全部去掉
-》hardware monitoring support 清空
-》Inpurt modules 清空

Libraries-》
-》Firewall-》去掉 libip6tc. IPv6 firewall - shared libiptc library ；

Network-》去掉Hostapd-common；iw.cfg80211；uclient-fetch
-》IP Addresses and Names-》清空
-》OpenLDAP-》清空

Utilities-》去掉coremark；iwinfo. ；lm-sensors；shell sync
openssl-util. Open source SSL toolkit (utility)
-》disc 去掉fdisk+ partx	
-》filesystem去掉e2fsprogs仅仅保留mkf2fs（必须保留）

norikohxb, 2020-03-14:
去掉了那么多以后，又喜欢追新lean大的lede，所以会造成经常无法用原来的config编译通过。

fgprodigal, 2020-03-14:
> 去掉了那么多以后，又喜欢追新lean大的lede，所以会造成经常无法用原来的config编译通过。

你可以看我的repo，每天都在自动编译

norikohxb, 2020-03-14:
https://github.com/fgprodigal?tab=repositories
你指的repo是这个吗？