coolsnowwolf/lede #3794: 20200314今日版本，配置时多拨当时可以上网，重启后多拨完成无法上网。

georip, 2020-03-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
PPPOE多拨，重启路由器后无法上网
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
没有
## 2. 详细叙述
20200314 今日版本，pppoe-单拨上网正常，重启后上网正常。在单拨之后再多拨，上网正常，然后重启，查看日志多拨成功，却无法上网，再改单拨也无法上网，请求修复。另外多拨的启动优先级较低，能否提高启动优先级？单拨基本在40多秒后就拨号成功了，多拨会延迟到90-110秒以后。
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
R619AC 128M 今日（20200314 git pull）
A：

### (3) 详细日志
Fri Mar 13 23:53:16 2020 kern.debug kernel: [   37.870228] ath: Country alpha2 being used: US
Fri Mar 13 23:53:16 2020 kern.debug kernel: [   37.870240] ath: Regpair used: 0x3a
Fri Mar 13 23:53:16 2020 user.info kernel: [   37.883943] kmodloader: done loading kernel modules from /etc/modules.d/*
Fri Mar 13 23:53:16 2020 daemon.debug zram_start: activating '/dev/zram0' for swapping (248 MegaBytes)
Fri Mar 13 23:53:16 2020 daemon.notice procd: /etc/rc.d/S15zram: zram_start: activating '/dev/zram0' for swapping (248 MegaBytes)
Fri Mar 13 23:53:16 2020 daemon.debug zram_reset: enforcing defaults via /sys/block/zram0/reset
Fri Mar 13 23:53:16 2020 daemon.notice procd: /etc/rc.d/S15zram: zram_reset: enforcing defaults via /sys/block/zram0/reset
Fri Mar 13 23:53:16 2020 daemon.debug zram_comp_streams: Set max compression streams to '4' for zram '/dev/zram0'
Fri Mar 13 23:53:16 2020 daemon.notice procd: /etc/rc.d/S15zram: zram_comp_streams: Set max compression streams to '4' for zram '/dev/zram0'
Fri Mar 13 23:53:16 2020 kern.info kernel: [   39.710477] zram0: detected capacity change from 0 to 260046848
Fri Mar 13 23:53:16 2020 daemon.notice procd: /etc/rc.d/S15zram: Setting up swapspace version 1, size = 248 MiB (260042752 bytes)
Fri Mar 13 23:53:16 2020 daemon.notice procd: /etc/rc.d/S15zram: no label, UUID=b2b7f2a3-d4dd-4426-ae31-31774d7dc3e4
Fri Mar 13 23:53:16 2020 kern.info kernel: [   39.815866] Adding 253948k swap on /dev/zram0.  Priority:-2 extents:1 across:253948k SS
Fri Mar 13 23:53:17 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Mar 13 23:53:17 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Mar 13 23:53:17 2020 authpriv.info dropbear[1697]: Not backgrounding
Fri Mar 13 23:53:18 2020 kern.info kernel: [   41.384889] xt_FULLCONENAT: RFC3489 Full Cone NAT module
Fri Mar 13 23:53:18 2020 kern.info kernel: [   41.384889] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
Fri Mar 13 23:53:20 2020 user.notice : Added device handler type: 8021ad
Fri Mar 13 23:53:20 2020 user.notice : Added device handler type: 8021q
Fri Mar 13 23:53:20 2020 user.notice : Added device handler type: macvlan
Fri Mar 13 23:53:20 2020 user.notice : Added device handler type: veth
Fri Mar 13 23:53:20 2020 user.notice : Added device handler type: bridge
Fri Mar 13 23:53:20 2020 user.notice : Added device handler type: Network device
Fri Mar 13 23:53:20 2020 user.notice : Added device handler type: tunnel
Fri Mar 13 23:53:21 2020 daemon.err block: No "mount.ubi" utility available
Fri Mar 13 23:53:21 2020 daemon.err block: mounting /dev/mtdblock0 (ubi) as /mnt/mtdblock0 failed (25) - Not a tty
Fri Mar 13 23:53:21 2020 daemon.err block: /dev/ubiblock0_1 is already mounted on /rom
Fri Mar 13 23:53:21 2020 daemon.err block: /dev/ubi0_2 is already mounted on /overlay
Fri Mar 13 23:53:21 2020 daemon.err block: /dev/sda1 is already mounted on /mnt/sda1
Fri Mar 13 23:53:21 2020 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: Not found
Fri Mar 13 23:53:21 2020 cron.info crond[1952]: crond (busybox 1.30.0) started, log level 5
Fri Mar 13 23:53:21 2020 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.
Fri Mar 13 23:53:22 2020 kern.info kernel: [   45.468469] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Fri Mar 13 23:53:22 2020 kern.info kernel: [   45.470971] device eth0 entered promiscuous mode
Fri Mar 13 23:53:22 2020 kern.info kernel: [   45.473862] br-lan: port 1(eth0.1) entered blocking state
Fri Mar 13 23:53:22 2020 kern.info kernel: [   45.478082] br-lan: port 1(eth0.1) entered disabled state
Fri Mar 13 23:53:22 2020 kern.info kernel: [   45.483664] device eth0.1 entered promiscuous mode
Fri Mar 13 23:53:22 2020 kern.info kernel: [   45.490818] br-lan: port 1(eth0.1) entered blocking state
Fri Mar 13 23:53:22 2020 kern.info kernel: [   45.493367] br-lan: port 1(eth0.1) entered forwarding state
Fri Mar 13 23:53:22 2020 daemon.notice netifd: Interface 'lan' is enabled
Fri Mar 13 23:53:22 2020 daemon.notice netifd: Interface 'lan' is setting up now
Fri Mar 13 23:53:22 2020 kern.info kernel: [   45.499124] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
Fri Mar 13 23:53:22 2020 daemon.notice netifd: Interface 'lan' is now up
Fri Mar 13 23:53:22 2020 daemon.notice netifd: Interface 'loopback' is enabled
Fri Mar 13 23:53:22 2020 daemon.notice netifd: Interface 'loopback' is setting up now
Fri Mar 13 23:53:22 2020 daemon.notice netifd: Interface 'loopback' is now up
Fri Mar 13 23:53:22 2020 daemon.notice netifd: Interface 'wan' is enabled
Fri Mar 13 23:53:22 2020 daemon.notice netifd: Interface 'wan6' is enabled
Fri Mar 13 23:53:22 2020 daemon.notice netifd: bridge 'br-lan' link is up
Fri Mar 13 23:53:22 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Fri Mar 13 23:53:22 2020 daemon.notice netifd: Network device 'eth0' link is up
[... 675 lines cut ...]
Fri Mar 13 23:54:09 2020 daemon.info hostapd: wlan1: STA d8:6c:02:ca:8f:ff IEEE 802.11: associated (aid 3)
Fri Mar 13 23:54:09 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED d8:6c:02:ca:8f:ff
Fri Mar 13 23:54:09 2020 daemon.info hostapd: wlan1: STA d8:6c:02:ca:8f:ff RADIUS: starting accounting session 4CFC535C34F1A6F2
Fri Mar 13 23:54:09 2020 daemon.info hostapd: wlan1: STA d8:6c:02:ca:8f:ff WPA: pairwise key handshake completed (RSN)
Fri Mar 13 23:54:09 2020 daemon.info : 10[KNL] 2408:8206:25e0:5747:651d:f385:5906:4c60 appeared on pppoe-vwan2
Fri Mar 13 23:54:09 2020 daemon.info : 12[KNL] 2408:8207:25e5:7c20::1 appeared on br-lan
Fri Mar 13 23:54:09 2020 daemon.notice netifd: Interface 'vwan2_6' is now up
Fri Mar 13 23:54:10 2020 daemon.info : 06[KNL] 2408:8206:25e0:5746:f1d6:a293:f5ce:e189 appeared on pppoe-vwan1
Fri Mar 13 23:54:10 2020 daemon.info : 10[KNL] 2408:8207:25e5:7c10::1 appeared on br-lan
Fri Mar 13 23:54:10 2020 daemon.notice netifd: Interface 'vwan1_6' is now up
Fri Mar 13 23:54:10 2020 user.notice simple-adblock [8344]: simple-adblock is currently disabled.
Fri Mar 13 23:54:10 2020 user.notice simple-adblock [8344]: Run the following commands before starting service again:
Fri Mar 13 23:54:10 2020 user.notice simple-adblock [8344]: uci set simple-adblock.config.enabled='1'; uci commit;
Fri Mar 13 23:54:12 2020 kern.warn kernel: [   95.874844] hrtimer: interrupt took 198792 ns
Fri Mar 13 23:54:12 2020 cron.info crond[9974]: crond (busybox 1.30.0) started, log level 5
Fri Mar 13 23:54:19 2020 daemon.notice netifd: Interface 'vwan3_6' has lost the connection
Fri Mar 13 23:54:19 2020 daemon.info : 15[KNL] 2408:8206:25e0:5748:81a9:9893:45d6:795d disappeared from pppoe-vwan3
Fri Mar 13 23:54:19 2020 daemon.notice netifd: Interface 'vwan3_6' is now up
Fri Mar 13 23:54:19 2020 daemon.info : 05[KNL] 2408:8206:25e0:5748:81a9:9893:45d6:795d appeared on pppoe-vwan3
Fri Mar 13 23:54:19 2020 daemon.info : 10[KNL] 2408:8207:25e5:7c00::1 appeared on br-lan
Fri Mar 13 23:55:36 2020 user.notice simple-adblock [4402]: simple-adblock is currently disabled.
Fri Mar 13 23:55:36 2020 user.notice simple-adblock [4402]: Run the following commands before starting service again:
Fri Mar 13 23:55:37 2020 user.notice simple-adblock [4402]: uci set simple-adblock.config.enabled='1'; uci commit;

A：

qq1316237299, 2020-03-13:
我也是这情况，所以干脆不多拨了 把mawan3配置都删除了

4593645, 2020-03-14:
若若问句，我为啥编译不出WNDR3800的包。之前的2019版本是可以的。

919129850, 2020-03-14:
x86的也有类似情况，多拨都可以获得ip，但mwan3负载均衡状态那里始终显示已停用，若要配置mwan3，则直接在保存应用那卡死转圈。。。3月5日固件正常，3月8日往后的固件就一直有这个问题

qq1316237299, 2020-03-14:
对的，我也是这个问题，根本就是转圈圈，所以我干脆把mwan3   多播插件全部不编译进去了

georip, 2020-03-15:
楼上的朋友，有正解的办法么？

qq1316237299, 2020-03-15:
多拨影响ss

iwowcn, 2020-03-17:
同样的问题，3.5日固件一切正常，3.11之后的一直不行，删除MWAN3等多拨插件就能恢复正常。

iwowcn, 2020-03-17:
最近老大更新了去广告插件，应该DNSMASQ的问题。

bluestar800, 2020-03-18:
多拨后，mwan3配置好可以使用，重启后我也是不可以用，看mwan3的状态，所有端口都显示的down，但各个端口获取到了公网IP。我都是把mwan3的配置在保存并应用一下，较慢，大概2-3分钟左右，然后mwan3的状态就正常了，就可以上网了……先这样临时用，反正除了停电，不重启X86

Kisoul, 2020-03-23:
我也是这个问题，我一度认为是ipv6的问题现在看来不是，一线多播然后，重启后wan网以获得ip但是就是无法上网，路由本身也是，除非在lan的配置里面随便改一个参数，然后再改回来，保存一下就好了。可能是保存的时候是从新触发某个组件重启。希望能修复，这个很重要。话说我着多播并没有出现影响ss的情况。

georip, 2020-03-23:
这个很可能与防火墙有关系。

919129850, 2020-03-23:
个人推测有可能是3月8号更新ppp导致的问题

Kisoul, 2020-03-23:
各位，由办法救吗？还是要等大雕修复啊？

Kisoul, 2020-03-23:
[image]
貌似有网友解决了，回去试试看

iwowcn, 2020-03-23:
> 我也是这个问题，我一度认为是ipv6的问题现在看来不是，一线多播然后，重启后网网ip但是就是无法上网，路由本身也是，除非在lan的配置里面随便改一个参数，然后再改回来，保存一下就好了。可能是保存的时候是从新触发某个组件重启。希望能修复，这个很重要。话说我着多播并没有出现影响ss的情况。
> 再多说一嘴：多播虽然已已不会增加下行速率，但是上行速度会增加非常明显，在国内家庭宽带上下行速率不对等的情况下这是曲线就
> 援救的唯一机会。话说我是22日白天最新编译的版本，之前某个时间的确是好的

3.5版是正常的，我3.5号编译的一切ok

Kisoul, 2020-03-23:
> > 我也是这个问题，我一度认为是ipv6的问题现在看来不是，一线多播然后，重启后网网ip但是就是无法上网，路由本身也是，除非在lan的配置里面随便改一个参数，然后再改回来，保存一下就好了。可能是保存的时候是从新触发某个组件重启。希望能修复，这个很重要。话说我着多播并没有出现影响ss的情况。
> > 再多说一嘴：多播虽然已已不会增加下行速率，但是上行速度会增加非常明显，在国内家庭宽带上下行速率不对等的情况下这是曲线就
> > 援救的唯一机会。话说我是22日白天最新编译的版本，之前某个时间的确是好的
> 
> 3.5版是正常的，我3.5号编译的一切ok

如何回到那天？你能看一下你mwan3的版本是号吗？谢谢

qq1316237299, 2020-03-23:
单线多拨，会影响到好多地方的实际使用，我现在都不敢多拨了

   

  
  
 
&mdash;

georip, 2020-03-23:
> 
> 
> 个人推测有可能是3月8号更新ppp导致的问题

应该与PPP更新无关。我这里就是最新的ppp。没记错应该是2.4.8。我最后一次是3月9号编译的。就正常。之后的都无不正常了。

Kisoul, 2020-03-23:
[image]
目前我直接在软件包中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，都能上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重启后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。

Kisoul, 2020-03-23:
[image]
[image]
有一点小问题是upnp不显示，但是测试下来upnp实际上应该是生效的。

Kisoul, 2020-03-23:
[image]
上行速度叠加4倍生效。

iwowcn, 2020-03-25:
3月25日最新代码编译，网络——负载均衡——全局——本地源地址改为“wan”，恢复正常。

iwowcn, 2020-03-25:
> [image]
> 目前我直接在一节段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。

请问如何更新到最新版本？或者如何单独编译最新版的ipk包？

Kisoul, 2020-03-25:
> > [image]
> > 目前我直接在一节段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。
> 
> 请问如何更新到最新版本？或者如何单独编译最新版的ipk包？

在软件包里更新一下，然后在可用软件包里有直接更新，貌似upnp不显示也一起修复了，我查了恩山，有人和你解决方案类似，把本地源改为none，我更新后看了一下也是none，估计更新的时候系统自动默认了。

qq1316237299, 2020-03-25:
> 3月25日最新代码编译，网络——负载均衡——全局——本地源地址改为“wan”，恢复正常。

谢谢，这个方法很好

Kisoul, 2020-03-25:
> > [image]
> > 目前我直接在一节段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。
> 
> 请问如何更新到最新版本？或者如何单独编译最新版的ipk包？

升级后即使源改为lan，重启也可以访问

qq1316237299, 2020-03-25:
> > > [image]
> > > 目前我直接在一节段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。
> > 
> > 
> > 请问如何更新到最新版本？或者如何单独编译最新版的ipk包？
> 
> 升级后即使源改为lan，重启也可以访问

我是最新的 默认lan是重启以后就废了 然后改成wan就成功了

qq1316237299, 2020-03-25:
> > > > [image]
> > > > 目前我直接在分段段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。
> > > 
> > > 
> > > 请问如何更新到最新版本？或者如何单独编译最新版的ipk包？
> > 
> > 
> > 升级后即使源替换lan，重启也可以访问
> 
> 我是最新的默认lan是重启以后就废了然后改成wan就成功了

[image]

Kisoul, 2020-03-25:
> > > > > [image]
> > > > > 目前我直接在分段段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。
> > > > 
> > > > 
> > > > 请问如何更新到最新版本？或者如何单独编译最新版的ipk包？
> > > 
> > > 
> > > 升级后即使源替换lan，重启也可以访问
> > 
> > 
> > 我是最新的默认lan是重启以后就废了然后改成wan就成功了
> 
> [image]

不知道再改回lan会不会又不行了，我是更新到最新版的mwan3了，改回lan依然没问题

qq1316237299, 2020-03-25:
> > > > > > [image]
> > > > > > 目前我直接在分段段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。
> > > > > 
> > > > > 
> > > > > 请问如何更新到最新版本？或者如何单独编译最新版的ipk包？
> > > > 
> > > > 
> > > > 升级后即使源替换lan，重启也可以访问
> > > 
> > > 
> > > 我是最新的默认lan是重启以后就废了然后改成wan就成功了
> > 
> > 
> > [image]
> 
> 不知道再改回lan会不会又不行了，我是更新到2.3.3-2的mwan3了，改回lan依然没问题

好多问题  可道云上不去了  内网才能上去 晕了

qq1316237299, 2020-03-25:
> > > > > > > [image]
> > > > > > > 目前我直接在分段段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。
> > > > > > 
> > > > > > 
> > > > > > 请问如何更新到最新版本？或者如何单独编译最新版的ipk包？
> > > > > 
> > > > > 
> > > > > 升级后即使源替换lan，重启也可以访问
> > > > 
> > > > 
> > > > 我是最新的默认lan是重启以后就废了然后改成wan就成功了
> > > 
> > > 
> > > [image]
> > 
> > 
> > 不知道再改回lan会不会又不行了，我是更新到2.3.3-2的mwan3了，改回lan依然没问题
> 
> 好多问题 可道云上不去了 内网才能上去 晕了

好了 已修复

Kisoul, 2020-03-25:
> > > > > > > > [image]
> > > > > > > > 目前我直接在分段段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。
> > > > > > > 
> > > > > > > 
> > > > > > > 请问如何更新到最新版本？或者如何单独编译最新版的ipk包？
> > > > > > 
> > > > > > 
> > > > > > 升级后即使源替换lan，重启也可以访问
> > > > > 
> > > > > 
> > > > > 我是最新的默认lan是重启以后就废了然后改成wan就成功了
> > > > 
> > > > 
> > > > [image]
> > > 
> > > 
> > > 不知道再改回lan会不会又不行了，我是更新到2.3.3-2的mwan3了，改回lan依然没问题
> > 
> > 
> > 好多问题 可道云上不去了 内网才能上去 晕了
> 
> 好了 已修复

你是升级mwan3了？

qq1316237299, 2020-03-25:
对的 最新的

iwowcn, 2020-03-26:
> > > [image]
> > > 目前我直接在一节段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。
> > 
> > 
> > 请问如何更新到最新版本？或者如何单独编译最新版的ipk包？
> 
> 在软件包里更新一下，然后在可用软件包里有直接更新，貌似upnp不显示也一起修复了，我查了恩山，有人和你解决方案类似，把本地源改为none，我更新后看了一下也是none，估计更新的时候系统自动默认了。

软件包更新报错，何故？没有下载完整？
Upgrading mwan3 on root from 2.6.18-1 to 2.8.3-2...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/mwan3_2.8.3-2_all.ipk
Collected errors:
 * opkg_install_pkg: Package size mismatch: mwan3 is 13983 bytes, expecting 13981 bytes
 * opkg_install_cmd: Cannot install package mwan3.

iwowcn, 2020-03-26:
> > > [image]
> > > 目前我直接在一节段中升级了mwan3到最新，好像修复了，重启后物论有没有ipv6，可以上网了，大家可以试一下，不过最好升级前备份一次，因为我只测试了是否重新启动后能直接上网，如果没问题，希望lean大雕能看到，然后更新一下。
> > 
> > 
> > 请问如何更新到最新版本？或者如何单独编译最新版的ipk包？
> 
> 在软件包里更新一下，然后在可用软件包里有直接更新，貌似upnp不显示也一起修复了，我查了恩山，有人和你解决方案类似，把本地源改为none，我更新后看了一下也是none，估计更新的时候系统自动默认了。

其实最好的方法应该是在编译的时候就直接更新到2.8.3，就是不知道要源码的哪个位置，可以研究一下。

Kisoul, 2020-03-26:
> 软件包更新报错，何故？没有下载完整？
> Upgrading mwan3 on root from 2.6.18-1 to 2.8.3-2...
> Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/mwan3_2.8.3-2_all.ipk
> Collected errors:
>
> * [...]

对文件大小不一致，偶尔抽风，过段时间就好了

georip, 2020-03-27:
> 其实最好的方法应该是在编译的时候就直接更新到2.8.3，就是不知道要源码的哪个位置，可以研究一下。

我昨天在编译的时候更新了mwan3到最新版。编译执行通过了。刷入系统后多拨和上网都正常了。但貌似均衡负载有问题，在MWAN配置里看不到接口情况。也有能是我更新的时候继承了配置。晚上再回去试试双清rom，重新配置是什么情况。
[image]

iwowcn, 2020-03-27:
> 我昨天在编译的时候更新了mwan3到最新版。编译执行通过了。刷入系统后多拨和上网都正常了。但貌似均衡负载有问题，在MWAN配置里看不到接口情况。也有能是我更新的时候继承了配置。晚上再回去试试双清rom，重新配置是什么情况。
> [image]

我昨天倒是没编译，今天回去试试看看更新了没。

qq1316237299, 2020-03-27:
多拨的问题太多了，现在又不敢多拨了

momonoshiori, 2020-03-27:
这几天朋友那边多拨出现过这个问题，mwan里接口很久才检测完而且如果开机中途断网后就一直是禁用状态了，让他make clean后删除配置文件重新选好插件编译完就正常了，不知道什么原因，我这里没多拨没碰到过。

Christopher-Songs, 2020-04-04:
哇，我刚好是这两天开始尝试用hyper v跑lede，在这个问题上卡了好多天，一直怀疑是虚拟机哪里出了问题，慢慢的才发现可能是mwan3的问题，把我坑惨了，欲哭无泪。没有责怪大佬的意思，只是想表达一下debug的过程太艰难。

qq1316237299, 2020-04-08:
单线多拨，会影响到好多地方的实际使用，我现在都不敢多拨了

xuwangli45, 2020-04-27:
同样的问题，要在启动项把mwan3启用才能上网，但是状态里负载均衡一直显示禁用，实际接口里都正常了

qq1316237299, 2020-04-28:
> 同样的问题，要在启动项把mwan3启用才能上网，但是状态里负载均衡一直显示禁用，实际接口里都正常了

启动项默认是关的吗，还真没注意，我是把负载均衡里面的全局lan改成none就好了，但我感觉又不需要多拨干脆又没多拨了

xuwangli45, 2020-04-28:
> > 同样的问题，要在启动项把mwan3启用才能上网，但是状态里负载均衡一直显示禁用，实际接口里都正常了
> 
> 启动项默认是关的吗，还真没注意，我是把负载均衡里面的全局lan改成none就好了，但我感觉又不需要多拨干脆又没多拨了

启动项里大部分都不是启用的，但是这个开了启用后，负载均衡里的状态一直显示禁用，但是正常上网

qq1316237299, 2020-04-28:
> > > 同样的问题，要在启动项把mwan3启用才能上网，但是状态里负载均衡一直显示禁用，实际接口里都正常了
> > 
> > 
> > 启动项默认是关的吗，还真没注意，我是把负载均衡里面的全局lan改成none就好了，但我感觉又不需要多拨干脆又没多拨了
> 
> 启动项里大部分都不是启用的，但是这个开了启用后，负载均衡里的状态一直显示禁用，但是正常上网

都应该是启动的吧，这个没怎么注意，反正多拨以后就只有一个在线的，其他都是禁用的，把负债均衡里面的全局改成none就完美解决了，很奇怪

dreamer2q, 2020-05-08:
[image]

每次配置mwan3的时候，点击应用后，会卡在上面的，卡很久。

我想问一下，怎么解决这个？

closed 2020-07-04 by coolsnowwolf