coolsnowwolf/lede #3796: adbyby每到6点，日志一片狼藉

Fyred, 2020-03-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [√ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：6点adbyby自动更新，HTTP连续错误429，是把adbyby服务器干瘫了吗？然后我自己op也瘫了，8点多一看ad和ssr双双未运行。

### (2) 路由器型号和固件版本

A：x86，OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7)

### (3) 详细日志

A：Sat Mar 14 00:35:09 2020 daemon.warn pdnsd[8236]: Caught signal 15. Exiting.
Sat Mar 14 00:35:09 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Mar 14 00:35:09 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Mar 14 00:35:12 2020 cron.info crond[32565]: crond (busybox 1.30.0) started, log level 8
Sat Mar 14 00:35:16 2020 daemon.info pdnsd[393]: pdnsd-1.2.9b-par starting.
Sat Mar 14 00:35:29 2020 daemon.warn pdnsd[393]: Caught signal 15. Exiting.
Sat Mar 14 00:35:29 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Mar 14 00:35:29 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Mar 14 00:35:32 2020 cron.info crond[1043]: crond (busybox 1.30.0) started, log level 8
Sat Mar 14 00:35:36 2020 daemon.info pdnsd[1343]: pdnsd-1.2.9b-par starting.
Sat Mar 14 01:00:00 2020 cron.info crond[1043]: USER root pid 10790 cmd echo "" > /tmp/ssrplus.log
Sat Mar 14 04:00:00 2020 cron.info crond[1043]: USER root pid 26039 cmd /usr/bin/lua /usr/share/shadowsocksr/subscribe.lua
Sat Mar 14 04:00:00 2020 daemon.warn pdnsd[1343]: Caught signal 15. Exiting.
Sat Mar 14 04:00:04 2020 cron.info crond[26211]: crond (busybox 1.30.0) started, log level 8
Sat Mar 14 06:00:00 2020 cron.info crond[26211]: USER root pid 1103 cmd /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1
Sat Mar 14 06:00:16 2020 cron.info crond[1154]: crond (busybox 1.30.0) started, log level 8
Sat Mar 14 06:00:17 2020 daemon.err adbyby[1769]: --2020-03-14 06:00:17--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/md5.json
Sat Mar 14 06:00:17 2020 daemon.err adbyby[1769]: Resolving adbyby.coding.net... 118.126.70.252
Sat Mar 14 06:00:17 2020 daemon.err adbyby[1769]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sat Mar 14 06:00:18 2020 daemon.err adbyby[1769]: HTTP request sent, awaiting response... 429
Sat Mar 14 06:00:18 2020 daemon.err adbyby[1769]: 2020-03-14 06:00:18 ERROR 429: (no description).
Sat Mar 14 06:00:18 2020 daemon.err adbyby[1769]:
Sat Mar 14 06:00:18 2020 daemon.err adbyby[1769]: --2020-03-14 06:00:18--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/lazy.txt
Sat Mar 14 06:00:18 2020 daemon.err adbyby[1769]: Resolving adbyby.coding.net... 118.126.70.252
Sat Mar 14 06:00:18 2020 daemon.err adbyby[1769]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sat Mar 14 06:00:19 2020 daemon.err adbyby[1769]: HTTP request sent, awaiting response... 429
Sat Mar 14 06:00:19 2020 daemon.err adbyby[1769]: 2020-03-14 06:00:19 ERROR 429: (no description).
Sat Mar 14 06:00:19 2020 daemon.err adbyby[1769]:
Sat Mar 14 06:00:19 2020 daemon.err adbyby[1769]: --2020-03-14 06:00:19--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/video.txt
Sat Mar 14 06:00:19 2020 daemon.err adbyby[1769]: Resolving adbyby.coding.net... 118.126.70.252
Sat Mar 14 06:00:19 2020 daemon.err adbyby[1769]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sat Mar 14 06:00:20 2020 daemon.err adbyby[1769]: HTTP request sent, awaiting response... 429
Sat Mar 14 06:00:20 2020 daemon.err adbyby[1769]: 2020-03-14 06:00:20 ERROR 429: (no description).
Sat Mar 14 06:00:20 2020 daemon.err adbyby[1769]:
Sat Mar 14 06:00:25 2020 daemon.err adbyby[1814]: --2020-03-14 06:00:25--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/md5.json
Sat Mar 14 06:00:25 2020 daemon.err adbyby[1814]: Resolving adbyby.coding.net... 118.126.70.252
Sat Mar 14 06:00:25 2020 daemon.err adbyby[1814]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sat Mar 14 06:00:25 2020 daemon.err adbyby[1814]: HTTP request sent, awaiting response... 429
Sat Mar 14 06:00:25 2020 daemon.err adbyby[1814]: 2020-03-14 06:00:25 ERROR 429: (no description).
Sat Mar 14 06:00:25 2020 daemon.err adbyby[1814]:
Sat Mar 14 06:00:25 2020 daemon.err adbyby[1814]: --2020-03-14 06:00:25--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/lazy.txt
Sat Mar 14 06:00:25 2020 daemon.err adbyby[1814]: Resolving adbyby.coding.net... 118.126.70.252
Sat Mar 14 06:00:25 2020 daemon.err adbyby[1814]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sat Mar 14 06:00:27 2020 daemon.err adbyby[1814]: HTTP request sent, awaiting response... 429
Sat Mar 14 06:00:27 2020 daemon.err adbyby[1814]: 2020-03-14 06:00:27 ERROR 429: (no description).
Sat Mar 14 06:00:27 2020 daemon.err adbyby[1814]:
Sat Mar 14 06:00:27 2020 daemon.err adbyby[1814]: --2020-03-14 06:00:27--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/video.txt
Sat Mar 14 06:00:27 2020 daemon.err adbyby[1814]: Resolving adbyby.coding.net... 118.126.70.252
Sat Mar 14 06:00:27 2020 daemon.err adbyby[1814]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sat Mar 14 06:00:27 2020 daemon.err adbyby[1814]: HTTP request sent, awaiting response... 429
Sat Mar 14 06:00:27 2020 daemon.err adbyby[1814]: 2020-03-14 06:00:27 ERROR 429: (no description).
[... 53 lines cut ...]
Sat Mar 14 06:00:48 2020 daemon.err adbyby[1935]: 2020-03-14 06:00:48 ERROR 429: (no description).
Sat Mar 14 06:00:48 2020 daemon.err adbyby[1935]:
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: --2020-03-14 06:00:53--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/md5.json
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: Resolving adbyby.coding.net... 118.126.70.252
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: HTTP request sent, awaiting response... 429
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: 2020-03-14 06:00:53 ERROR 429: (no description).
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]:
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: --2020-03-14 06:00:53--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/lazy.txt
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: Resolving adbyby.coding.net... 118.126.70.252
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: HTTP request sent, awaiting response... 429
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: 2020-03-14 06:00:53 ERROR 429: (no description).
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]:
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: --2020-03-14 06:00:53--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/video.txt
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: Resolving adbyby.coding.net... 118.126.70.252
Sat Mar 14 06:00:53 2020 daemon.err adbyby[1976]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sat Mar 14 06:00:54 2020 daemon.err adbyby[1976]: HTTP request sent, awaiting response... 429
Sat Mar 14 06:00:54 2020 daemon.err adbyby[1976]: 2020-03-14 06:00:54 ERROR 429: (no description).
Sat Mar 14 06:00:54 2020 daemon.err adbyby[1976]:
Sat Mar 14 06:00:54 2020 daemon.info procd: Instance adbyby::instance1 s in a crash loop 6 crashes, 1 seconds since last crash
Sat Mar 14 08:26:12 2020 cron.info crond[13041]: crond (busybox 1.30.0) started, log level 8
Sat Mar 14 08:26:14 2020 daemon.info pdnsd[13256]: pdnsd-1.2.9b-par starting.
Sat Mar 14 08:26:14 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Mar 14 08:26:14 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

guoyuc, 2020-03-14:
一样的情况，我早上这个时间手动打开 https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/video.txt 这个链接也是会报错，应该是大量的人同时在这个时间访问超过了服务器连接数限制，所以拒绝访问。过几分钟就能正常访问了。
我手动把adbyby脚本的更新地址换成GitHub上的地址暂时避免了这个问题。

leishao1993, 2020-03-14:
一样的情况    我把自动更新关了

fgprodigal, 2020-03-14:
是的。。每天早晨wifi都是挂的。。

liwenjie119, 2020-03-14:
那可以加个选项，自定义自动更新时间

leishao1993, 2020-03-14:
现在无法正常运行adb了   日志提示:instance1 s in a crash loop 6 crashes, 31 seconds since last crash
重启也没有用   开启不了

coolsnowwolf, 2020-03-14:
服务器那边的问题，等就好了

hplee0120, 2020-03-14:
同样的情况，然后插件停止运行了。刚才发现抖音连不上网排查了下才发现这插件死掉了，不过不清除为什么插件死掉后不应该是去广告功能失效吗，导致某些网络也出问题了

hplee0120, 2020-03-14:
遇到同样的问题，然后adbyby自己停止运行了，后来发现抖音连不上网查了下才发现。比较好奇插件未运行状态不是去广告失效吗。我这边是直接影响到了其他网络。比如抖音

Wilson-coder, 2020-03-15:
我估计是因为这个编译工具包太普及，导致所有人都是在早晨6点去连接服务器，成变相的DDOS攻击使对方服务器过载了。因为刚好我设置了路由器周日早上5点自动重启，从系统日志看，获取adbyby服务器更新一次成功，到了6点的时候adbyby模块自己更新就和楼主反馈的情况一样了。

建议把这个自动更新的时间设置成可以由用户自己修改，以避开同一时间同时去连接服务器带来的超负荷？

附带系统日志：

Sun Mar 15 05:00:42 2020 daemon.notice netifd: Interface 'wan' is now up
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]: --2020-03-15 05:00:43--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/md5.json
Sun Mar 15 05:00:43 2020 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]: Resolving adbyby.coding.net... 118.126.70.252
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]: HTTP request sent, awaiting response... 200 OK
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]: Length: 131 [text/plain]
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]: Saving to: '/tmp/md5.json'
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]:
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]:      0K                                                       100% 2.24M=0s
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]:
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]: 2020-03-15 05:00:43 (2.24 MB/s) - '/tmp/md5.json' saved [131/131]
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]:
Sun Mar 15 05:00:43 2020 cron.info crond[6073]: crond (busybox 1.30.0) started, log level 8
Sun Mar 15 05:00:43 2020 daemon.err adbyby[4705]: --2020-03-15 05:00:43--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/lazy.txt
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]: --2020-03-15 05:00:44--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/md5.json
Sun Mar 15 05:00:44 2020 cron.info crond[6946]: crond (busybox 1.30.0) started, log level 8
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]: Resolving adbyby.coding.net... 118.126.70.252
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sun Mar 15 05:00:44 2020 cron.info crond[7060]: crond (busybox 1.30.0) started, log level 8
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]: HTTP request sent, awaiting response... 200 OK
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]: Length: 131 [text/plain]
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]: Saving to: '/tmp/md5.json'
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]:
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]:      0K                                                       100% 22.4M=0s
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]:
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]: 2020-03-15 05:00:44 (22.4 MB/s) - '/tmp/md5.json' saved [131/131]
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]:
Sun Mar 15 05:00:44 2020 daemon.err adbyby[6737]: --2020-03-15 05:00:44--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/lazy.txt
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]: Resolving adbyby.coding.net... 118.126.70.252
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]: HTTP request sent, awaiting response... 200 OK
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]: Length: 559336 (546K) [text/plain]
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]: Saving to: '/tmp/lazy.txt'
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:      0K .......... .......... .......... .......... ..........  9%  488K 1s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:     50K .......... .......... .......... .......... .......... 18%  495K 1s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:    100K .......... .......... .......... .......... .......... 27% 1.43M 1s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:    150K .......... .......... .......... .......... .......... 36% 26.7M 0s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:    200K .......... .......... .......... .......... .......... 45% 1.66M 0s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:    250K .......... .......... .......... .......... .......... 54% 13.3M 0s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:    300K .......... .......... .......... .......... .......... 64% 32.1M 0s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:    350K .......... .......... .......... .......... .......... 73% 30.7M 0s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:    400K .......... .......... .......... .......... .......... 82% 1.70M 0s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:    450K .......... .......... .......... .......... .......... 91% 24.6M 0s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:    500K .......... .......... .......... .......... ......    100% 28.6M=0.3s
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]: 2020-03-15 05:00:45 (1.73 MB/s) - '/tmp/lazy.txt' saved [559336/559336]
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]:
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]: --2020-03-15 05:00:45--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/video.txt
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]: Resolving adbyby.coding.net... 118.126.70.252
Sun Mar 15 05:00:45 2020 daemon.err adbyby[6737]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
[... cut ...]
Sun Mar 15 06:00:12 2020 daemon.err adbyby[26641]: --2020-03-15 06:00:12--  https://adbyby.coding.net/p/xwhyc-rules/d/xwhyc-rules/git/raw/master/lazy.txt
Sun Mar 15 06:00:12 2020 daemon.err adbyby[26641]: Resolving adbyby.coding.net... 118.126.70.252
Sun Mar 15 06:00:13 2020 daemon.err adbyby[26641]: Connecting to adbyby.coding.net|118.126.70.252|:443... connected.
Sun Mar 15 06:00:14 2020 daemon.err adbyby[26641]: HTTP request sent, awaiting response... 429
Sun Mar 15 06:00:14 2020 daemon.err adbyby[26641]: 2020-03-15 06:00:14 ERROR 429: (no description).

rogerjx87, 2020-03-15:
> 我估计是因为这个编译工具包太普及，导致所有人都是在早晨6点去连接服务器，成变相的DDOS攻击使对方服务器过载了。因为刚好我设置了路由器周日早上5点自动重启，从系统日志看，获取adbyby服务器更新一次成功，到了6点的时候adbyby模块自己更新就和楼主反馈的情况一样了。
>
> 建议把这个自动更新的时间设置成可以由用户自己修改，以避开同一时间同时去连接服务器带来的超负荷？
>
> 附带系统日志：
>
> Sun [...]

系统-计划任务里面可以自行修改更新时间。

iamggs, 2020-03-15:
> 系统-计划任务里面可以自行修改更新时间。

支持这个建议，更新时间可以开放接口自定义。

rogerjx87, 2020-03-15:
> 支持这个建议，更新时间可以开放接口自定义。

??? 系统 - 计划任务 里面，现在就可以自定义更新时间啊。可以自己修改的。

iamggs, 2020-03-15:
> ??? 系统 - 计划任务 里面，现在就可以自定义更新时间啊。可以自己修改的。

这里已修改，谢谢。

Fyred, 2020-03-16:
> > ??? 系统 - 计划任务 里面，现在就可以自定义更新时间啊。可以自己修改的。
> 
> 这里已修改，谢谢。

不建议修改计划任务，建议修改/etc/init.d/adbyby中
52行：echo '0 6 * * * /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1' >>

计划任务改了重启就没了

closed 2020-03-16 by coolsnowwolf