coolsnowwolf/lede #3797: IPV6配置问题

ertcds, 2020-03-14:
直接连接光猫可以正常获取IPV6，用这个安装后需要特殊设置吗？
试了修改网口也不行。谢谢

you-mu, 2020-03-14:
make menuconfig 时候

选择
Extra packages —> ipv6helper （选定这个后下面几项会自动选择）

Network —> odhcp6c
Network —> odhcpd-ipv6only
LuCI —> Protocols —> luci-proto-ipv6
LuCI —> Protocols —> luci-proto-ppp

ertcds, 2020-03-14:
> make menuconfig 时候
> 
> 选择
> Extra packages —> ipv6helper （选定这个后下面几项会自动选择）
> 
> Network —> odhcp6c
> Network —> odhcpd-ipv6only
> LuCI —> Protocols —> luci-proto-ipv6
> LuCI —> Protocols —> luci-proto-ppp

谢谢。我试试。

ertcds, 2020-03-14:
> make menuconfig 时候
> 
> 选择
> Extra packages —> ipv6helper （选定这个后下面几项会自动选择）
> 
> Network —> odhcp6c
> Network —> odhcpd-ipv6only
> LuCI —> Protocols —> luci-proto-ipv6
> LuCI —> Protocols —> luci-proto-ppp

[image]

是不是需要设置？lan口有ipv6地址，电脑那边没有，神奇

Aspiriniii, 2020-03-14:
光猫改桥接，用路由拨号

lengyuqingqiu, 2020-03-14:
光猫桥接，删除接口-ipv6ULA前缀，dhcp/dns-高级设置-过滤掉ipv6（AAAA)前面的勾去掉。我的直接就可以了。不过刚发现目测开启ipv6会导致adbyby的adblock host规则无法更新。

you-mu, 2020-03-15:
> 
> 
> > make menuconfig 时候
> > 选择
> > Extra packages —> ipv6helper （选定这个后下面几项会自动选择）
> > Network —> odhcp6c
> > Network —> odhcpd-ipv6only
> > LuCI —> Protocols —> luci-proto-ipv6
> > LuCI —> Protocols —> luci-proto-ppp
> 
> [image]
> 
> 是不是需要设置？lan口有ipv6地址，电脑那边没有，神奇

点开详细信息里面看看?

ertcds, 2020-03-16:
> > > make menuconfig 时候
> > > 选择
> > > Extra packages —> ipv6helper （选定这个后下面几项会自动选择）
> > > Network —> odhcp6c
> > > Network —> odhcpd-ipv6only
> > > LuCI —> Protocols —> luci-proto-ipv6
> > > LuCI —> Protocols —> luci-proto-ppp
> > 
> > 
> > [image]
> > 是不是需要设置？lan口有ipv6地址，电脑那边没有，神奇
> 
> 点开详细信息里面看看?
好多临时ipv6地址
[image]

you-mu, 2020-03-16:
> 
> 
> > > > make menuconfig 时候
> > > > 选择
> > > > Extra packages —> ipv6helper （选定这个后下面几项会自动选择）
> > > > Network —> odhcp6c
> > > > Network —> odhcpd-ipv6only
> > > > LuCI —> Protocols —> luci-proto-ipv6
> > > > LuCI —> Protocols —> luci-proto-ppp
> > > 
> > > 
> > > [image]
> > > 是不是需要设置？lan口有ipv6地址，电脑那边没有，神奇
> > 
> > 
> > 点开详细信息里面看看?
> > 好多临时ipv6地址
> > [image]

==============================

那你还有啥问题?

nftbty, 2020-03-17:
@you-mu 很明显，他的问题是获取到IPv6地址但是不能访问IPv6网络，看他的图片，网络状态里IPv6连接是“无Internet访问权限”，没有分配到IPv6地址直接是“无网络访问权限”。关于这个问题我最近也遇到，并且一直不知道问题出在哪，没法解决，本来想发个issue求助的，最近不知道是刷了R20.3.11还是开关了其他某个服务的原因，突然好了。之前也搜了一些IPv6有关的issue，有些情况是类似的，但都没什么人回复就关闭了。这里的情况感觉跟我很像，就在这借帖描述一下我的情况，让大神们分析分析是什么原因，也正好让题主看看他的问题是不是一样的。

nftbty, 2020-03-17:
我是用的联通宽带，路由器K3，在R20.1.24及之前的版本，IPv6都是正常的，后来重新编译了R20.2.15之后，R20.3.5之前的一个版本，刷上去后，就发现IPv6不能用了，路由器接口获取到了地址和前缀，电脑也获取到了地址，但是打开状态就是显示“无Internet访问权限”，ipv6-test.com测试不支持IPv6，换手机测，也是有地址但是不通，ping 240c::6666也不通，直接登路由器后台ping 240c::6666和其他v6服务器，都不通。但是从外网其他宽带或流量通过IPv6访问路由器和下面的电脑，都可以正常访问。
然后上GitHub查issues和commits，先是看到20.1.24和R20.2.15之间有一个dnsmasq禁止解析AAAA记录的更新，上luci，取消勾选禁止解析IPv6的选项，然后情况还是依旧。期间有怀疑是保留配置升级的问题，重置、进cfe重刷factory然后不保留配置升级都做过了，还是不行。刷回R20.1.24，又可以了，但是发现从路由器拨号成功到电脑显示有Internet连接有接近10秒时间，而用恩山下的factory固件（17年的lede）测试，从拨上号，几乎1秒内，电脑IPv4和IPv6连接状态就变成Internet了，测试是通的。然后又换成R20.2.15，发现刚拨号成功时，有一瞬间，IPv6显示Internet，然后马上变成“无Internet访问权限”。然后到最近刷了R20.3.11，我记得刚刷好时也测过了，不行。
我家里还有一条宽带，也是联通，路由器也是K3，用的一样的固件。上面说的是在店里。这两个地方在一个县下面的不同镇，原来这两个宽带号在不同机房，后来应该是在一个机房了，2个号在两边都能成功拨号。两边局域网通过zerotier互通。家里的一直用的R20.1.24，因为新固件IPv6还不知道什么问题，就一直没升级。之前出R20.3.5的时候，本来是要升店里面的，但是浏览器开的是家里的luci，没注意，误升了，但是发现家里的升级之后，IPv6照样能用，但店里面的就是不行。前几天店子附近联通宽带下行异常但上行没问题。有的家里下行只有2Mbps，我的还好，只降到30Mbps，同时家里的测速可以跑满。联通来人排查问题，恢复后一直没有测IPv6，不知道是否有关。因为sourceforge在我这边DNS被污染很久了，一直没上去，用流量也打不开，但那天我远程连接家里电脑发现sourceforge可以打开，但店里面还是不行，对比了两边的设置，家里的开了flowoffload里的DNS加速，店里没开。把店里面的DNS加速也开了之后，sourceforge可以上了。然后第二天无意发现电脑IPv6连接状态是Internet，然后ipv6-test.com测了一下，还真的可以了。又用手机和路由器测，都可以了。
至此，虽然问题没有了，但不知道问题出在哪里，也不知道是怎么解决的，是联通那边的问题，还是之前更新odhcp6c的问题、R20.3.11之前某个commit解决了这个问题，或者和某个软件的某个版本有冲突。

nftbty, 2020-03-17:
对了，曾经还编译过官方op-19.07加上lean的packages，IPv6也是有地址但不通。怕软件包影响，重新没改任何东西，默认配置编译，不保留配置刷机后，不配置任何其他东西，只配置PPPoE和IPv6，还是用不了，后来就放弃了，刷回R20.1.24。再之后就是R20.2.15以上版本遇到的问题和各种折腾。

ertcds, 2020-03-17:
> 对了，曾经还编译过官方op-19.07加上lean的packages，IPv6也是有地址但不通。怕软件包影响，重新没改任何东西，默认配置编译，不保留配置刷机后，不配置任何其他东西，只配置PPPoE和IPv6，还是用不了，后来就放弃了，刷回R20.1.24。再之后就是R20.2.15以上版本遇到的问题和各种折腾。

我的问题不在联通，因为我另一个固件是可以正常访问ipv6的，我尝试一下你说的固件试试。

you-mu, 2020-03-17:
> 
> 
> > 对了，曾经还编译过官方op-19.07加上lean的packages，IPv6也是有地址但不通。怕软件包影响，重新没改任何东西，默认配置编译，不保留配置刷机后，不配置任何其他东西，只配置PPPoE和IPv6，还是用不了，后来就放弃了，刷回R20.1.24。再之后就是R20.2.15以上版本遇到的问题和各种折腾。
> 
> 我的问题不在联通，因为我另一个固件是可以正常访问ipv6的，我尝试一下你说的固件试试。

那现在 ping 2400:3200::1 不通?

nftbty, 2020-03-18:
@ertcds 我遇到的问题估计也不在运营商，因为R20.1.24及以前的版本，还有我从恩山下的一个17年的lede没问题，就是官方op-19.07的不行，然后大雕的可能是更新odhcp6之后就不行了。因为官方op那边的版本肯定也是新的。我猜会不会是某个新版odhcp6引入的问题，但新版也不是都有问题，现在我这边就又可以了，不明原因的。还有个猜测，是不是更新后的版本和某些地区ISP机房的服务端存在些兼容性上的问题，比如机器固件比较老之类的，因为我这边是小地方，联通也是三家里最没钱的，IPv6的部署也搞的最慢。

ertcds, 2020-03-18:
> > > 对了，曾经还编译过官方op-19.07加上lean的packages，IPv6也是有地址但不通。怕软件包影响，重新没改任何东西，默认配置编译，不保留配置刷机后，不配置任何其他东西，只配置PPPoE和IPv6，还是用不了，后来就放弃了，刷回R20.1.24。再之后就是R20.2.15以上版本遇到的问题和各种折腾。
> > 
> > 
> > 我的问题不在联通，因为我另一个固件是可以正常访问ipv6的，我尝试一下你说的固件试试。
> 
> 那现在 ping 2400:3200::1 不通?

[image]
[image]

我删除了wlan6接口，又重新新建了一个。现在可以获取ipv6地址了，但是还是无法访问ipv6的一些网站。
我今天再查查

ertcds, 2020-03-18:
今天折腾好了，光猫拨号，最新版编译的环境
记录一下我的操作过程希望能给其他人点启发：
1：删除了wlan6口，重新建一个。有点玄学~~
2:DHCP/DNS 禁止解析 IPv6 DNS 记录去掉勾选。
@you-mu @Aspiriniii @lengyuqingqiu @nftbty 
谢谢几位。

closed 2020-03-18 by ertcds