coolsnowwolf/lede #3799: 原版openwrt的ssr plus+连接问题

nikitazep, 2020-03-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
原版openwrt安装的ssr plus+无法连网

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：通过原版openwrt单独安装的ssr plus+，大多主流外网无法连接，目前google可以连接，YouTube连接不了，有些网站会提示“你的连接不是专用连接”，本人使用Trojan协议，服务器节点连接都是正常，但【谷歌】连通性检查显示连接错误

### (2) 路由器型号和固件版本

A：路由器型号 Ubnt-ERX + 无线AP，固件OpenWrt R20.3.5

coolsnowwolf, 2020-03-14:
原版默认开 IPv6 ，自然地址被污染了，自行解决吧

lesliedh, 2020-03-14:
我用的allwinner-h5做旁路由，friendlywrt和openwrt都用ssr-plus，协议是ssr，都是可以连外网的，没有问题。就是adbyby-plus在openwrt下启动不了，也没去管它。

norikohxb, 2020-03-15:
太多人都喜欢自己去整一个oepnrt然后再来用ssrplus，才造成政府不想要让你看的网站给所谓的dns劫持了。我是用lede原汁原味自己编译的。我连加速Turbo ACC 网络加速设置都不开，所以啊，一切的根源就是在于不用别人整合好，自己弄，然后才来引入一个smartdns来所谓国内国外分开解析。证明了，问题其实人都是这么折腾，其实我在说别人的时候我自己也是，总喜欢把没有用的lean的模块给不打勾，做的过分精简了以后，每次lean大的一个更新，我有这个bug那个bug，所以其实都是折腾惹的祸

closed 2020-03-19 by coolsnowwolf