coolsnowwolf/lede #3802: ACME certs设置后无法获取证书

minfengyk, 2020-03-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
ACME certs设置后无法获取证书，保存应用后没有运行和报错，系统日志上获取不了该程序的日子。

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
我是公网IP，已经正确设置了阿里云DDNS解析，准备为阿里上的域名申请证书。但是ACME certs设置后无法获取证书，保存应用后没有运行和报错，系统日志上获取不了该程序的日志。
### (1) 具体问题
ACME certs设置后无法获取证书。我不是程序员，但是感觉luci-app-acme只是一个操作界面，并没有安装程序内核，如acme.sh。设置不知道有没有问题，请参考下图。

### (2) 路由器型号和固件版本
型号：Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz : 1 Core 2 Thread
固件：OpenWrt R20.3.11 / LuCI Master (git-20.062.50551-a6189c5)

### (3) 详细日志
获取不到程序运行日志
[image]

garypang13, 2020-03-14:
还需要 CONFIG_PACKAGE_acme-dnsapi=y
另外你的DNS API格式错了 不需要前面的export

minfengyk, 2020-03-14:
> 还需要 CONFIG_PACKAGE_acme-dnsapi=y
> 另外你的DNS API格式错了 不需要前面的export

感谢您的指点！我是小白，再请教：CONFIG_PACKAGE_acme-dnsapi=y，如何找到这个参数设置的地方？

trackycn, 2020-03-14:
> > 还需要 CONFIG_PACKAGE_acme-dnsapi=y
> > 另外你的DNS API格式错了 不需要前面的export
> 
> 感谢您的指点！我是小白，再请教：CONFIG_PACKAGE_acme-dnsapi=y，如何找到这个参数设置的地方？

在 Network 里

minfengyk, 2020-03-14:
> > > 还需要 CONFIG_PACKAGE_acme-dnsapi=y
> > > 另外你的DNS API格式错了 不需要前面的export
> > 
> > 
> > 感谢您的指点！我是小白，再请教：CONFIG_PACKAGE_acme-dnsapi=y，如何找到这个参数设置的地方？
> 
> 在 Network 里

您好！按照您的指点，将Network里的acme有关两项全部勾选，重新制作固件，并且vi .config里检查确认CONFIG_PACKAGE_acme-dnsapi=y正确设置，DNS API格式也修改了，但是问题依旧。考虑还是缺acme.sh核心。就像AdGuard Home一样，没安装核心，并且安装困难，又没有教程，使用都非常不友好，准备放弃了。

trackycn, 2020-03-15:
> 您好！按照您的指点，将Network里的acme有关两项全部勾选，重新制作固件，并且vi .config里检查确认CONFIG_PACKAGE_acme-dnsapi=y正确设置，DNS API格式也修改了，但是问题依旧。考虑还是缺acme.sh核心。就像AdGuard Home一样，没安装核心，并且安装困难，又没有教程，使用都非常不友好，准备放弃了。

我建议你手工将 acme 和 acme 里的 dnsapi 脚本更新到官方的最新版本, 源码里版本带的dnsapi好像版本比较旧, 之后dnsapi更新过api接口。

[https://github.com/acmesh-official/acme.sh](url)

minfengyk, 2020-03-15:
> > > > > 还需要CONFIG_PACKAGE_acme-dnsapi = y另外
> > > > > 你的DNS API格式错了不需要前面的export
> > > > 
> > > > 
> > > > 感谢您的指点！我是小白，再请教：CONFIG_PACKAGE_acme-dnsapi = y，如何找到这个参数设置的地方？
> > > 
> > > 
> > > 在网络里
> > 
> > 
> > 您好！按照您的指点，将网络里的acme有关全部全部重置，重新制作固件，并且vi .config里检查确认CONFIG_PACKAGE_acme-dnsapi = y正确设置，DNS API格式也被修改了，但是问题依旧。考虑还是缺acme.sh核心。就像AdGuard Home一样，没安装核心，并且安装困难，又没有教程，使用都非常不友好，准备放弃了。
> 
> 我建议你手工将acme和acme里的dnsapi脚本更新到官方的最新版本，子程序版本版本的dnsapi好像版本比较旧，之后dnsapi更新过api接口。
> 
> [https://github.com/acmesh-official/acme.sh](url)

按照您的建议，下载最新acme源码，打包，替换dl文件夹中的旧版源码，重新编译，还是获取不到证书。申请证书是为了远程访问家里的OpenWrt安全点，准备换个思路，看VPN能不能访问OpenWrt。

wspzf, 2020-03-16:
你配置少了个DNS API吧，我用的dnspod，给你参考一下
[image]

minfengyk, 2020-03-16:
> 你配置少了个DNS API吧，我用的dnspod，给你参考一下
> [image]
您上传的图片看不到，您能否抽空再上传一次，谢谢！

minfengyk, 2020-03-16:
> 你配置少了个DNS API吧，我用的dnspod，给你参考一下
> [image]

按照您的设置，保存应用，还是获取不了证书，阿里云查看没有绑定证书，本地/etc/acme也找不到证书。

wspzf, 2020-03-16:
运行一下计划任务里的acme的命令，默认是0点自动运行

minfengyk, 2020-03-16:
> 运行一下计划任务里的acme的命令，默认是0点自动运行
感谢您的指点，终于成功获取到证书。但是发现证书是下载到软路由里面，不能自动上传到阿里云绑定吗？还是需要从软路由里导出再人工上传到阿里云？

garypang13, 2020-03-17:
> > 运行一下计划任务里的acme的命令，默认是0点自动运行
> > 感谢您的指点，终于成功获取到证书。但是发现证书是下载到软路由里面，不能自动上传到阿里云绑定吗？还是需要从软路由里导出再人工上传到阿里云？

只有在申请证书的时候需要验证域名所有权,申请好的证书和阿里云已经没关系了,

wspzf, 2020-03-18:
> > 运行一下计划任务里的acme的命令，默认是0点自动运行
> > 感谢您的指点，终于成功获取到证书。但是发现证书是下载到软路由里面，不能自动上传到阿里云绑定吗？还是需要从软路由里导出再人工上传到阿里云？

为什么要上传到阿里云？不是很明白

minfengyk, 2020-03-18:
> > > 运行一下计划任务里的acme的命令，默认是0点自动运行
> > > 感谢您的指点，终于成功获取到证书。但是发现证书是下载到软路由里面，不能自动上传到阿里云绑定吗？还是需要从软路由里导出再人工上传到阿里云？
> 
> 只有在申请证书的时候需要验证域名所有权,申请好的证书和阿里云已经没关系了,

> > > 运行一下计划任务里的acme的命令，默认是0点自动运行
> > > 感谢您的指点，终于成功获取到证书。但是发现证书是下载到软路由里面，不能自动上传到阿里云绑定吗？还是需要从软路由里导出再人工上传到阿里云？
> 
> 为什么要上传到阿里云？不是很明白

是我小白了点，看到阿里云有个ssl证书管理，可以上传证书，后来才明白那是租服务器，开网站用的。从koolshare转过来的，koolshare上有个软件，申请成功后直接就可以用了。而acme还需要配合luci-app-uhttpd才能生效。要是路由器上安装证书，内网的NAS、电脑都能生效就好了，发现外网访问NAS使用加密链接的话，还需要将证书安装到NAS才行，否则3月就要安装一次，或者在NAS上也安装acme，不知道我的理解是否正确？

garypang13, 2020-03-18:
> 是我小白了点，看到阿里云有个ssl证书管理，可以上传证书，后来才明白那是租服务器，开网站用的。从koolshare转过来的，koolshare上有个软件，申请成功后直接就可以用了。而acme还需要配合luci-app-uhttpd才能生效。要是路由器上安装证书，内网的NAS、电脑都能生效就好了，发现外网访问NAS使用加密链接的话，还需要将证书安装到NAS才行，否则3月就要安装一次，或者在NAS上也 [...]

把uhttpd换成nginx,使用反向代理,就能实现内网所有设备都使用证书.

minfengyk, 2020-03-18:
> 把uhttpd换成nginx,使用反向代理,就能实现内网所有设备都使用证书.

感谢您的指点，百度了一下，介绍最多的是终端登录后台，敲代码安装，调试。对我这样的小白很不友好，出了问题也不知道怎么解决。就搜索了一下官方源里的软件包，看有没有现成的。结果很多，请教有没有能用的？源里的软件包兼容也不一定好，装个acme就导致luci界面出问题！但还是想试试看。
[image]

garypang13, 2020-03-18:
安装这个nginx-mod-luci-ssl,其他依赖会自动装上.完全没接触过nginx的话还是不建议折腾.

minfengyk, 2020-03-20:
> 安装这个nginx-mod-luci-ssl,其他依赖会自动装上.完全没接触过nginx的话还是不建议折腾.
> 或者你可以试试我自用的https://github.com/garypang13/Actions-OpenWrt-Snapshot/releases ,默认已配置好nginx

感谢！软件源的nginx-mod-luci-ssl安装不了。您的固件在我的路由上滚屏报错，进后台改network配置都困难。加上nginx配置比较复杂，这方面知识还欠缺，还是不折腾了。

garypang13, 2020-03-20:
> > 安装这个nginx-mod-luci-ssl,其他依赖会自动装上.完全没接触过nginx的话还是不建议折腾.
> > 或者你可以试试我自用的https://github.com/garypang13/Actions-OpenWrt-Snapshot/releases ,默认已配置好nginx
> 
> 感谢！软件源的nginx-mod-luci-ssl安装不了。您的固件在我的路由上滚屏报错，进后台改network配置都困难。加上nginx配置比较复杂，这方面知识还欠缺，还是不折腾了。

我也是用的i3-7100U,不过是在pve中运行,会不会是开启了uefi导致的.

boatmac, 2020-03-20:
我使用的例子，编译的时候选上了 luci-app-acme，DNS 使用的是Cloudflare：

mv /usr/lib/acme/acme.sh /usr/lib/acme/acme.sh.bak
curl https://raw.githubusercontent.com/Neilpang/acme.sh/master/acme.sh > /usr/lib/acme/acme.sh
chmod a+x /usr/lib/acme/acme.sh

opkg update
opkg install socat
curl https://raw.githubusercontent.com/Neilpang/acme.sh/master/dnsapi/dns_cf.sh > /usr/lib/acme/dnsapi/dns_cf.sh
chmod +x /usr/lib/acme/dnsapi/dns_cf.sh
/etc/init.d/acme restart

wspzf, 2020-03-21:
> > 安装这个nginx-mod-luci-ssl,其他依赖会自动装上.完全没接触过nginx的话还是不建议折腾.
> > 或者你可以试试我自用的https://github.com/garypang13/Actions-OpenWrt-Snapshot/releases ,默认已配置好nginx
> 
> 感谢！软件源的nginx-mod-luci-ssl安装不了。您的固件在我的路由上滚屏报错，进后台改network配置都困难。加上nginx配置比较复杂，这方面知识还欠缺，还是不折腾了。

用docker容器 吧,nginx:alpine 比较简单,只需要配置nginx.config就可以了

minfengyk, 2020-03-22:
> > > 安装这个nginx-mod-luci-ssl,其他依赖会自动装上.完全没接触过nginx的话还是不建议折腾.
> > > 或者你可以试试我自用的https://github.com/garypang13/Actions-OpenWrt-Snapshot/releases ,默认已配置好nginx
> > 
> > 
> > 感谢！软件源的nginx-mod-luci-ssl安装不了。您的固件在我的路由上滚屏报错，进后台改network配置都困难。加上nginx配置比较复杂，这方面知识还欠缺，还是不折腾了。
> 
> 用docker容器 吧,nginx:alpine 比较简单,只需要配置nginx.config就可以了

谢谢，我试试！

closed 2020-07-04 by coolsnowwolf