coolsnowwolf/lede #3804: x64编译失败

Yofo-me, 2020-03-14:
` 路由器版本 :x86_64 `
> 问题：
```  
checking for clock_gettime in -lrt... yes
checking µTP... yes
checking system miniupnpc library... >= 1.7
checking whether NLS is requested... yes
Unescaped left brace in regex is illegal here in regex; marked by <-- HERE in m/^(.*)\${ <-- HERE ?([A-Z_]+)}?(.*)$/ at /home/code/lede/staging_dir/hostpkg/bin/intltool-update line 986.
checking for intltool >= 0.35.0...  found
configure: error: Your intltool is too old.  You need intltool 0.35.0 or later.
Makefile:177: recipe for target '/home/code/lede/build_dir/target-x86_64_musl/transmission-openssl/transmission-2.94/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/code/lede/build_dir/target-x86_64_musl/transmission-openssl/transmission-2.94/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/code/lede/feeds/packages/net/transmission'
time: package/feeds/packages/transmission/openssl/compile#3.60#1.15#4.86
package/Makefile:111: recipe for target 'package/feeds/packages/transmission/compile' failed
make[2]: *** [package/feeds/packages/transmission/compile] Error 2
make[2]: Leaving directory '/home/code/lede'
package/Makefile:107: recipe for target '/home/code/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/code/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/code/lede'
/home/code/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

```

> initltool目前安装的是0.4.6版本，请问怎么修复？

Yofo-me, 2020-03-14:
> ` /home/code/lede/staging_dir/hostpkg/bin/intltool-update ` 986行内容：

[image]

Yofo-me, 2020-03-14:
> intltool版本：
```
code@CODE:~/lede$ intltoolize  --version 
intltoolize (GNU intltool) 0.40.6
```

[image]

Yofo-me, 2020-03-14:
> 但还是一直报这个错误

```
Your intltool is too old. You need intltool 0.35.0 or later.
```

closed 2020-07-04 by coolsnowwolf

c005machao5, 2020-11-02:
这个问题最后怎么解决的