coolsnowwolf/lede #3806: ShadowsocksR Plus+ socks5 全局模式问题

luyu6056, 2020-03-14:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我把socks5服务端，上传到路由器上，服务器端口127.0.0.1:10808，运行模式选全局，在终端运行这个socks5服务端，由于服务端需要连接到远程服务器，实现socks5代理穿透，这时候就出现尴尬的事，路由器上的socks5的外网请求，被路由器的全局透明代理拦截下来转发回到本地socks5，形成一个死循环，无法连接到远端服务器。
最后我想说的是，能否判断路由器本地ip的连接不走代理？强制所有用户要走代理是ok的。

### (2) 路由器型号和固件版本

A：竞斗云2.0,PandoraBox R20.3.8 By Lean / LuCI Master (git-19.357.21022-51cd4c9)

### (3) 详细日志

A：
[image]

coolsnowwolf, 2020-03-14:
那么这个功能是用来干嘛的呢。。。

[image]

luyu6056, 2020-03-14:
想偷懒啊，v2ray也是连远端开个本地socks5，你的插件是如何避免这种死循环请求呢？------------------ 原始邮件 ------------------

 

 

 
&mdash;

coolsnowwolf, 2020-03-14:
写入节点的 IP 被自动加入跳过列表

luyu6056, 2020-03-14:
好吧,等下我找找改哪个文件，把我节点ip加进去------------------ 原始邮件 ------------------

coolsnowwolf, 2020-03-14:
明明可以配置，你还多此一举

closed 2020-03-14 by luyu6056