coolsnowwolf/lede #3809: 编译v2ray-plugin时失败

Zack0617, 2020-03-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
本地编译最新固件，如果不勾选v2ray-plugin可以编译成功，如果勾选则编译失败，看日志是golang的问题，试了很久，不知道如何解决
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：部分日志如下：
SHELL= flock /home/zack/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zack/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.pkgdir/v2ray/. /home/zack/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zack/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.v2ray_installed
echo "v2ray" >> /home/zack/lede/staging_dir/target-x86_64_musl/pkginfo/v2ray.default.install
make[3]: Leaving directory '/home/zack/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#0.23#0.29#0.45
make[3]: Entering directory '/home/zack/lede/package/lean/v2ray-plugin'
rm -f /home/zack/lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0/.built
touch /home/zack/lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0/.built_check
( cd /home/zack/lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0/.go_work/build ; export            GOOS=linux GOARCH=amd64 GO386= GOARM= GOMIPS= GOMIPS64= CGO_ENABLED=1 CC=x86_64-openwrt-linux-musl-gcc CXX=x86_64-openwrt-linux-musl-g++ CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/zack/lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0:v2ray-plugin-1.3.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/zack/lede/staging_dir/target-x86_64_musl/usr/include -I/home/zack/lede/staging_dir/target-x86_64_musl/include -I/home/zack/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/zack/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/zack/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/zack/lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0:v2ray-plugin-1.3.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/zack/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/zack/lede/staging_dir/target-x86_64_musl/lib -L/home/zack/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/lib -L/home/zack/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib -znow -zrelro"   GOPATH=/home/zack/lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0/.go_work/build GOCACHE=/home/zack/lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0/.go_work/cache GOENV=off ; echo "Finding targets" ; targets=$(go list github.com/shadowsocks/v2ray-plugin/...) ; for pattern in ; do targets=$(echo "$targets" | grep -v "$pattern") ; done ; echo ; if [ "" = 1 ]; then echo "Calling go generate" ; go generate -v  $targets ; echo ; fi ; if [ "" != 1 ]; then echo "Building targets" ; case amd64 in arm)             installsuffix="v" ;; mips|mipsle)     installsuffix="" ;; mips64|mips64le) installsuffix="" ;; esac ; ldflags="-linkmode external -extldflags '-L/home/zack/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/zack/lede/staging_dir/target-x86_64_musl/lib -L/home/zack/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/lib -L/home/zack/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib -Wl,-z,now -Wl,-z,relro'" ; pkg_gcflags="" ; pkg_ldflags="-s -w" ; for def in ; do pkg_ldflags="$pkg_ldflags -X $def" ; done ; go install  ${installsuffix:+-installsuffix $installsuffix} -trimpath -ldflags "all=$ldflags" -v ${pkg_gcflags:+-gcflags "$pkg_gcflags"} ${pkg_ldflags:+-ldflags "$pkg_ldflags $ldflags"}  $targets ; retval=$? ; echo ; if [ "$retval" -eq 0 ] && [ -z "$(find /home/zack/lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0/.go_work/build/bin -maxdepth 0 -type d \! -empty 2>/dev/null)" ]; then echo "No binaries were generated, consider adding GO_PKG_SOURCE_ONLY:=1 to Makefile" ; echo ; fi ; echo "Cleaning module download cache (golang/go#27455)" ; go clean -modcache ; echo ; fi ; exit $retval ; )
Finding targets
go: golang.org/x/net@v0.0.0-20190619014844-b5b0513f8c1b: Get "https://proxy.golang.org/golang.org/x/net/@v/v0.0.0-20190619014844-b5b0513f8c1b.mod": EOF

Building targets
go: github.com/v2ray/v2ray-core@v4.22.1+incompatible: Get "https://proxy.golang.org/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod": EOF

Cleaning module download cache (golang/go#27455)

make[3]: *** [Makefile:62: /home/zack/lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0/.built] Error 1
make[3]: Leaving directory '/home/zack/lede/package/lean/v2ray-plugin'
time: package/lean/v2ray-plugin/compile#0.62#0.95#3.86
make[2]: *** [package/Makefile:113: package/lean/v2ray-plugin/compile] Error 2
make[2]: Leaving directory '/home/zack/lede'
make[1]: *** [package/Makefile:107: /home/zack/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zack/lede'
make: *** [/home/zack/lede/include/toplevel.mk:227: world] Error 2

Zack0617, 2020-03-14:
云编译没问题，本地编译一直通不过

haxc, 2020-03-15:
架梯子编译就可以了

closed 2020-03-15 by coolsnowwolf