coolsnowwolf/lede #3811: 开启 IPv6 内网机器名解析成外网 IPv6了

tanst, 2020-03-14:
开启 IPv6 以后，使用smb下载群晖上的文件，发现速度只有3MB/S，和外网上传速度一致，检测解析发现解析成公网 IPv6 了。

```
Server:		127.0.0.1
Address:	127.0.0.1#53

*** Can't find DS3617: No answer
Name:      DS3617
Address 1: 240e:898:4c0:9d10::5
Address 2: fdd5:2ee5:eb3f::5
```

能不能在不关闭 IPv6 解析的情况下优先解析 IPv4 地址或优先返回内网 IPv6 地址。

3mile, 2020-03-15:
群晖改MTU 2000试试
对我而言，有效
[来源点此](http://www.gebi1.cn/thread-292338-1-1.html)

dongjuanyong, 2020-03-24:
你群晖是固定ip还是dhcp（在路由的dhcp中保留ip也算dhcp）
你要是固定ip的话很正常，路由解析地址是从自己dhcp表里读的，v4固定了没有，v6走的dhcpv6，自然就返回这个地址
不过话说回来，你的路由表没错的话，及时访问240e的电信global地址，路由也不会傻傻的绕外网一圈的（ping<=1ms）

tanst, 2020-03-24:
> 你群晖是固定ip还是dhcp（在路由的dhcp中保留ip也算dhcp）
> 你要是固定ip的话很正常，路由解析地址是从自己dhcp表里读的，v4固定了没有，v6走的dhcpv6，自然就返回这个地址
> 不过话说回来，你的路由表没错的话，及时访问240e的电信global地址，路由也不会傻傻的绕外网一圈的（ping<=1ms）

感谢回答，我的群晖的确是固定IP，但是我在 dhcp 分配里面也是分配的一样的IP，至于路由表，请教一下如何排查呢？

tanst, 2020-03-24:
> 你群晖是固定ip还是dhcp（在路由的dhcp中保留ip也算dhcp）
> 你要是固定ip的话很正常，路由解析地址是从自己dhcp表里读的，v4固定了没有，v6走的dhcpv6，自然就返回这个地址
> 不过话说回来，你的路由表没错的话，及时访问240e的电信global地址，路由也不会傻傻的绕外网一圈的（ping<=1ms）

我刚刚试了一下，默认 ping DS3617 返回的是240 开头的 ipv6 地址，ping<=1ms，但是即使这样访问群晖 DS3617 的 smb 共享速度仍然只有 2MB/S 左右，换成 ipv4 格式的 smb 共享复制电影到本地有 110MB/S 左右。

tanst, 2020-03-24:
刚刚又试了一下用 ipv6 地址直接访问 smb (IPV6地址已处理)

```
\\240e-312-4c8-2ac0--5.ipv6-literal.net\Media1\电影\The.Fly
\\fdf9-b9c7-cc6e--5.ipv6-literal.net\Media1\电影\The.Fly
```

发现内网 ipv6 外网 ipv6 速度均只有 1.5MB/S 左右。
也就是说现在以下3种方式访问 smb 共享速度均只有 1.5MB/S 左右。

```
\\240e-312-4c8-2ac0--5.ipv6-literal.net\Media1\电影\The.Fly
\\fdf9-b9c7-cc6e--5.ipv6-literal.net\Media1\电影\The.Fly
\\DS3617\Media1\电影\The.Fly
···

只有 `\\192.168.1.5\Media1\电影\The.Fly`  速度能达到 110MB/S

客户端是 WIN10 64位系统

dongjuanyong, 2020-03-25:
群晖的锅，看看是不是有访问限制（限速）
ping 1ms的话路由肯定没绕出去，不信的话tracert也能知道
这样一下只剩下客户端、交换机和群晖了，v4正常、v6不正常只能是群晖的锅

closed 2020-03-25 by coolsnowwolf