coolsnowwolf/lede #3812: lede版本20200314，v2ray h2+tls 的allowsecure 勾选没有在后台配置体现出来

flyhigao, 2020-03-14:
lede x86_64版本20200314，v2ray h2+tls 的allowsecure勾选没有在后台配置体现出来
我发现，如果我不选界面上这项，那么后台/etc/config/shadowsocksr中存在
option insecure '0'
而如果我界面上勾选了allowsecure ， 后台/etc/config/shadowsocksr
就不存在option insecure  这一行，从而
导致v2ray的运行配置文件/var/etc/v2-ssr-retcp.json中的
"allowInsecure": false   #不论界面是否选，这里始终都是false。
这就连不上了。
我的服务器是正常的，因为我手动改动"allowInsecure": true ，手动重启v2ray就正常使用了。

flyhigao, 2020-03-14:
[image]
就是这个选项，不论是否选，/var/etc/v2-ssr-retcp.json都是："allowInsecure": false

yonggedebaqi, 2020-03-15:
我也遇到同样问题，不设置TLS Host
就可以上网

coolsnowwolf, 2020-03-15:
这里有个逻辑上的问题。。。就是都启用了 TLS 了， 为啥不验证证书呢。。。？

coolsnowwolf, 2020-03-15:
目前默认的逻辑就是 V2RAY 的官方推荐配置

coolsnowwolf, 2020-03-15:
推送了commit 了，试试看

flyhigao, 2020-03-16:
好的

closed 2020-03-16 by flyhigao