coolsnowwolf/lede #3813: 奇怪的CPU占用率

ricksuzade-maker, 2020-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用top命令查看CPU占用率仅有1%，但是路由器Luci界面显示占用率高达25%以上，实际网速也受到很大影响。

### (2) 路由器型号和固件版本

A：3月15日最新版本编译，使用了mwan多拨，开启dns和bbr加速。硬件是i5-7200u，esxi虚拟机。

### (3) 详细日志

Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.623645] ata29: SATA max UDMA/133 abar m4096@0xfd5ff000 port 0xfd5fff00 irq 26
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.623788] ata30: SATA max UDMA/133 abar m4096@0xfd5ff000 port 0xfd5fff80 irq 26
Sun Mar 15 08:30:10 2020 kern.debug kernel: [    0.623976] ata_piix 0000:00:07.1: version 2.13
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.624410] scsi host30: ata_piix
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.624811] scsi host31: ata_piix
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.624929] ata31: PATA max UDMA/33 cmd 0x1f0 ctl 0x3f6 bmdma 0x1060 irq 14
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.625086] ata32: PATA max UDMA/33 cmd 0x170 ctl 0x376 bmdma 0x1068 irq 15
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.625430] VMware vmxnet3 virtual NIC driver - version 1.4.16.0-k-NAPI
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.625619] vmxnet3 0000:1b:00.0: # of Tx queues : 4, # of Rx queues : 4
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.626756] vmxnet3 0000:1b:00.0 eth0: NIC Link is Up 10000 Mbps
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.626903] hv_vmbus: registering driver hv_netvsc
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.627169] Fusion MPT base driver 3.04.20
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.627253] Copyright (c) 1999-2008 LSI Corporation
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.627352] Fusion MPT SPI Host driver 3.04.20
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.627475] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.627608] ehci-pci: EHCI PCI platform driver
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.627749] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.627908] ohci-pci: OHCI PCI platform driver
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.628008] ohci-platform: OHCI generic platform driver
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.628125] uhci_hcd: USB Universal Host Controller Interface driver
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.628550] xhci_hcd 0000:04:00.0: xHCI Host Controller
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.628657] xhci_hcd 0000:04:00.0: new USB bus registered, assigned bus number 1
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.629136] xhci_hcd 0000:04:00.0: hcc params 0x0388f081 hci version 0x100 quirks 0x0000000000000010
Sun Mar 15 08:30:10 2020 kern.debug kernel: [    0.629392] xhci_hcd 0000:04:00.0: cache line size of 64 is not supported
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.630238] hub 1-0:1.0: USB hub found
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.630364] hub 1-0:1.0: 4 ports detected
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.630739] xhci_hcd 0000:04:00.0: xHCI Host Controller
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.630844] xhci_hcd 0000:04:00.0: new USB bus registered, assigned bus number 2
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.630987] xhci_hcd 0000:04:00.0: Host supports USB 3.0 SuperSpeed
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.631262] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] hub 2-0:1.0: USB hub found
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] hub 2-0:1.0: 4 ports detected
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] usbcore: registered new interface driver usb-storage
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] i8042: PNP: PS/2 Controller [PNP0303:KBC,PNP0f13:MOUS] at 0x60,0x64 irq 1,12
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] serio: i8042 KBD port at 0x60,0x64 irq 1
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] serio: i8042 AUX port at 0x60,0x64 irq 12
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] hv_vmbus: registering driver hyperv_keyboard
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] rtc_cmos 00:01: registered as rtc0
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] rtc_cmos 00:01: alarms up to one month, y3k, 114 bytes nvram, hpet irqs
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] sdhci: Secure Digital Host Controller Interface driver
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] sdhci: Copyright(c) Pierre Ossman
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] hidraw: raw HID events driver (C) Jiri Kosina
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] hv_vmbus: registering driver hid_hyperv
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] usbcore: registered new interface driver usbhid
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] usbhid: USB HID core driver
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] hv_utils: Registering HyperV Utility Driver
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.639343] hv_vmbus: registering driver hv_util
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.672679] hv_vmbus: registering driver hv_balloon
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.682556] NET: Registered protocol family 10
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.682556] Segment Routing with IPv6
Sun Mar 15 08:30:10 2020 kern.info kernel: [    0.682556] NET: Registered protocol family 17
[... 623 lines cut ...]
Sun Mar 15 08:30:47 2020 cron.info crond[9771]: crond (busybox 1.30.0) started, log level 5
Sun Mar 15 08:30:51 2020 daemon.err uhttpd[3009]: cut: standard output: Broken pipe
Sun Mar 15 08:30:58 2020 user.notice mwan3[10184]: Execute ifup event on interface wan (pppoe-wan)
Sun Mar 15 08:30:58 2020 user.notice mwan3[10184]: Starting tracker on interface wan (pppoe-wan)
Sun Mar 15 08:30:58 2020 user.info mwan3rtmon[5192]: Detect rtchange event.
Sun Mar 15 08:30:58 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sun Mar 15 08:30:59 2020 cron.info crond[12349]: crond (busybox 1.30.0) started, log level 5
Sun Mar 15 08:30:59 2020 user.notice ddns-scripts[12755]: myddns_ipv4: PID '12755' started at 2020-03-15 08:30
Sun Mar 15 08:31:00 2020 user.warn ddns-scripts[12755]: myddns_ipv4: Service section disabled! - TERMINATE
Sun Mar 15 08:31:00 2020 user.warn ddns-scripts[12755]: myddns_ipv4: PID '12755' exit WITH ERROR '1' at 2020-03-15 08:31
Sun Mar 15 08:31:02 2020 daemon.err uhttpd[3009]: cut: standard output: Broken pipe
Sun Mar 15 08:31:10 2020 user.info mwan3rtmon[5192]: Detect rtchange event.
Sun Mar 15 08:31:10 2020 user.notice firewall: Reloading firewall due to ifup of vwan1_6 (pppoe-vwan1)
Sun Mar 15 08:31:10 2020 cron.info crond[15863]: crond (busybox 1.30.0) started, log level 5
Sun Mar 15 08:31:20 2020 daemon.info pppd[4640]: System time change detected.
Sun Mar 15 08:31:20 2020 daemon.info pppd[4644]: System time change detected.
Sun Mar 15 08:31:21 2020 user.info mwan3rtmon[5192]: Detect rtchange event.
Sun Mar 15 08:31:21 2020 user.notice firewall: Reloading firewall due to ifup of wan2_6 (pppoe-wan2)
Sun Mar 15 08:31:22 2020 cron.info crond[17294]: crond (busybox 1.30.0) started, log level 5
Sun Mar 15 08:31:33 2020 user.info mwan3rtmon[5192]: Detect rtchange event.
Sun Mar 15 08:31:33 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sun Mar 15 08:31:33 2020 cron.info crond[18396]: crond (busybox 1.30.0) started, log level 5
Sun Mar 15 08:32:19 2020 daemon.err /usr/bin/ss-redir[4863]: server recv: Connection reset by peer
Sun Mar 15 08:34:20 2020 authpriv.info dropbear[6512]: Child connection from 192.168.1.60:5791
Sun Mar 15 08:34:20 2020 authpriv.notice dropbear[6512]: Password auth succeeded for 'root' from 192.168.1.60:5791

ricksuzade-maker, 2020-03-15:
[image]
[image]

ricksuzade-maker, 2020-03-15:
经过排查，问题似乎出在多拨上，只要一开多拨，CPU占用率就超高，系统变得非常卡。

DongfeiSay, 2020-03-15:
新版问题挺多，要说为啥咱也不知道，选的插件就没变过，闲置的小米mini刷进最新源码的固件进luci cpu使用率居然高达70~99% 操作极其卡顿缓慢 socks server速率70mbps+降到30+ 开启多拨直接降到10+ 新添加的socks用户认证我也不知道是不是个摆设 不完善，设置了用户认证后我是不知道该要怎么连上了，chrome Proxy SwitchyOmega就说浏览器不支持用户认证，tm里倒是设不设密码都能连上？直接滚回去4号的版本一切正常了233 唯一感到有提升的就是mt7620 wifi算是到勉强能用的程度了 之前的话直接禁用2.4gwifi

DongfeiSay, 2020-03-15:
> 新版问题挺多，要说为啥咱也不知道，选的插件就没变过(只少不多)，闲置的小米mini刷进最新源码的固件进luci cpu使用率居然高达70~99% 操作极其卡顿缓慢 socks server速率70mbps+降到30+ 开启多拨直接降到10+ 新添加的socks用户认证我也不知道是不是个摆设 不完善，设置了用户认证后我是不知道该要怎么连上了，chrome Proxy SwitchyOmega就说浏览器不支持用户认证，tm里倒是设不设密码都能连上？直接滚回去4号的版本一切正常了233 唯一感到有提升的就是mt7620 wifi算是到勉强能用的程度了 之前的话直接禁用2.4gwifi

tone88, 2020-03-15:
我也遇到CPU占用异常的问题，在不登陆LUCI的时候CPU占用正常，只要登陆LUCI，CPU占用立马上来，在概况页面的时候占用是最高的，切换到其它界面的时候CPU占用率会出现波动，之前的版本并没有这样的问题。

htop看不出有什么进程存在高CPU占用率的情况。。

下面3张图是在ESXI里的截图，在没有登陆LUCI的时候一般占用是1.几，当登陆后占用至少都是10多，特别是在概况页面时占用最高。
[image]
[image]
[image]

tone88, 2020-03-15:
补充一下，刚刚手机用Chrome登陆也存在同样的问题

sirfromprc, 2020-03-15:
autocore里面的cpuinfo和ethinfo的原因

EnnawYang, 2020-03-15:
luci占用高正常 关闭自动刷新

ricksuzade-maker, 2020-03-15:
> luci占用高正常 关闭自动刷新

应该不是吧，关了浏览器界面，资源占用依然很高，ESXi界面看得到，非常非常高。而且网速也受到影响，千兆宽带都跑不满了，之前版本千兆跑满也没这么高的占用率。

georip, 2020-03-15:
每次重启一下ADBYBY就可以了。

ricksuzade-maker, 2020-03-15:
> 每次重启一下ADBYBY就可以了。

adbyby关掉也没什么变化

tone88, 2020-03-15:
> luci占用高正常 关闭自动刷新

luci有占用是正常，但是之前的版本占用没有这么突出，要不然也不会被明显的感觉到了。

georip, 2020-03-15:
adbyby非常占用CPU，特别是重启后。我这每次重启路由器以后，手动停用ADBYBY,然后再开启即可。目前3月9号的自编固件算正常，之前的和之后的都不行。。。

tone88, 2020-03-15:
> > luci占用高正常 关闭自动刷新
> 
> 应该不是吧，关了浏览器界面，资源占用依然很高，ESXi界面看得到，非常非常高。而且网速也受到影响，千兆宽带都跑不满了，之前版本千兆跑满也没这么高的占用率。

我是关闭luci后占用就正常了。奇怪

georip, 2020-03-15:
在web查看下进程，就能看到哪些进程占用CPU。

panther706, 2020-03-15:
确实CPU占用高，界面操作卡顿，感觉是升级了内核开始就这样。

sososoyoung, 2020-03-15:
竟斗云也是，我开了多拨，top 命令偶尔一个 lua 命令占cpu 挺高，但是top 中的进程都是0%，负载还是50%+

mjyhj, 2020-03-15:
我也是，待机CPU一直占用20-25%，频率都在1000-1200Mhz，以前600-800。
这一个月内开始的

dolphinpaopao, 2020-03-15:
[image]
最新版，并未出现异常占用情况。怀疑是添加了其他引入的ipk，或者有功能冲突。

mcucore, 2020-03-15:
今天编译的，只加了IPERF3和PASSWALL，CPU占用异常高，60%+！系统进程里没发现CPU占用高的。
刷之前是3月8日版本，没有这问题！

dolphinpaopao, 2020-03-15:
> 今天编译的X86，只默认再加了IPERF3和PASSWALL，CPU占用异常高，60%+！系统进程里没发现CPU占用高的。
> 刷之前是3月8日版本，没有这问题！

引入了本项目外的功能，首先排除是否是这个导致的问题。我一直都是使用原版编译，很少出现问题。

baygixu, 2020-03-15:
为什么也有could not open lease file miniupnp啊，这个错误需要解决吗

kl9800, 2020-03-15:
感觉是luci页面上数据的问题?
我用netdata看，cpu一直0.3-0.4,
在luci页面看，就是20%左右，不知道怎么回事

kl9800, 2020-03-15:
> 感觉是luci页面上数据的问题?
> 我用netdata看，cpu一直0.3-0.4,
> 在luci页面看，就是20%左右，不知道怎么回事

不对。。。用web版的netdata看，也是20%左右，手机端上后台看netdata，就是0.3-0.4

ricksuzade-maker, 2020-03-15:
> > 感觉是luci页面上数据的问题?
> > 我用netdata看，cpu一直0.3-0.4,
> > 在luci页面看，就是20%左右，不知道怎么回事
> 
> 不对。。。用web版的netdata看，也是20%左右，手机端上后台看netdata，就是0.3-0.4

不是的，我是用的esxi虚拟机，虚拟机资源监测发现CPU占用率确实很高，几乎把分配的CPU时间全部用完了，而且千兆网速也受到影响，最高只能跑到700兆了，整个系统显得非常卡，步履蹒跚。

kl9800, 2020-03-15:
> > > 感觉是luci页面上数据的问题?
> > > 我用netdata看，cpu一直0.3-0.4,
> > > 在luci页面看，就是20%左右，不知道怎么回事
> > 
> > 
> > 不对。。。用web版的netdata看，也是20%左右，手机端上后台看netdata，就是0.3-0.4
> 
> 不是的，我是用的esxi虚拟机，虚拟机资源监测发现CPU占用率确实很高，几乎把分配的CPU时间全部用完了，而且千兆网速也受到影响，最高只能跑到700兆了，整个系统显得非常卡，步履蹒跚。

会不会是新内核闹的呀？

Aes64X, 2020-03-15:
不知道你们都加了些什么玩意。。。。。我也没发现这问题。

lengyuqingqiu, 2020-03-15:
我也一样，今天新编译的固件X86，没虚拟机，top没问题，进luci以后cpu占用直接上30

ricksuzade-maker, 2020-03-15:
> 不知道你们都加了些什么玩意。。。。。我也没发现这问题。

加了mwan，syncdial，proto-bonding，ipv6，就这几个，都是大雕源码包含的内容，没有任何第三方包了。

dolphinpaopao, 2020-03-15:
> > 不知道你们都加了些什么玩意。。。。。我也没发现这问题。
> 
> 加了mwan，syncdial，proto-bonding，ipv6，就这几个，都是大雕源码包含的内容，没有任何第三方包了。

试试Actions里面的.img是否会出现相同问题。目前物理软路由使用新固件没有遇到，待会儿我再全新编译一个测试一下。

EstFoxs, 2020-03-15:
LUCI页面CPU偏高，htop基本维持5%一下

kl9800, 2020-03-15:
> htop

还真是。。。感觉这个问题可以忽略了，反正我是不会在乎。。。。毕竟也不会天天打开那个看哈哈

zhangdp, 2020-03-15:
[image]
[image]
同样发现这个问题，没任何任务流量在走的情况下，单独打开netdata页面cpu占用0.3%左右，在管理界面打开“实时监控”或者“实时信息”cpu使用率或者负载都偏高，而且一直偏高，直到管理界面关闭。早上刚编译的最新发现这样，之前的版本不会。

kl9800, 2020-03-15:
我刷回了大概3月7,8,9号编译的版本，
然后这么操作，cpu大概10%左右，再往前就没注意过有没有这个问题了，也懒得刷了…

n0way, 2020-03-15:
以为只有我存在这个问题 软路由都快融了 退回到3.3版本了 开启dns 加速就卡

tty228, 2020-03-16:
> > > luci占用高正常 关闭自动刷新
> > 
> > 
> > 应该不是吧，关了浏览器界面，资源占用依然很高，ESXi界面看得到，非常非常高。而且网速也受到影响，千兆宽带都跑不满了，之前版本千兆跑满也没这么高的占用率。
> 
> 我是关闭luci后占用就正常了。奇怪

因为 luci 占用特别高的问题，我升级了 luci 19……不过兼容性有点蛋疼

coolsnowwolf, 2020-03-16:
编译 mwan3 就会这样，删除即可

lengyuqingqiu, 2020-03-16:
> 编译 mwan3 就会这样，删除即可

大神，我的编译里都没有mwan3。但是卸载掉autocore就好了。大神有时间了麻烦看一下吧。

lengyuqingqiu, 2020-03-16:
> 编译 mwan3 就会这样，删除即可

我编译的时候没选mwan3，现在也是一进luci就cpu使用率飙升。

elonj, 2020-03-16:
遇到同样问题。换回R20.3.11的commit, 内核4.19.106没有问题

nono8273, 2020-03-17:
> 遇到同样问题。换回R20.3.11的commit, 内核4.19.106没有问题

108内核也没问题，不知道问题出在哪

hiliuyj, 2020-03-17:
咱退回到3月4日后才行

kl9800, 2020-03-17:
> 编译 mwan3 就会这样，删除即可

如果选了多拨功能，就默认选中mwan3，是不是近期不能使用多拨功能了？

nono8273, 2020-03-17:
> > 编译 mwan3 就会这样，删除即可
> 
> 如果选了多拨功能，就默认选中mwan3，是不是近期不能使用多拨功能了？

我去掉了多mwan占用率还是20+

elonj, 2020-03-17:
> > > 编译 mwan3 就会这样，删除即可
> > 
> > 
> > 如果选了多拨功能，就默认选中mwan3，是不是近期不能使用多拨功能了？
> 
> 我去掉了多mwan占用率还是20+

我也没有用syncdial和mwan，最后的编译cpu占用高，R20.3.11的commit没有问题

ricksuzade-maker, 2020-03-17:
经过测试，3月17日的固件，似乎CPU高占用的问题已经解决。但是又出现一个新的问题，就是不开SSR系统尚正常，可是一打开SSR，luci界面响应就变得极慢，而且SSR完全无法使用。

panther706, 2020-03-19:
昨天编译了3月18日的源码固件，发现还是老问题，实在无法忍受路由器操作界面的卡顿，反应慢，CPU占用高的问题。想退回以前的版本，本人小白，于是网上查找退回方法，终于找到。具体操作如下：
1，git pull，下载全新的源下，把LEAN文件夹复制出来备用
2，git reset --hard f9cb999，退回以前的版本。
3，再把全新的LEAN文件夹复制到package，这样就可以用新版插件，不知道不复制行不行，没试过。
4，再重新编译，我只选了，luci-app-accesscontrol，luci-app-adbyby-plus，luci-app-aria2，luci-app-ddns，luci-app-filetransfer，luci-app-hd-idle，luci-app-upnp，luci-app-usb-printer，luci-app-sfe，ipv6helper，automount，autosamba，科学上网。其它默认全去掉。
编译完成重新刷入，进入路由器界面，呵呵，终于找回初恋的感觉，操作行云流水，基本都是秒开（比如打开科学上网新版本要等上那么几秒），老版本那感觉爽得不要不要的，呵呵。
或者我用的是WNDR4300，对4.9的字眼比较敏感（因为一直在用4.9的内核），往上看到linux: backport kernel 4.9 support for crypto.mk，我对代码一窃不通，但看到这个英文大概知道，对4.9内核进行了什么支持，于是又用刚学会的退回命令回到git reset --hard 09f1cf4 回到这里重新编译刷入，悲剧再次上演，那蜗牛般的界面反应又回来了，唉，于是又刷回了git reset --hard f9cb999的版本。

我分析应该是f9cb999到09f1cf4之间的源码修改了才导致出问题了，或者是增加了一些功能导致操作卡顿，反应慢，CPU占用高。但无奈对代码不懂，无法解决。另外L大的新插件更新的没有问题，因为我用老版本，新插件是没有任何卡顿和CPU占用高的情况。还有就是f9cb999和09f1cf4同样是4.9.214内核，也证明了卡顿，CPU占用高跟内核版本无关。

以上是我发现的一点点资料，希望各位会看代码的看看f9cb999到09f1cf4之间的源码是不是出问题。如何没有问题就当做个案吧。

hcym, 2020-03-19:
没什么感觉，只一个ss+，官方master编译的界面经常出事

Clackz, 2020-03-19:
目前我这边是SSR的问题，启用SSR一段时间后CPU使用率就开始提高，网速降低明显，并且SSR中打开广告屏蔽和关闭广告屏蔽也有明显的速度差别

panther706, 2020-03-19:
我是AD和科学上网一直开，老版本新插件，用着不错

hcym, 2020-03-19:
今天用7800有点感觉，什么都没开也有0.78，然后慢慢恢复正常，不管了，难得用用

ricksuzade-maker, 2020-03-19:
目前的问题就是一开SSR，界面就变得奇卡无比，而且还没法上网。

geerda123, 2020-03-19:
@coolsnowwolf  不是mwan3问题，最新几个版本都这样，默认插件，不开ssr，mwan3等，只要进入luci就会20%以上占用率，无论哪个CPU。用top进后台不用luci查看，只有5%以下。大家都反映这个问题

dolphinpaopao, 2020-03-19:
> @coolsnowwolf 不是mwan3问题，最新几个版本都这样，默认插件，不开ssr，mwan3等，只要进入luci就会20%以上占用率，无论哪个CPU。用top进后台不用luci查看，只有5%以下。大家都反映这个问题

假如没有详细信息作者无法复现的话，反应上天也解决不了的。因为各自使用环境差异太大。用Actions里面自动默认编译的固件试一下是否有相同情况出现。目前我物理机使用没有出现异常。

geerda123, 2020-03-19:
看别人的日志也行，我也是actions跑的

dolphinpaopao, 2020-03-19:
> […](#)

本项目的Actions里面自动生成的固件，或者自己默认编译不修改配置文件。这样方便输出信息定位问题。

jxph1123, 2020-03-20:
现在最新版解决了吗

geerda123, 2020-03-20:
暂时无解，不进luci就没事，对付用吧

xjboss, 2020-03-21:
我这LUCI同样有占吃的问题，我是歌华链路由，装了MWAN没有$$R

ricksuzade-maker, 2020-03-21:
> 我这LUCI同样有吃CPU的问题，我是歌华链路由，装了MWAN没有$$R

似乎最新版本的x86编译，已经解决这个问题，我目前用起来挺正常。就是adbyby规则显示0条，除此以外没有别的问题了。

[11 more replies]

closed 2020-03-25 by coolsnowwolf