coolsnowwolf/lede #3816: 今天很奇怪，多次系统宕机，需要热重启

AmibaWorld, 2020-03-15:
UBNT-ERX
OpenWrt R20.3.11 / LuCI Master (git-20.062.50551-a6189c5)

用着突然系统宕机，无法连接管理192.168.1.1

热重启，然后又正常了
log

Sun Mar 15 15:26:54 2020 user.info e2guardian[3530]: Started sucessfully.
Sun Mar 15 15:26:55 2020 daemon.info pdnsd[3901]: pdnsd-1.2.9b-par starting.
Sun Mar 15 15:26:55 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:26:55 2020 daemon.info pppd[1569]: System time change detected.
Sun Mar 15 15:26:56 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Mar 15 15:26:56 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Mar 15 15:27:05 2020 daemon.info procd: Instance unbound::unbound s in a crash loop 6 crashes, 0 seconds since last crash
Sun Mar 15 15:27:10 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:27:11 2020 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0.2)
Sun Mar 15 15:27:13 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:27:14 2020 user.notice ddns-scripts[4307]: myddns_ipv6: PID '4307' started at 2020-03-15 15:27
Sun Mar 15 15:27:17 2020 user.warn ddns-scripts[4307]: myddns_ipv6: Service section disabled! - TERMINATE
Sun Mar 15 15:27:17 2020 user.warn ddns-scripts[4307]: myddns_ipv6: PID '4307' exit WITH ERROR '1' at 2020-03-15 15:27
Sun Mar 15 15:27:23 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sun Mar 15 15:27:24 2020 user.notice ddns-scripts[5687]: myddns_ipv4: PID '5687' started at 2020-03-15 15:27
Sun Mar 15 15:27:24 2020 user.warn ddns-scripts[5687]: myddns_ipv4: Service section disabled! - TERMINATE
Sun Mar 15 15:27:24 2020 user.warn ddns-scripts[5687]: myddns_ipv4: PID '5687' exit WITH ERROR '1' at 2020-03-15 15:27
Sun Mar 15 15:27:28 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:27:34 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:27:34 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sun Mar 15 15:27:49 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:27:49 2020 cron.err crond[3565]: time disparity of 145 minutes detected
Sun Mar 15 15:27:54 2020 daemon.info procd: Instance unbound::unbound s in a crash loop 6 crashes, 0 seconds since last crash
Sun Mar 15 15:27:58 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:28:13 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:28:25 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:28:40 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:28:55 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:29:10 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:29:28 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:29:43 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:30:04 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:30:20 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:30:44 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:30:59 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:31:26 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:31:41 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:32:11 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:32:26 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates
Sun Mar 15 15:32:59 2020 daemon.info dnscrypt-proxy[2491]: dnscrypt-proxy Refetching server certificates
Sun Mar 15 15:33:14 2020 daemon.err dnscrypt-proxy[2491]: dnscrypt-proxy Unable to retrieve server certificates

kl9800, 2020-03-15:
我前天也出现过这个，重新编译了一下，把dnscrypt-proxy与dnsforwarder全去掉了，目前还未发现问题，还得再观察几天

AmibaWorld, 2020-03-16:
试试，dnscrypt-proxy dnsforwarder 应该用的不多

tmpaxx, 2020-03-19:
我也是ERX 3月9号之后编译的几个固件 也会有使用一段时间后无法上网，ssh登录重启无效，需要拔电源。

jdjingdian, 2020-03-20:
您好，我的小娱C1也是7621芯片，在更新3.13版本（mt7621: TCP and UDP traffic is forwarded correctly by HWNAT (1% CPU usage））后不定时重启，现在关了hwnat正常运行，你们要不要看看是不是hwnat的问题？

AmibaWorld, 2020-03-21:
我的是管理后台也进不去，因为获取不到正确ip，连不上，正在刷机，问题是按照步骤刷机后仍然无法解决，正在试图刷原厂固件，再刷大雕的固件，
因为仍然无法刷好，所以还在试验中，希望能找到完整的刷机办法

AmibaWorld, 2020-03-21:
dnscrypt-proxy与dnsforwarder全 删除后，
在重刷 lean固件时候，log 有提示 dnscrypt-proxy 是dead directory

realoyrz, 2020-03-22:
7621的HWNAT的问题，大雕已经撤回了3.9号更新的Flow的版本，重新编译一下吧，不过现在好像HWNAT失效了

closed 2020-03-25 by coolsnowwolf