coolsnowwolf/lede #3817: 编译出错

warrenhoo, 2020-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
全局翻墙香港，换过不同出国方式,ssr或者v2rau，默认.config，无论是先make download下载了相关package再make，还是直接make -j1 V=s， 都出错，换了ubuntu 14  16，重装了几系统，都同样的错误，
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
See <http://bugs.openwrt.org/> for instructions.
tools/v8_gypfiles/v8_base_without_compiler.target.mk:590: recipe for target '/home/administrator/lede/build_dir/target-x86_64_musl/node-v12.15.0/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/api/api.o' failed
make[5]: *** [/home/administrator/lede/build_dir/target-x86_64_musl/node-v12.15.0/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/api/api.o] Error 4
rm dde9db9743ea72b7d89714abc892d153f5042fb7.intermediate 8f756472d5a33e62b78e30856570a3650d3c08f3.intermediate aed3096f8d3a07dd409ed332316fc782a8563de2.intermediate c4b3955b0171fa0da079ac4b78f94dbd161ce7b3.intermediate
Makefile:101: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/administrator/lede/build_dir/target-x86_64_musl/node-v12.15.0'
Makefile:150: recipe for target '/home/administrator/lede/build_dir/target-x86_64_musl/node-v12.15.0/.built' failed
make[3]: *** [/home/administrator/lede/build_dir/target-x86_64_musl/node-v12.15.0/.built] Error 2
make[3]: Leaving directory '/home/administrator/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#22.12#53.63#289.87
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/administrator/lede'
package/Makefile:107: recipe for target '/home/administrator/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/administrator/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/administrator/lede'
/home/administrator/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

hcym, 2020-03-15:
1804和2004都正常编译，应该还是代理的问题，尤其golang编译考验梯子质量

warrenhoo, 2020-03-15:
> 1804和2004都正常编译，应该还是代理的问题，尤其golang编译考验梯子质量

感谢感谢

0987363, 2020-03-17:
> 
> 
> 1804和2004都正常编译，应该还是代理的问题，尤其golang编译考验梯子质量

ubuntu 20.04下无法编译呢，go1.14编译出错，提示内核有bug
https://github.com/golang/go/issues/35777

hcym, 2020-03-17:
三月初的2004版本可以的，我编译几个机型，没有go问题，十五号的有系统依赖问题，这两天又退到1804了，等年底再说吧

closed 2020-07-04 by coolsnowwolf