coolsnowwolf/lede #3819: 挂载overlay失败，重启50次才成功挂载两次，手动挂载可以成功但是会丢失所有配置

764763903a, 2020-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
挂载overlay失败，重启50次才成功挂载两次，手动挂载可以成功但是会丢失所有配置
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
挂载overlay失败，重启50次才成功挂载两次，手动挂载可以成功但是会丢失所有配置
### (1) 具体问题
挂载点中已经将sda2挂载为overlay，生效后完成配置，安装一部分软件后重启就GG，就又挂载不上去，反复重启了三十次什么方式都试了都不行，但是手动挂载可以，但是手动挂载后虽然软件包有增加，但是我的配置已经全部丢失，希望百忙之中能解决这个问题，谢谢，挂载不上overlay实在太不方便。
A：

### (2) 路由器型号和固件版本

P&W R619AC 128M
OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7)
--
A：

### (3) 详细日志

A：   13.727833] mount_root: loading kmods from internal overlay
[   13.758276] kmodloader: loading kernel modules from //etc/modules-boot.d/*
[   13.761812] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
[   14.767118] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" started, PID 188
[   14.852618] UBIFS (ubi0:2): recovery needed
[   15.060087] UBIFS (ubi0:2): recovery completed
[   15.060404] UBIFS (ubi0:2): UBIFS: mounted UBI device 0, volume 2, name "rootfs_data"
[   15.063481] UBIFS (ubi0:2): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[   15.071583] UBIFS (ubi0:2): FS size: 70471680 bytes (67 MiB, 555 LEBs), journal size 3555328 bytes (3 MiB, 28 LEBs)
[   15.081342] UBIFS (ubi0:2): reserved for root: 3328551 bytes (3250 KiB)
[   15.091574] UBIFS (ubi0:2): media format: w4/r0 (latest is w5/r0), UUID 7CCB65C8-5FAA-49DC-83BB-5B5F4ED8D379, small LPT model
[   15.101328] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
[   15.119990] block: extroot: device not present, retrying in 5 seconds
[   20.134706] block: extroot: cannot find device with UUID 4edbea99-d71f-46b1-baa4-34a0f9944a68
[   20.176875] UBIFS (ubi0:2): un-mount UBI device 0
[   20.177470] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
[   20.193856] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" started, PID 190
[   20.379229] UBIFS (ubi0:2): UBIFS: mounted UBI device 0, volume 2, name "rootfs_data"
[   20.379318] UBIFS (ubi0:2): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[   20.386238] UBIFS (ubi0:2): FS size: 70471680 bytes (67 MiB, 555 LEBs), journal size 3555328 bytes (3 MiB, 28 LEBs)
[   20.395995] UBIFS (ubi0:2): reserved for root: 3328551 bytes (3250 KiB)
[   20.406239] UBIFS (ubi0:2): media format: w4/r0 (latest is w5/r0), UUID 7CCB65C8-5FAA-49DC-83BB-5B5F4ED8D379, small LPT model
[   20.419560] mount_root: loading kmods from internal overlay
[   20.451220] kmodloader: loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
[   20.454077] kmodloader: done loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
[   21.128602] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
[   21.135445] block: extroot: device not present, retrying in 5 seconds
[   26.151020] block: extroot: cannot find device with UUID 4edbea99-d71f-46b1-baa4-34a0f9944a68
[   26.217345] mount_root: switching to ubifs overlay
[   26.262969] overlayfs: upper fs does not support xattr, falling back to index=off and metacopy=off.
[   26.268377] urandom-seed: Seeding with /etc/urandom.seed
[   26.470624] procd: - early -
[   26.470761] procd: - watchdog -
[   27.130507] procd: - watchdog -
[   27.131566] procd: - ubus -
[   27.277215] procd: - init -
[   28.191076] kmodloader: loading kernel modules from /etc/modules.d/*
[   28.191475] urngd: v1.0.2 started.
[   28.217025] NET: Registered protocol family 38
[   28.233393] NET: Registered protocol family 15
[   28.236612] Initializing XFRM netlink socket
[   28.239127] tun: Universal TUN/TAP device driver, 1.6
[   28.302717] cryptodev: driver 1.10 loaded.
[   28.304317] Mirror/redirect action on
[   28.316138] u32 classifier
[   28.316167]     input device check on
[   28.317749]     Actions configured
[   28.348770] fuse init (API version 7.27)
[   28.356595] Loading modules backported from Linux version v4.19.98-0-gd183c8e2647a
[   28.356629] Backport generated by backports.git v4.19.98-1-0-g8204eb99
[   28.450678] usbcore: registered new interface driver ums-alauda
[   28.452539] usbcore: registered new interface driver ums-cypress
[   28.457411] usbcore: registered new interface driver ums-datafab
[   28.463472] usbcore: registered new interface driver ums-freecom
[   28.469616] usbcore: registered new interface driver ums-isd200
[   28.475721] usbcore: registered new interface driver ums-jumpshot
[   28.481105] usbcore: registered new interface driver ums-karma
[   28.487946] usbcore: registered new interface driver ums-sddr09
[   28.493216] usbcore: registered new interface driver ums-sddr55
[   28.499274] usbcore: registered new interface driver ums-usbat
[   28.504315] usbcore: registered new interface driver usblp
[   28.548371] xt_time: kernel timezone is -0000
[   28.603993] PPP generic driver version 2.4.2
[   28.605460] PPP MPPE Compression module registered
[   28.608961] NET: Registered protocol family 24
[   37.084711] firmware ath10k!QCA4019!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   37.908389] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   37.908435] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   37.920314] ath10k_ahb a000000.wifi: firmware ver 10.4-3.6-00140 api 5 features no-p2p,mfp,peer-flow-ctrl,btcoex-param,allows-mesh-bcast,no-ps crc32 ba79b746
[   37.966903] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:16 crc32 6ce1f887
[   39.363762] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
[   39.388640] ath: EEPROM regdomain: 0x0
[   39.388661] ath: EEPROM indicates default country code should be used
[   39.388673] ath: doing EEPROM country->regdmn map search
[   39.388696] ath: country maps to regdmn code: 0x3a
[   39.388712] ath: Country alpha2 being used: US
[   39.388724] ath: Regpair used: 0x3a
[   39.938207] firmware ath10k!QCA4019!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   39.938538] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   39.946166] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   39.960928] ath10k_ahb a800000.wifi: firmware ver 10.4-3.6-00140 api 5 features no-p2p,mfp,peer-flow-ctrl,btcoex-param,allows-mesh-bcast,no-ps crc32 ba79b746
[   40.004542] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:17 crc32 6ce1f887
[   40.174916] sd 0:0:0:0: tag#1 uas_eh_abort_handler 0 uas-tag 2 inflight: CMD IN 
[   40.174964] sd 0:0:0:0: tag#1 CDB: opcode=0x9e, sa=0x10 9e 10 00 00 00 00 00 00 00 00 00 00 00 20 00 00
[   40.214932] scsi host0: uas_eh_device_reset_handler start
[   40.365250] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[   40.566927] scsi host0: uas_eh_device_reset_handler success
[   40.588676] sd 0:0:0:0: [sda] 31246336 4096-byte logical blocks: (128 GB/119 GiB)
[   40.589628] sd 0:0:0:0: [sda] Write Protect is off
[   40.595183] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
[   40.597017] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[   40.600960] sd 0:0:0:0: [sda] Optimal transfer size 268431360 bytes
[   40.618248]  sda: sda1 sda2 sda3
[   40.628868] sd 0:0:0:0: [sda] Attached SCSI removable disk
[   41.411603] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
[   41.437242] ath: EEPROM regdomain: 0x0
[   41.437264] ath: EEPROM indicates default country code should be used
[   41.437275] ath: doing EEPROM country->regdmn map search
[   41.437298] ath: country maps to regdmn code: 0x3a
[   41.437316] ath: Country alpha2 being used: US
[   41.437328] ath: Regpair used: 0x3a
[   41.449377] kmodloader: done loading kernel modules from /etc/modules.d/*
[   44.558339] xt_FULLCONENAT: RFC3489 Full Cone NAT module
[   44.558339] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
[   46.350564] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: 
[   46.854311] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   46.856696] device eth0 entered promiscuous mode
[   46.859518] br-lan: port 1(eth0.1) entered blocking state
[   46.863818] br-lan: port 1(eth0.1) entered disabled state
[   46.869527] device eth0.1 entered promiscuous mode
[   46.876216] br-lan: port 1(eth0.1) entered blocking state
[   46.879183] br-lan: port 1(eth0.1) entered forwarding state

764763903a, 2020-03-15:
[21.135445]块：extroot：设备不存在，在5秒内重试
[26.151020]块：extroot：找不到具有UUID 4edbea99-d71f-46b1-baa4-34a0f9944a68
[26.217345]的设备mount_root：切换到ubifs overlay
主要是这个，但是启动后通过ssh挂载又能成功挂载 UUID是正确的

shenzt68, 2020-03-15:
我也有这样问题

mjyhj, 2020-03-15:
128M分区还不够用？？

closed 2020-03-25 by coolsnowwolf