coolsnowwolf/lede #3820: 恳请各位大佬帮忙瞅瞅编译错误！！谢谢！！

TendernessRadif, 2020-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译

### (2) 路由器型号和固件版本

A：ARM3328

### (3) 详细日志
src/Makefile.in:44: recipe for target 'src/libuClibc++.a' failed
make[4]: *** [src/libuClibc++.a] Error 1
make[4]: Leaving directory '/home/lbx/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/uClibc++-0.2.5'
Makefile:97: recipe for target '/home/lbx/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/uClibc++-0.2.5/.built' failed
make[3]: *** [/home/lbx/friendlywrt-rk3328/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/uClibc++-0.2.5/.built] Error 2
make[3]: Leaving directory '/home/lbx/friendlywrt-rk3328/friendlywrt/package/libs/uclibc++'
time: package/libs/uclibc++/compile#0.26#0.05#0.48
package/Makefile:111: recipe for target 'package/libs/uclibc++/compile' failed
make[2]: *** [package/libs/uclibc++/compile] Error 2
make[2]: Leaving directory '/home/lbx/friendlywrt-rk3328/friendlywrt'
package/Makefile:107: recipe for target '/home/lbx/friendlywrt-rk3328/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/home/lbx/friendlywrt-rk3328/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lbx/friendlywrt-rk3328/friendlywrt'
/home/lbx/friendlywrt-rk3328/friendlywrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
Command exited with non-zero status 1
you take 2:44:37 to build friendlywrt
lbx@lbx:~/friendlywrt-rk3328$ 

A：

qyh214, 2020-03-15:
尝试下先`make clean` 再`make j8`

TendernessRadif, 2020-03-15:
> 尝试下先`make clean` 再`make j8`

老哥我有尝试过 DL之类的全部下载全了

closed 2020-03-17 by coolsnowwolf