coolsnowwolf/lede #3821: 小娱C5 问题

chiyu521, 2020-03-15:
刚刚更新的源码 编译后 内置SATA好用 USB接入硬盘找不到了~
不知道什么原因~

chiyu521, 2020-03-16:
补充下
[  409.976596] usb 2-1: new SuperSpeed USB device number 3 using xhci-mtk
[  410.020179] scsi host2: uas
[  410.027323] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 4
[  410.044141] xhci-mtk 1e1c0000.xhci: @000000000f6d68a0 0e63f200 00000000 05000000 01058000
[  410.060410] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 6
[  410.077187] xhci-mtk 1e1c0000.xhci: @000000000f6d68b0 0e63f100 00000000 05000000 01078000

AmadeusGhost, 2020-03-17:
看了下日志，别选上`kmod-usb-storage-uas`试试，这个会被`automount`依赖上，所以也别选`automount`，然后把这几个都选上`block-mount kmod-usb-storage kmod-usb-storage-extras kmod-fs-vfat kmod-fs-ext4 kmod-fs-exfat +ntfs-3g`

chiyu521, 2020-03-17:
感谢楼上大佬 我去测试下 回来报告效果

chiyu521, 2020-03-17:
编译完毕也刷好了~~ USB3.0移动硬盘问题解决 感谢AmadeusGhost大佬提示

closed 2020-03-17 by chiyu521