coolsnowwolf/lede #3822: 跟新后编译出错

shbjian, 2020-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译几次，每次都卡这儿，不知怎么解决？百度无果

### (2) 路由器型号和固件版本

A：x64

### (3) 详细日志

A：sed -i "s/Installed-Time: .*/Installed-Time: 1584111140/" /home/sun/openwrt/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/status
rm -rf /home/sun/openwrt/lede/build_dir/target-x86_64_musl/root-x86/tmp/*
rm -f /home/sun/openwrt/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/lists/*
rm -f /home/sun/openwrt/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/info/*.postinst*
rm -f /home/sun/openwrt/lede/build_dir/target-x86_64_musl/root-x86/var/lock/*.lock
rm -rf /home/sun/openwrt/lede/build_dir/target-x86_64_musl/root-x86/boot
find /home/sun/openwrt/lede/build_dir/target-x86_64_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1584111140" "{}" +
find: The relative path 'AVA_HOME/bin' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/sun/openwrt/lede'
package/Makefile:108: recipe for target '/home/sun/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/sun/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/sun/openwrt/lede'
/home/sun/openwrt/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

xxp2277, 2020-03-15:
百度搜索不到日志中英文的有用信息吧，用google。
估计你是用WSL编译的吧，Please remove that entry from $PATH，让你删除$PATH变量中相关的东东。

> $ sudo vim /etc/wsl.conf
> # 不加载Windows中的PATH内容
> [interop]
> appendWindowsPath = false

https://lpwmm.blog.csdn.net/article/details/104326310

shbjian, 2020-03-18:
我用的ubuntu16.04，我装的双系统，在vps上编译也是一样

xxp2277, 2020-03-18:
既然提示是PATH变量的问题就看看你的PATH变量呢，我这里的PATH变量
```
$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
```

closed 2020-03-26 by coolsnowwolf