coolsnowwolf/lede #3825: JMS 机场节点在ShadowSocksR Plus+ 无法使用

warzh1, 2020-03-15:
不知道有没有用搬瓦工JMS 机场的小伙伴，ShadowSocksR Plus+ 节点显示运行，但是都无法连接，节点是正常可以使用的，没有被墙，koolss和merlin 系统都可以正常使用，想请教下大神们，该如何配置解决。

感谢

SharerMax, 2020-03-16:
1. 用的是那个代码版本的固件？（固件版本，app版本）
2. 节点信息是啥（v2ray?, ss?, ssr？用插件了没？等等..）
3. 如果用了ss的SIP003插件，固件有没有安装所需插件？
4. 有没有什么可用的日志？

kl9800, 2020-03-16:
我的jms没问题呀，我是x86 64    j1900

warzh1, 2020-03-16:
> 我的jms没问题呀，我是x86 64 j1900

你用的是shadowsocks R+ 的插件吗

warzh1, 2020-03-16:
> 1. 用的是那个代码版本的固件？（固件版本，app版本）
> 2. 节点信息是啥（v2ray?, ss?, ssr？用插件了没？等等..）
> 3. 如果用了ss的SIP003插件，固件有没有安装所需插件？
> 4. 有没有什么可用的日志？

JMS 机场只有SS的节点， 用的就是lean最新的固件，日志暂时没有 您说的SIP003是什么插件

SharerMax, 2020-03-16:
> > 1. 用的是那个代码版本的固件？（固件版本，app版本）
> > 2. 节点信息是啥（v2ray?, ss?, ssr？用插件了没？等等..）
> > 3. 如果用了ss的SIP003插件，固件有没有安装所需插件？
> > 4. 有没有什么可用的日志？
> 
> JMS 机场只有SS的节点， 用的就是lean最新的固件，日志暂时没有 您说的SIP003是什么插件

simple-obfs(obfs-local), v2ray-plugin 等等

warzh1, 2020-03-16:
> > 我的jms没问题呀，我是x86 64 j1900
> 
> 你用的是shadowsocks R+ 的插件吗 我和你的配置是一模一样，你用的固件版本可以分享一下吗

> > > 1. 用的是那个代码版本的固件？（固件版本，app版本）
> > > 2. 节点信息是啥（v2ray?, ss?, ssr？用插件了没？等等..）
> > > 3. 如果用了ss的SIP003插件，固件有没有安装所需插件？
> > > 4. 有没有什么可用的日志？
> > 
> > 
> > JMS 机场只有SS的节点， 用的就是lean最新的固件，日志暂时没有 您说的SIP003是什么插件
> 
> simple-obfs(obfs-local), v2ray-plugin 等等

我要怎么安装，大哥我小白，能不能给个指点。

kl9800, 2020-03-16:
> > 我的jms没问题呀，我是x86 64 j1900
> 
> 你用的是shadowsocks R+ 的插件吗 我和你的配置是一模一样，你用的固件版本可以分享一下吗

我目前用的就是3月7号左右自己编译的那版（自己没修改过任何东西），你可以回到那几次提交，然后编译一下，再不行就估计是你节点问题了

warzh1, 2020-03-16:
> 译的那版（自己没修改过任何东西），你可以回到那几次提交，然后编译一下，再不行就估计是你节点问题了

节点应该没有问题，我用的梅林路由固件和Koolshare Lede 的KOOLSS都可以正常使用

kl9800, 2020-03-16:
你用的是最新版本么，可以退回去试试，我用的不是当下最新版，昨天刚退回去的

LEFrost, 2020-03-16:
我的也没问题，jms

SharerMax, 2020-03-16:
> > > > 1. 用的是那个代码版本的固件？（固件版本，app版本）
> > > > 2. 节点信息是啥（v2ray?, ss?, ssr？用插件了没？等等..）
> > > > 3. 如果用了ss的SIP003插件，固件有没有安装所需插件？
> > > > 4. 有没有什么可用的日志？
> > > 
> > > 
> > > JMS 机场只有SS的节点， 用的就是lean最新的固件，日志暂时没有 您说的SIP003是什么插件
> > 
> > 
> > simple-obfs(obfs-local), v2ray-plugin 等等
> 
> 我要怎么安装，大哥我小白，能不能给个指点。

先确认是否是插件没安装的问题
查看软件包已安装列表，是否有相关插件
若没有，则需要安装插件

安装插件有有两种方式
1. 使用包含插件的固件
2. 找和你路由架构相同的相关ipk文件自己安装

jyy5206, 2020-03-16:
同JMS机场。
所有 5 个节点，在 Netgear 4300V1 和 X86 上均无问题。

注意：
JMS 的第 3、第 4 个节点不是纯 SS，是 SS+obfs 组合。
需要输入 obfs 参数：

> 插件：obfs-local
> 插件参数：obfs=tls;obfs-host=www.bing.com

自 R20.1.15 版本（即2020年1月15日）之后编译出来的，都是有 SS+obfs 组合功能了。

你再试试看？

warzh1, 2020-03-17:
感谢各位大神，我的X86软路由刷挂了，正在翻修，我先关掉issue，等路由器拿回来接着试

closed 2020-03-17 by warzh1

lynnial, 2020-03-24:
> 同JMS机场。
> 所有 5 个节点，在 Netgear 4300V1 和 X86 上均无问题。
> 
> 注意：
> JMS 的第 3、第 4 个节点不是纯 SS，是 SS+obfs 组合。
> 需要输入 obfs 参数：
> 
> > 插件：obfs-local
> > 插件参数：obfs=tls;obfs-host=www.bing.com
> 
> 自 R20.1.15 版本（即2020年1月15日）之后编译出来的，都是有 SS+obfs 组合功能了。
> 
> 你再试试看？

回去试一下，还以为SSR+不支持SS+obfs，差点刷回LEDE