coolsnowwolf/lede #3826: OpenWrt 18.06 中subscribe.lua找不到uci模块问题

agutools, 2020-03-15:
## 1.关于你要提交的问题
OpenWrt 18.06 中subscribe.lua找不到uci模块问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X ] 没有类似的issue

## 2. 详细叙述
SSH截图：
[image]

### (1) 具体问题
已安装luc、uci，请问require 'uci'中的uci是什么包？

### (2) 路由器型号和固件版本
主机型号：DomyWifi DW33D
固件版本：OpenWrt 18.06.6 r7957-d81a8a3e29 / LuCI openwrt-18.06 branch (git-20.064.26341-4fc4c79)
内核版本：4.9.208

### (3) 详细日志

	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:10: in main chunk
	[C]: in function 'require'
stack traceback:
	no file '/usr/lib/lua/loadall.so'
	no file '/usr/lib/lua/uci.so'
	no file './uci.so'
	no file '/usr/lib/lua/uci/init.lua'
	no file '/usr/lib/lua/uci.lua'
	no file '/usr/share/lua/uci/init.lua'
	no file '/usr/share/lua/uci.lua'
	no file './uci.lua'
	no field package.preload['uci']
/usr/bin/lua: /usr/share/shadowsocksr/subscribe.lua:10: module 'uci' not found:

agutools, 2020-03-16:
已找到：libuci-lua

closed 2020-03-16 by agutools

Zack0617, 2020-03-16:
> 已找到：libuci-lua

怎么解决的？

agutools, 2020-03-17:
> > 已找到：libuci-lua
> 
> 怎么解决的？

opkg update && opkg install libuci-lua

Zack0617, 2020-03-17:
> > > 已找到：libuci-lua
> > 
> > 
> > 怎么解决的？
> 
> opkg update && opkg install libuci-lua

谢谢，大佬已经在最新版本的依赖中添加了libuci-lua，直接编译后就ok