coolsnowwolf/lede #3828: gfw无法起作用

miroda, 2020-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：gfw开启之后，进入全局模式，腾讯视频等应用显示没有版权无法观看

### (2) 路由器型号和固件版本

A：Intel(R) Celeron(R) CPU J1900 @ 1.99GHz : 4 Core 4 Thread
OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7)

### (3) 详细日志

A：日志不知道在哪里看

SharerMax, 2020-03-16:
你在说什么？
> gfw开启之后，进入全局模式 

到底是GFW模式还是全局模式？

miroda, 2020-03-16:
我开启的是gfw模式，实际上变成了全局模式。

SharerMax, 2020-03-16:
无法复现
有使用其它服务么

miroda, 2020-03-16:
我这边一直是这个情况 ，我不知道为何，lean的固件 之前都是好的，后面就不行了。

miroda, 2020-03-16:
腾讯和国内的服务都被代理了 ，为何？
如何排查这个问题，是否有办法？

miroda, 2020-03-16:
我就ssr plus+这一个 服务

EnnawYang, 2020-03-16:
实测并没有发现任何异常

miroda, 2020-03-16:
是否能有看日志的方法，帮助排查一下这个问题的原因

miroda, 2020-03-16:
[image]

miroda, 2020-03-16:
[image]

miroda, 2020-03-16:
[image]

SharerMax, 2020-03-16:
看看访问控制，强制代理的域名

miroda, 2020-03-16:
dlercloud.me
cn.engadget.com
www.keystonejs.com
https://justmysocks.net/

miroda, 2020-03-16:
找到问题了，最后一行 造成的，去掉就好了 ，啊，受不了了

SharerMax, 2020-03-16:
这里填的域名不是url，注意一下

closed 2020-03-16 by coolsnowwolf