coolsnowwolf/lede #3831: 编译失败

linxuankent, 2020-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：多次尝试编译失败

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：make[1]: Entering directory '/home/lin/ceshi/lede'
make[2]: Entering directory '/home/lin/ceshi/lede'
+ mkdir -p /home/lin/ceshi/lede/staging_dir/target-i386_pentium4_musl
+ cd /home/lin/ceshi/lede/staging_dir/target-i386_pentium4_musl
+ mkdir -p bin lib include stamp
mkdir -p /home/lin/ceshi/lede/build_dir/target-i386_pentium4_musl/stamp
touch /home/lin/ceshi/lede/staging_dir/target-i386_pentium4_musl/.prepared
+ mkdir -p /home/lin/ceshi/lede/staging_dir/host
+ cd /home/lin/ceshi/lede/staging_dir/host
+ mkdir -p bin lib include stamp
mkdir -p /home/lin/ceshi/lede/build_dir/host/stamp /home/lin/ceshi/lede/staging_dir/host/include/sys
install -m0644 /home/lin/ceshi/lede/tools/include/*.h /home/lin/ceshi/lede/staging_dir/host/include/
install -m0644 /home/lin/ceshi/lede/tools/include/sys/*.h /home/lin/ceshi/lede/staging_dir/host/include/sys/
ln -sf lib /home/lin/ceshi/lede/staging_dir/host/lib64
touch /home/lin/ceshi/lede/staging_dir/host/.prepared
make[3]: Entering directory '/home/lin/ceshi/lede/tools/flock'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/flock'
time: tools/flock/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/xz'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/xz'
time: tools/xz/compile#0.04#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/sed'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/sed'
time: tools/sed/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/patch'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/patch'
time: tools/patch/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/tar'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/tar'
time: tools/tar/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/m4'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/m4'
time: tools/m4/compile#0.02#0.02#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/autoconf'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/autoconf'
time: tools/autoconf/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/pkg-config'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/pkg-config'
time: tools/pkg-config/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/automake'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/automake'
time: tools/automake/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/missing-macros'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/missing-macros'
time: tools/missing-macros/compile#0.02#0.00#0.03
make[3]: Entering directory '/home/lin/ceshi/lede/tools/libtool'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/libtool'
time: tools/libtool/compile#0.04#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/gmp'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/gmp'
time: tools/gmp/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/mpfr'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/mpfr'
time: tools/mpfr/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/mpc'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/mpc'
time: tools/mpc/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/libelf'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/libelf'
time: tools/libelf/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/expat'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/expat'
time: tools/expat/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/flex'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/flex'
time: tools/flex/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/bison'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/bison'
time: tools/bison/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/mklibs'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/mklibs'
time: tools/mklibs/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/libressl'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/libressl'
time: tools/libressl/compile#0.04#0.01#0.05
make[3]: Entering directory '/home/lin/ceshi/lede/tools/cmake'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/cmake'
time: tools/cmake/compile#0.06#0.06#0.08
make[3]: Entering directory '/home/lin/ceshi/lede/tools/zlib'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/zlib'
time: tools/zlib/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/sstrip'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/sstrip'
time: tools/sstrip/compile#0.02#0.00#0.03
make[3]: Entering directory '/home/lin/ceshi/lede/tools/make-ext4fs'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/make-ext4fs'
time: tools/make-ext4fs/compile#0.09#0.03#0.12
make[3]: Entering directory '/home/lin/ceshi/lede/tools/e2fsprogs'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/e2fsprogs'
time: tools/e2fsprogs/compile#0.03#0.02#0.05
make[3]: Entering directory '/home/lin/ceshi/lede/tools/mtd-utils'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/mtd-utils'
time: tools/mtd-utils/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/mkimage'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/mkimage'
time: tools/mkimage/compile#0.07#0.05#0.08
make[3]: Entering directory '/home/lin/ceshi/lede/tools/firmware-utils'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/firmware-utils'
time: tools/firmware-utils/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/patch-image'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/patch-image'
time: tools/patch-image/compile#0.02#0.00#0.03
make[3]: Entering directory '/home/lin/ceshi/lede/tools/findutils'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/findutils'
time: tools/findutils/compile#0.04#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/quilt'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/quilt'
time: tools/quilt/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/padjffs2'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/padjffs2'
time: tools/padjffs2/compile#0.02#0.00#0.03
make[3]: Entering directory '/home/lin/ceshi/lede/tools/mm-macros'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/mm-macros'
time: tools/mm-macros/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/bc'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/bc'
time: tools/bc/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/gengetopt'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/gengetopt'
time: tools/gengetopt/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/patchelf'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/patchelf'
time: tools/patchelf/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/mtools'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/mtools'
time: tools/mtools/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/dosfstools'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/dosfstools'
time: tools/dosfstools/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/ucl'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/ucl'
time: tools/ucl/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/upx'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/upx'
time: tools/upx/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede'
make[3]: *** tools/qemu: No such file or directory.  Stop.
make[3]: Leaving directory '/home/lin/ceshi/lede'
time: tools/qemu/compile#0.00#0.00#0.00
tools/Makefile:161: recipe for target 'tools/qemu/compile' failed
make[2]: *** [tools/qemu/compile] Error 2
make[2]: Leaving directory '/home/lin/ceshi/lede'
tools/Makefile:159: recipe for target '/home/lin/ceshi/lede/staging_dir/target-i386_pentium4_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyynynnyyyynynyyy' failed
make[1]: *** [/home/lin/ceshi/lede/staging_dir/target-i386_pentium4_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyynynnyyyynynyyy] Error 2
make[1]: Leaving directory '/home/lin/ceshi/lede'
/home/lin/ceshi/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

SharerMax, 2020-03-16:
貌似是qemu的代码不存在，请确保编译环境能够正常下载所需代码，并确保下载代码的完整性

1. 检查dl目录下是否有qemu的代码，如果有请删除，重新编译（此举的目的是防止qemu的代码不完整，强制重新下载）

linxuankent, 2020-03-16:
> 貌似是qemu的代码不存在，请确保编译环境能够正常下载所需代码，并确保下载代码的实现
> 
> 1. 检查dl目录下是否有qemu的代码，如果有请删除，重新编译（此举的目的是防止qemu的代码不完整，强制重新下载）

dl目录下没有qemu

SharerMax, 2020-03-16:
那就下载
```
make download V=s
```

linxuankent, 2020-03-16:
> 那就下载
> 
> ```
> make download V=s
> ```
执行完还是一样报错

linxuankent, 2020-03-16:
> > 那就下载
> > ```
> > make download V=s
> > ```
> 
> 执行完还是一样报错

make[1]: Entering directory '/home/lin/ceshi/lede'
make[2]: Entering directory '/home/lin/ceshi/lede/tools/gmp'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/gmp'
time: tools/gmp/download#0.02#0.00#0.11
make[2]: Entering directory '/home/lin/ceshi/lede/tools/mpfr'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/mpfr'
time: tools/mpfr/download#0.01#0.00#0.04
make[2]: Entering directory '/home/lin/ceshi/lede/tools/mpc'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/mpc'
time: tools/mpc/download#0.02#0.00#0.02
make[2]: Entering directory '/home/lin/ceshi/lede/tools/libelf'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/libelf'
time: tools/libelf/download#0.02#0.00#0.03
make[2]: Entering directory '/home/lin/ceshi/lede/tools/expat'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/expat'
time: tools/expat/download#0.02#0.00#0.03
make[2]: Entering directory '/home/lin/ceshi/lede/tools/m4'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/m4'
time: tools/m4/download#0.02#0.00#0.07
make[2]: Entering directory '/home/lin/ceshi/lede/tools/libtool'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/libtool'
time: tools/libtool/download#0.02#0.00#0.04
make[2]: Entering directory '/home/lin/ceshi/lede/tools/autoconf'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/autoconf'
time: tools/autoconf/download#0.01#0.00#0.03
make[2]: Entering directory '/home/lin/ceshi/lede/tools/automake'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/automake'
time: tools/automake/download#0.02#0.00#0.04
make[2]: Entering directory '/home/lin/ceshi/lede/tools/flex'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/flex'
time: tools/flex/download#0.02#0.00#0.04
make[2]: Entering directory '/home/lin/ceshi/lede/tools/bison'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/bison'
time: tools/bison/download#0.01#0.01#0.06
make[2]: Entering directory '/home/lin/ceshi/lede/tools/pkg-config'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/pkg-config'
time: tools/pkg-config/download#0.02#0.01#0.05
make[2]: Entering directory '/home/lin/ceshi/lede/tools/mklibs'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/mklibs'
time: tools/mklibs/download#0.01#0.01#0.03
make[2]: Entering directory '/home/lin/ceshi/lede/tools/zlib'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/zlib'
time: tools/zlib/download#0.01#0.00#0.11
make[2]: Entering directory '/home/lin/ceshi/lede/tools/sstrip'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/tools/sstrip'
time: tools/sstrip/download#0.01#0.00#0.03
make[2]: Entering directory '/home/lin/ceshi/lede/tools/make-ext4fs'
make[2]: Nothing to be done for 'download'.
[... 803 lines cut ...]
make[2]: Entering directory '/home/lin/ceshi/lede/package/utils/f2fs-tools'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/package/utils/f2fs-tools'
time: package/utils/f2fs-tools/download#0.07#0.01#0.08
make[2]: Entering directory '/home/lin/ceshi/lede/package/utils/jsonfilter'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/package/utils/jsonfilter'
time: package/utils/jsonfilter/download#0.20#0.06#0.25
make[2]: Entering directory '/home/lin/ceshi/lede/package/utils/lua'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/package/utils/lua'
time: package/utils/lua/download#0.08#0.03#0.10
make[2]: Entering directory '/home/lin/ceshi/lede/package/utils/util-linux'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/lin/ceshi/lede/package/utils/util-linux'
time: package/utils/util-linux/download#0.08#0.02#0.10
make[1]: Leaving directory '/home/lin/ceshi/lede'
make[1]: Entering directory '/home/lin/ceshi/lede'
make[2]: Entering directory '/home/lin/ceshi/lede/target/linux'
make[3]: Entering directory '/home/lin/ceshi/lede/target/linux/x86'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/home/lin/ceshi/lede/target/linux/x86'
make[2]: Leaving directory '/home/lin/ceshi/lede/target/linux'
time: target/linux/download#0.03#0.00#0.04
make[1]: Leaving directory '/home/lin/ceshi/lede'

SharerMax, 2020-03-17:
试试最新代码

linxuankent, 2020-03-17:
git pull 
./scripts/feeds update -a 
 ./scripts/feeds install -a
全部执行完还是报错
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/mm-macros'
time: tools/mm-macros/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/bc'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/bc'
time: tools/bc/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/gengetopt'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/gengetopt'
time: tools/gengetopt/compile#0.02#0.02#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/patchelf'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/patchelf'
time: tools/patchelf/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/mtools'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/mtools'
time: tools/mtools/compile#0.03#0.01#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/dosfstools'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/dosfstools'
time: tools/dosfstools/compile#0.03#0.00#0.04
make[3]: Entering directory '/home/lin/ceshi/lede/tools/ucl'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/ucl'
time: tools/ucl/compile#0.03#0.01#0.11
make[3]: Entering directory '/home/lin/ceshi/lede/tools/upx'
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/upx'
time: tools/upx/compile#0.02#0.02#0.05
make[3]: Entering directory '/home/lin/ceshi/lede/tools/qemu'
make[3]: *** No rule to make target 'compile'.  Stop.
make[3]: Leaving directory '/home/lin/ceshi/lede/tools/qemu'
time: tools/qemu/compile#0.00#0.00#0.00
tools/Makefile:161: recipe for target 'tools/qemu/compile' failed
make[2]: *** [tools/qemu/compile] Error 2
make[2]: Leaving directory '/home/lin/ceshi/lede'
tools/Makefile:159: recipe for target '/home/lin/ceshi/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyynynnyyyynynyyy' failed
make[1]: *** [/home/lin/ceshi/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyynynnyyyynynyyy] Error 2
make[1]: Leaving directory '/home/lin/ceshi/lede'
/home/lin/ceshi/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

SharerMax, 2020-03-17:
执行下面的命令后，重新尝试编译
```
make clean
```

linxuankent, 2020-03-17:
> 执行下面的命令后，重新尝试编译
> 
> ```
> make clean
> ```

一样报错

ElectronicElephant, 2020-03-17:
Same problem.

SharerMax, 2020-03-17:
#3856 
会不会是这个

linxuankent, 2020-03-17:
> [＃3856](https://github.com/coolsnowwolf/lede/pull/3856)
> 这个这个要怎么使用？

linxuankent, 2020-03-17:
> [＃3856](https://github.com/coolsnowwolf/lede/pull/3856)
> 会不会是这个

尝试替换了代码还是报错

luxiaok, 2020-03-18:
编译x86的固件依赖qemu，可以手动下载到dl目录

> https://download.qemu.org/qemu-0.14.1.tar.gz

即便不编译虚拟机镜像也会需要这个。

linxuankent, 2020-03-18:
> 编译x86的固件依赖qemu，可以手动下载到dl目录
> 
> > https://download.qemu.org/qemu-0.14.1.tar.gz
> 
> 即便不编译虚拟机镜像（`vmdk`，`qcow2`，`vdi`等格式）也会需要这个。
还是报一样的错，文件名要改吗？

luxiaok, 2020-03-18:
不用改名，看这几个位置有文件或者目录没：

* dl/qemu-0.14.1.tar.gz (可以先解压一下，验证下文件的完整性)

* build_dir/host/qemu-0.14.1

* staging_dir/host/bin/qemu-img

**其他软件包报错也可以用类似的方法检查一下，墙里面编译经常发生文件不完整或者因下载超时而出错的问题。**

ElectronicElephant, 2020-03-18:
> 
> 
> Same problem.

Solved.

linxuankent, 2020-03-18:
> 不用改名，看这几个位置有文件或目录没：
> 
> * dl / qemu-0.14.1.tar.gz（可以先解压一下，验证下文件的本质）
> * build_dir / host / qemu-0.14.1
> * staging_dir / host / bin / qemu-img
> 
> **其他副本报错也可以使用类似的方法检查一下，墙里面编译经常发生文件不完整或因下载超时而出错的问题。**

我重新建了目录重新git，重新做了一遍，还在编译，暂时没有报错。

linxuankent, 2020-03-19:
问题已解决

closed 2020-03-19 by linxuankent

EriTang, 2020-12-27:
> 不用改名，看这几个位置有文件或者目录没：
> 
> * dl/qemu-0.14.1.tar.gz (可以先解压一下，验证下文件的完整性)
> * build_dir/host/qemu-0.14.1
> * staging_dir/host/bin/qemu-img
> 
> **其他软件包报错也可以用类似的方法检查一下，墙里面编译经常发生文件不完整或者因下载超时而出错的问题。**

staging_dir / host / bin / qemu-img

你好，请问这个目录不存在该怎么做