coolsnowwolf/lede #3832: luci-app-adbyby规则更新的问题

wangzhuok, 2020-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：luci-app-adbyby规则更新的问题
可否添加自定义更新时间的设置，另外可否增加更新第三方规则的计划任务并且可以自定义时间
现在只能手动添加计划任务，并不影响使用，添加这一功能可以改善界面友好程度。

### (2) 路由器型号和固件版本

A：x64

### (3) 详细日志

A：

523034406, 2020-03-16:
很赞同，对于更新时间其实可以更加细化，支持设置

liwenjie119, 2020-03-17:
目前可以在源码package/lean/luci-app-adbyby-plus/root/etc/init.d/adbyby修改52行
`echo '0 5 * * * /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1' >> $CRON_FILE`
改成自己想要的时间，这样编译完成的包自动更新时间就会错开，重启、更新也不会恢复6点。

wangzhuok, 2020-03-17:
> 目前可以在源码package/lean/luci-app-adbyby-plus/root/etc/init.d/adbyby修改52行
> `echo '0 5 * * * /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1' >> $CRON_FILE`
> 改成自己想要的时间，这样编译完成的包自动更新时间就会错开，重启、更新也不会恢复6点。

这个会改，但是不会提交commits，大佬这么忙估计小bug都没时间改
手动添加第三方规则更新的计划任务：
0 2 * * * /usr/share/adbyby/rule-update && /usr/share/adbyby/rule-count '/usr/share/adbyby/rules/'

closed 2020-07-04 by coolsnowwolf