coolsnowwolf/lede #3835: 841v8编译没有报错，无法生成固件，请问还需要修改哪些

acd889, 2020-03-16:
define Device/tl-wr841-v8
  $(Device/tplink-16mlzma)
  DEVICE_TITLE := TP-LINK TL-WR841N/ND v8
  BOARDNAME := TL-WR841N-v8
  DEVICE_PROFILE := TLWR841
  TPLINK_HWID := 0x08410008
我改成16m大小，生成不了固件，以前可以，更新源码以后不行了，谁帮忙指点一下还需要改哪些

acd889, 2020-03-16:
重新编译了一遍，成功了

closed 2020-03-17 by coolsnowwolf

fengchen-github, 2024-11-16:
> define Device/tl-wr841-v8 $(Device/tplink-16mlzma) DEVICE_TITLE := TP-LINK TL-WR841N/ND v8 BOARDNAME := TL-WR841N-v8 DEVICE_PROFILE := TLWR841 TPLINK_HWID := 0x08410008 我改成16m大小，生成不了固件，以前可以，更新源码以后不行了，谁帮忙指点一下还需要改哪些

最近几个版本image-builder固件都是循环重启，大佬遇到过吗，分区表还要改吗

fengchen-github, 2024-11-16:
> > define Device/tl-wr841-v8 $(Device/tplink-16mlzma) DEVICE_TITLE := TP-LINK TL-WR841N/ND v8 BOARDNAME := TL-WR841N-v8 DEVICE_PROFILE := TLWR841 TPLINK_HWID := 0x08410008 我改成16m大小，生成不了固件，以前可以，更新源码以后不行了，谁帮忙指点一下还需要改哪些
> 
> 最近几个版本image-builder固件都是循环重启，大佬遇到过吗，分区表还要改吗

`$ nano target/linux/ath79/dts/ar9341_tplink_tl-wr841-v8.dts
&spi {
        status = "okay";

        flash@0 {
                compatible = "jedec,spi-nor";
                reg = <0>;
                spi-max-frequency = <25000000>;

                partitions {
                        compatible = "fixed-partitions";
                        #address-cells = <1>;
                        #size-cells = <1>;

                        uboot: partition@0 {
                                label = "u-boot";
                                reg = <0x000000 0x020000>;
                                read-only;
                        };

                        partition@20000 {
                                compatible = "tplink,firmware";
                                label = "firmware";
                                reg = <0x020000 0xfd0000>;
                        };

                        art: partition@ff0000 {
                                label = "art";
                                reg = <0xff0000 0x010000>;
                                read-only;
                        };
                };
        };
};
`
`$ nano target/linux/ath79/image/tiny-tp-link.mk
define Device/tplink_tl-wr841-v8
  $(Device/tplink-16mlzma)
  SOC := ar9341
  DEVICE_MODEL := TL-WR841N/ND
  DEVICE_VARIANT := v8
  TPLINK_HWID := 0x08410008
  SUPPORTED_DEVICES += tl-wr841n-v8
endef
TARGET_DEVICES += tplink_tl-wr841-v8
`

需要重新编译，可以使用[openwrt-toolchain](https://downloads.openwrt.org/releases/23.05.5/targets/ath79/tiny/openwrt-toolchain-23.05.5-ath79-tiny_gcc-12.3.0_musl.Linux-x86_64.tar.xz)现成的工具链编译，会快一些
make menuconfig设置工具链如下：
 Advanced configuration options (for developers)  ---> 
Use external toolchain  --->
  │ │ (mips-openwrt-linux) Target name                                                                                     
  │ │   (mips-openwrt-linux-musl-) Toolchain prefix                                                                          
  │ │    (/home/debian/openwrt-make/openwrt/openwrt-toolchain/toolchain-mips_24kc_gcc-12.3.0_musl) Toolchain root