coolsnowwolf/lede #3836: 没人遇到过这个问题吗，来说说我新发现 当手机第一次安装网易云音乐打开能正常解锁，但是当手机杀掉这个进程或者重启之后 就不能解锁了，这是不是某些代码中的bug

1-github-1, 2020-03-16:
没人遇到过这个问题吗，来说说我新发现 当手机第一次安装网易云音乐打开能正常解锁，但是当手机杀掉这个进程或者重启之后 就不能解锁了，这是不是某些代码中的bug

_Originally posted by @1-github-1 in https://github.com/coolsnowwolf/lede/issues/3747#issuecomment-599425434_

camino13, 2020-03-16:
在下面添加个例外
[image]

IP地址填入你运行UnblockNeteaseMusic的局域网IP

1-github-1, 2020-03-16:
> 在下面添加个例外
> [image]
> 
> IP地址填入您运行UnblockNeteaseMusic的初始化IP

不行啊大神 你的可以吗 我是电脑能正常解锁 手机不设置代理不行

xxp2277, 2020-03-16:
> 在下面添加个例外
> [image]
> 
> IP地址填入你运行UnblockNeteaseMusic的局域网IP

搭车问下，在外面使用ssr连回家里面的路由器，云音乐怎么设置才能解锁。

Zack0617, 2020-03-16:
> > 在下面添加个例外
> > [image]
> > IP地址填入你运行UnblockNeteaseMusic的局域网IP
> 
> 搭车问下，在外面使用ssr连回家里面的路由器，云音乐怎么设置才能解锁。

softether vpn回家，不需要额外设置就可以云音乐解锁。

closed 2020-03-16 by coolsnowwolf