coolsnowwolf/lede #3837: 解锁网易云音乐三合一NodeJS版编译失败

505799690, 2020-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
环境：
Gitpod
配置：
CONFIG_PACKAGE_luci-app-unblockmusic=y
CONFIG_UnblockNeteaseMusic_Go=y
CONFIG_UnblockNeteaseMusic_NodeJS=y
没勾这三个的时候编译成功
这3个勾上后报错，好像是说node-v12.15.0有个.h找不到？
之后取消了CONFIG_UnblockNeteaseMusic_NodeJS还是报错
又取消了CONFIG_UnblockNeteaseMusic_Go继续报错
之后尝试make clean后不勾CONFIG_UnblockNeteaseMusic_NodeJS继续报错
尝试rm -rf ./build_dir/target-mipsel_24kc_musl/node-v12.15.0 无果
工作空间炸了233

### (2) 路由器型号和固件版本

A：歌华链

### (3) 详细日志

A：没有记录

whlsxl, 2020-03-16:
更新新的依赖了吗？

505799690, 2020-03-16:
> 更新新的依赖了吗？

更新了的

whlsxl, 2020-03-16:
如果自行编译，推荐docker。
不会扰乱现在环境https://github.com/whlsxl/lede_docker

505799690, 2020-03-17:
感觉可能是依赖问题

closed 2020-03-17 by 505799690

kivid, 2021-08-14:
> 感觉可能是依赖问题

是缺少依赖，centos8的话需安装这些依赖 `sudo yum install glibc-devel.i686 libgcc.i686 libstdc++-devel.i686`